OpenNuvoton/NUC980-linux-4.4.y #1: I2C reads on i2c-p2 and i2c-p3 hang (but writes work great)

jaydcarlson, 2019-08-01:
I'm trying to communicate with some I2C devices attached to I2C2, using PB5 as SCL and PB7 as SDA. I have a simple temperature sensor with an address of 0x18 attached to the bus.

If I write to the device:
```
i2cset -y 2 0x18 0x00 0x00
```
I get no problems:
[image]

But if I read from the device:
```
i2cget -y 2 0x18 0x00
```
The bus hangs:
[image]

Not shown in the above timescale, but SDA times out after ~25ms and returns high. SCL never gets released, though.

I've tested this on multiple I2C slave devices, and originally discovered the problem after noticing my kernel modules for the InvenSense MPU-6050 and Analog Devices AD5933 were both hanging when I attempt reads.

I switched to I2C3, but it has the same problem.

Any ideas?

yachen, 2019-08-02:
Hi,

Please pull  0ee9bda   and test again. Thanks.

Sincerely,

Yi-An Chen

jaydcarlson, 2019-08-02:
Thanks! Works a treat!

closed 2019-08-02 by jaydcarlson