OpenNuvoton/NUC980-linux-4.4.y #2: SPI-NAND下如何啟用SD卡0

1020165558, 2019-08-03:
在內核menuconfig中並沒有選擇SD0的選項卡，文檔中也只是提及
以下是SD卡介面使能的設置. NUC980支援一個 SD 卡介面
Device Drivers ---> 
<*>MMC/SD/SDIO card support --->
 <*> MMC block device driver 
 <*> Use bounce buffer for simple hosts
 <*> Nuvoton NUC980 SD Card support
沒有選擇用哪一個SD卡的地方

yachen, 2019-08-03:
Hi,
Another SD port share controller with NAND. Please enable it by select Device Drivers -> Generic Driver Options -> Nuvoton NUC980 FMI function selection -> Select FMI device to support (Support SD0 Device). 
Sincerely,

Yi-An Chen

closed 2019-08-12 by yachen

1020165558, 2019-08-19:
非常感謝，按您的方法可以正常使用@Yi-An Chen

xiaolu88856, 2020-09-30:
除了配置FMI，还要配置其他么？

yachen, 2020-09-30:
@xiaolu88856 ,

也要開以下這些:
<*>MMC/SD/SDIO card support ---> 
<*> MMC block device driver 
<*> Use bounce buffer for simple hosts

xiaolu88856, 2020-09-30:
感谢回复！不过我还是没识别出来，
首先硬件是没问题的，因为uboot可以识别。
日志这边：
mmc0: clock 0Hz busmode 2 powermode 1 cs 0 Vdd 21 width 1 timing 0
nuc980-fmi nuc980-fmi: Initial signal voltage of 3.3v
mmc0: clock 400000Hz busmode 2 powermode 2 cs 0 Vdd 21 width 1 timing 0
mmc0: clock 0Hz busmode 2 powermode 0 cs 0 Vdd 0 width 1 timing 0
然后就没有了关于mmc的信息了

xiaolu88856, 2020-09-30:
@yachen 方便留个联系方式么？

yachen, 2020-09-30:
請問你用什麼板子測試? 我們這邊使用 EVB 照這樣設置. 只要接上卡, 就會有訊息.
Uboot 目前 driver 沒處理 CD pin, 會不會是 CD 沒接上?