OpenNuvoton/NUC980-linux-4.4.y #3: NUC980 IOT SPI NAND 上是否支持ubifs根文件系统

tongshi203, 2019-08-05:
你好：
     在NUC980 IOT SPI NAND 上是否支不支持ubifs根文件系统？
     我在启动板子时出现错误如下：
NUC980 Crypto engine enabled.
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
nand: Winbond W25N01GV 1G 3.3V
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
Scanning device for bad blocks
nuc980-nadc: nuc980 Normal ADC adapter
nuc980-audio nuc980-audio: nau8822-hifi <-> nuc980-audio-i2s mapping ok
NET: Registered protocol family 17
lib80211: common routines for IEEE802.11 drivers
UBI error: cannot open mtd 2, error -19
ALSA device list:
  #0: nuc980_IIS
UBIFS error (pid: 1): cannot open "ubi0:user", error -19VFS: Cannot open root device "ubi0:user" or unknown-block(0,0): error -19
Please append a correct "root=" boot option; here are the available partitions:
0100           16384 ram0  (driver?)
0101           16384 ram1  (driver?)
0102           16384 ram2  (driver?)
0103           16384 ram3  (driver?)
0104           16384 ram4  (driver?)
0105           16384 ram5  (driver?)
0106           16384 ram6  (driver?)
0107           16384 ram7  (driver?)
0108           16384 ram8  (driver?)
0109           16384 ram9  (driver?)
010a           16384 ram10  (driver?)
010b           16384 ram11  (driver?)
010c           16384 ram12  (driver?)
010d           16384 ram13  (driver?)
010e           16384 ram14  (driver?)
010f           16384 ram15  (driver?)
1f00          131072 mtdblock0  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #28
Hardware name: NUC980
Backtrace: 
[<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
 r6:c3b2e000 r5:c3b2e000 r4:c04a87a4 r3:00000000
[<c0013000>] (show_stack) from [<c01b3b3c>] (dump_stack+0x20/0x28)
[<c01b3b1c>] (dump_stack) from [<c0075dc0>] (panic+0xa4/0x234)
[<c0075d20>] (panic) from [<c0523218>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fed8 r0:c04a87a4
 r7:c3fed5c0
[<c0522ff4>] (mount_block_root) from [<c05234e4>] (prepare_namespace+0x98/0x1c0)
 r10:00000074 r9:c053f838 r8:c0546e4c r7:c053f830 r6:c053f858 r5:c053f858
 r4:c053f868

yachen, 2019-08-05:
可以支持, 有其他客戶使用, 以下是我們在 SPI NAND 放 rootfs 使用的 bootarg
bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel

tongshi203, 2019-08-06:
谢谢你的回答！
但是使用这个bootargs还是提示如下的错误： 麻烦指导！谢谢
U-Boot 2016.11-g3ac85d0-dirty (Jul 22 2019 - 11:19:39 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x400000
SF: 4194304 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.179+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2779944 Bytes = 2.7 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.179+ (wcl@localhost.localdomain) (gcc version 4.8.4 (GCC) ) #28 PREEMPT Mon Aug 5 11:08:41 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c058e080, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd ubi.mtd=2 root=ubi0:user rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-bool
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 59004K/65536K available (3954K kernel code, 282K rwdata, 1268K rodata, 152K init, 217K bss, 6532K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0521f9c   (5224 kB)
      .init : 0xc0522000 - 0xc0548000   ( 152 kB)
      .data : 0xc0548000 - 0xc058e8a0   ( 283 kB)
       .bss : 0xc058e8a0 - 0xc05c4cd8   ( 218 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 139 lines cut ...]
1f00            2048 mtdblock0  (driver?)
1f01           20480 mtdblock1  (driver?)
1f02          108544 mtdblock2  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #28
Hardware name: NUC980
Backtrace: 
[<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
 r6:c3b99000 r5:c3b99000 r4:c04a87a4 r3:00000000
[<c0013000>] (show_stack) from [<c01b3b3c>] (dump_stack+0x20/0x28)
[<c01b3b1c>] (dump_stack) from [<c0075dc0>] (panic+0xa4/0x234)
[<c0075d20>] (panic) from [<c0523218>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fed8 r0:c04a87a4
 r7:c3fee320
[<c0522ff4>] (mount_block_root) from [<c05234e4>] (prepare_namespace+0x98/0x1c0)
 r10:00000074 r9:c053f838 r8:c0546e4c r7:c053f830 r6:c053f858 r5:c053f858
 r4:c053f868
[<c052344c>] (prepare_namespace) from [<c0522e40>] (kernel_init_freeable+0x184/0x1cc)
 r6:c058e8a0 r5:c058e8a0 r4:00000007
[<c0522cbc>] (kernel_init_freeable) from [<c03dea98>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03dea88
 r4:00000000
[<c03dea88>] (kernel_init) from [<c000fd58>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:c383e000
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

tongshi203, 2019-08-06:
谢谢你的指导！
但是我用了bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel 
还是出错如下：
U-Boot 2016.11-g3ac85d0-dirty (Jul 22 2019 - 11:19:39 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x400000
SF: 4194304 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.179+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2779944 Bytes = 2.7 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.179+ (wcl@localhost.localdomain) (gcc version 4.8.4 (GCC) ) #28 PREEMPT Mon Aug 5 11:08:41 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c058e080, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd ubi.mtd=2 root=ubi0:user rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-bool
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 59004K/65536K available (3954K kernel code, 282K rwdata, 1268K rodata, 152K init, 217K bss, 6532K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0521f9c   (5224 kB)
      .init : 0xc0522000 - 0xc0548000   ( 152 kB)
      .data : 0xc0548000 - 0xc058e8a0   ( 283 kB)
       .bss : 0xc058e8a0 - 0xc05c4cd8   ( 218 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
[... 148 lines cut ...]
1f01           20480 mtdblock1  (driver?)
1f02          108544 mtdblock2  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #28
Hardware name: NUC980
Backtrace: 
[<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
 r6:c3b4a000 r5:c3b4a000 r4:c04a87a4 r3:00000000
[<c0013000>] (show_stack) from [<c01b3b3c>] (dump_stack+0x20/0x28)
[<c01b3b1c>] (dump_stack) from [<c0075dc0>] (panic+0xa4/0x234)
[<c0075d20>] (panic) from [<c0523218>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fed8 r0:c04a87a4
 r7:c3fed940
[<c0522ff4>] (mount_block_root) from [<c05234e4>] (prepare_namespace+0x98/0x1c0)
 r10:00000074 r9:c053f838 r8:c0546e4c r7:c053f830 r6:c053f858 r5:c053f858
 r4:c053f868
[<c052344c>] (prepare_namespace) from [<c0522e40>] (kernel_init_freeable+0x184/0x1cc)
 r6:c058e8a0 r5:c058e8a0 r4:00000007
[<c0522cbc>] (kernel_init_freeable) from [<c03dea98>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03dea88
 r4:00000000
[<c03dea88>] (kernel_init) from [<c000fd58>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:c383e000
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
random: nonblocking pool is initialized

yachen, 2019-08-06:
bootargs 是  root=ubi0:system, 但 console log 是 root=ubi0,  kernel 沒收 uboot 傳來的 arg?
以下是我們使用的 rootfs cfg 設定.  vol_name 要跟 bootarg 裡的相同. 

[rootfs-volume]
mode=ubi
image=rootfs_ubifs.img
vol_id=0
vol_size=92946432
vol_type=dynamic
vol_name=system
vol_flags=autoresize

tongshi203, 2019-08-06:
1.利用下面命令生成ubi
mkfs.ubifs -x lzo -m 2048 -e 126976 -c 732 -o rootfs_ubifs.img -d ../rootfs
ubinize -o ubi.img -m 2048 -p 131072 -O 2048 -s 2048 ubinize.cfg 
[root@localhost mk_ubifs]# cat ubinize.cfg
[rootfs-volume]
mode=ubi
image=rootfs_ubifs.img
vol_id=0
vol_size=92946432
vol_type=dynamic
vol_name=system
vol_flags=autoresize
2. 烧写文件（SPI Nand）：
u-boot-spl.bin   (Image Type：uBoot型  Image start offset： 0x0)
u-boot.bin       (Image Type：Data型  Image start offset： 0x100000)
980uimage     (Image Type：Data型  Image start offset： 0x200000)
env.txt            (Image Type：Environment型  Image start offset： 0x80000)
ubi.img           (Image Type：Data型  Image start offset： 0x1600000) 
env.txt 文件内容如下：
baudrate=115200
bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
bootdelay=3
stderr=serial
stdin=serial
stdout=serial
setspi=sf probe 0 30000000
loadkernel=sf read 0x7fc0 0x200000 0x400000
bootcmd=run setspi;run loadkernel;bootm 0x7fc0

烧写成功后第一次是能正常启动了，log如下：
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow�load

 SPL load main U-Boot from SPI NAND Flash! (Jul 22 2019 11:20:34)

U-Boot 2016.11-g3ac85d0-dirty (Jul 22 2019 - 11:19:39 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x400000
SF: 4194304 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.179+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2779944 Bytes = 2.7 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.179+ (wcl@localhost.localdomain) (gcc version 4.8.4 (GCC) ) #28 PREEMPT Mon Aug 5 11:08:41 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
[... 845 lines cut ...]
1f02          108544 mtdblock2  (driver?)
No filesystem could mount root, tried:  ubifs
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #28
Hardware name: NUC980
Backtrace: 
[<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
 r6:c3b4c006 r5:c3b4c006 r4:c04a87a4 r3:00000000
[<c0013000>] (show_stack) from [<c01b3b3c>] (dump_stack+0x20/0x28)
[<c01b3b1c>] (dump_stack) from [<c0075dc0>] (panic+0xa4/0x234)
[<c0075d20>] (panic) from [<c0523288>] (mount_block_root+0x294/0x2d8)
 r3:c383fea4 r2:80000000 r1:c383fed8 r0:c04a87a4
 r7:c3fed980
[<c0522ff4>] (mount_block_root) from [<c05234e4>] (prepare_namespace+0x98/0x1c0)
 r10:00000074 r9:c053f838 r8:c0546e4c r7:c053f830 r6:c053f858 r5:c053f858
 r4:c053f868
[<c052344c>] (prepare_namespace) from [<c0522e40>] (kernel_init_freeable+0x184/0x1cc)
 r6:c058e8a0 r5:c058e8a0 r4:00000007
[<c0522cbc>] (kernel_init_freeable) from [<c03dea98>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03dea88
 r4:00000000
[<c03dea88>] (kernel_init) from [<c000fd58>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:c383e000
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
random: nonblocking pool is initialized

tongshi203, 2019-08-06:
1.利用下面命令生成ubi
mkfs.ubifs -x lzo -m 2048 -e 126976 -c 732 -o rootfs_ubifs.img -d ../rootfs
ubinize -o ubi.img -m 2048 -p 131072 -O 2048 -s 2048 ubinize.cfg
[root@localhost mk_ubifs]# cat ubinize.cfg
[rootfs-volume]
mode=ubi
image=rootfs_ubifs.img
vol_id=0
vol_size=92946432
vol_type=dynamic
vol_name=system
vol_flags=autoresize
2. 烧写文件（SPI Nand）：
u-boot-spl.bin (Image Type：uBoot型 Image start offset： 0x0)
u-boot.bin (Image Type：Data型 Image start offset： 0x100000)
980uimage (Image Type：Data型 Image start offset： 0x200000)
env.txt (Image Type：Environment型 Image start offset： 0x80000)
ubi.img (Image Type：Data型 Image start offset： 0x1600000)
env.txt 文件内容如下：
baudrate=115200
bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
bootdelay=3
stderr=serial
stdin=serial
stdout=serial
setspi=sf probe 0 30000000
loadkernel=sf read 0x7fc0 0x200000 0x400000
bootcmd=run setspi;run loadkernel;bootm 0x7fc0

烧写成功后第一次是能正常启动了，log如下：（reboot重起arm板子，就又出错了）
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow�load

SPL load main U-Boot from SPI NAND Flash! (Jul 22 2019 11:20:34)

U-Boot 2016.11-g3ac85d0-dirty (Jul 22 2019 - 11:19:39 +0800)

CPU: NUC980
Board: NUC980
DRAM: 64 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In: serial
Out: serial
Err: serial
Net: Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot: 0
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x400000
SF: 4194304 bytes @ 0x200000 Read: OK
Booting kernel from Legacy Image at 00007fc0 ...

Image Name: Linux-4.4.179+
Image Type: ARM Linux Kernel Image (uncompressed)
Data Size: 2779944 Bytes = 2.7 MiB
Load Address: 00008000
Entry Point: 00008000
Verifying Checksum ... OK
XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.179+ (wcl@localhost.localdomain) (gcc version 4.8.4 (GCC) ) #28 PREEMPT Mon Aug 5 11:08:41 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
[... 847 lines cut ...]
1f02 108544 mtdblock2 (driver?)
No filesystem could mount root, tried: ubifs
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #28
Hardware name: NUC980
Backtrace:
[] (dump_backtrace) from [] (show_stack+0x18/0x1c)
r6:c3b4c006 r5:c3b4c006 r4:c04a87a4 r3:00000000
[] (show_stack) from [] (dump_stack+0x20/0x28)
[] (dump_stack) from [] (panic+0xa4/0x234)
[] (panic) from [] (mount_block_root+0x294/0x2d8)
r3:c383fea4 r2:80000000 r1:c383fed8 r0:c04a87a4
r7:c3fed980
[] (mount_block_root) from [] (prepare_namespace+0x98/0x1c0)
r10:00000074 r9:c053f838 r8:c0546e4c r7:c053f830 r6:c053f858 r5:c053f858
r4:c053f868
[] (prepare_namespace) from [] (kernel_init_freeable+0x184/0x1cc)
r6:c058e8a0 r5:c058e8a0 r4:00000007
[] (kernel_init_freeable) from [] (kernel_init+0x10/0xf4)
r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03dea88
r4:00000000
[] (kernel_init) from [] (ret_from_fork+0x14/0x3c)
r4:00000000 r3:c383e000
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
random: nonblocking pool is initialized

yachen, 2019-08-06:
改 mdev.conf 或是 auromount.sh, 讓 yaffs2 不要來 mount mtd 試試. 這邊應該是已經 mount 好的 rootfs. 不應該出現下面這段 log.

mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags "inband-tags"
yaffs: yaffs: Attempting MTD mount of 31.2,"mtdblock2"
yaffs: 0 blocks to be sorted...
yaffs: yaffs_read_super: is_checkpointed 0
[Mount YAFFS2]: /dev/mtdblock2 --> /mnt/mtdblock2

tongshi203, 2019-08-07:
我在编译NUC980-linux-4.4y时，取消了yaffs2 file system support 选项，只选择UBIFS file system support,第一次启动提示如下错误，reboot后又起不来
ubi0: attaching mtd2
ubi0: scanning is finished
ubi0: volume 0 ("system") re-sized from 732 to 824 LEBs
ubi0: attached mtd2 (name "user", size 106 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 848, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 1, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 2124215395
ubi0: available PEBs: 0, total reserved PEBs: 848, PEBs reserved for bad PEB handling: 20
ubi0: background thread "ubi_bgt0d" started, PID 1092
ALSA device list:
  #0: nuc980_IIS
UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 1094
UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "system"
UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
UBIFS (ubi0:0): FS size: 91549696 bytes (87 MiB, 721 LEBs), journal size 9023488 bytes (8 MiB, 72 LEBs)
UBIFS (ubi0:0): reserved for root: 0 bytes (0 KiB)
UBIFS (ubi0:0): media format: w4/r0 (latest is w4/r0), UUID 8D7D35BC-B912-418C-8C56-A8044F10AD4E, small LPT model
VFS: Mounted root (ubifs filesystem) on device 0:13.
devtmpfs: mounted
Freeing unused kernel memory: 144K
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
UBIFS error (pid: 1119): cannot open "/dev/mtdblock2", error -22mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: Invalid argument

tongshi203, 2019-08-07:
请问 你有没有详细的关于在NUC980 iot 上挂载ubifs的操作文档？ 谢谢！！！

yachen, 2019-08-07:
抓整包Linux BSP裡的文件試試: https://www.nuvoton.com/hq/products/microprocessors/arm9-mpus/nuc980-industrial-control-iot-series/Software/?__locale=zh_TW&resourcePage=Y

tongshi203, 2019-08-07:
还是一样的错误
ubi0: attaching mtd2
ubi0: scanning is finished
ubi0: volume 0 ("system") re-sized from 732 to 824 LEBs
ubi0: attached mtd2 (name "user", size 106 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 848, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 1, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 2124215395
ubi0: available PEBs: 0, total reserved PEBs: 848, PEBs reserved for bad PEB handling: 20
ubi0: background thread "ubi_bgt0d" started, PID 1092
ALSA device list:
#0: nuc980_IIS
UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 1094
UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "system"
UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
UBIFS (ubi0:0): FS size: 91549696 bytes (87 MiB, 721 LEBs), journal size 9023488 bytes (8 MiB, 72 LEBs)
UBIFS (ubi0:0): reserved for root: 0 bytes (0 KiB)
UBIFS (ubi0:0): media format: w4/r0 (latest is w4/r0), UUID 8D7D35BC-B912-418C-8C56-A8044F10AD4E, small LPT model
VFS: Mounted root (ubifs filesystem) on device 0:13.
devtmpfs: mounted
Freeing unused kernel memory: 144K
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
UBIFS error (pid: 1119): cannot open "/dev/mtdblock2", error -22mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: Invalid argument

yachen, 2019-08-07:
把mdev.conf 跟MTD相P那行移除。console 已@示rootfs 已mount成功了。

Yi-An Chen

The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

tongshi203, 2019-08-08:
mdev.conf 文件中没有MTD 相关的， 烧完程序后是可以进入系统的，但是reboot后就出错了

[root@localhost rootfs]# find ./ -name "mdev.conf"
./etc/mdev.conf
[root@localhost rootfs]# cat etc/mdev.conf 
mmcblk([0-9]+)p([0-9]+) 0:0 660 */sbin/automount.sh $MDEV X${ACTION}
mmcblk([0-9]+)          0:0 660 */sbin/automount.sh $MDEV X${ACTION}
sd([a-z]+)([0-9]+)      0:0 660 */sbin/automount.sh $MDEV X${ACTION}
sd([a-z]+)              0:0 660 */sbin/automount.sh $MDEV X${ACTION}
mtdblock([2-9]+)        0:0 660 */sbin/automount.sh $MDEV X${ACTION}

reboot后出错的信息：
NUC980 Crypto engine enabled.
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
nand: Winbond W25N01GV 1G 3.3V
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
Scanning device for bad blocks
Creating 3 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000001600000 : "Kernel"
0x000001600000-0x000008000000 : "user"
nuc980-nadc: nuc980 Normal ADC adapter
nuc980-audio nuc980-audio: nau8822-hifi <-> nuc980-audio-i2s mapping ok
NET: Registered protocol family 17
lib80211: common routines for IEEE802.11 drivers
ubi0: attaching mtd2
ubi0: scanning is finished
ubi0: attached mtd2 (name "user", size 106 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 848, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 1, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 2/0, WL threshold: 4096, image sequence number: 619663878
ubi0: available PEBs: 0, total reserved PEBs: 848, PEBs reserved for bad PEB handling: 20
ubi0: background thread "ubi_bgt0d" started, PID 1092
ALSA device list:
  #0: nuc980_IIS
mt29f spi0.0: ecc error, page=11460
UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 1095
spinand: ECC error
mt29f spi0.0: ecc error, page=11461
spinand: ECC error
ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 126976 bytes from PEB 3:4096, read only 126976 bytes, retry
mt29f spi0.0: ecc error, page=11460
spinand: ECC error
mt29f spi0.0: ecc error, page=11461
spinand: ECC error
ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 126976 bytes from PEB 3:4096, read only 126976 bytes, retry
mt29f spi0.0: ecc error, page=11460
spinand: ECC error
mt29f spi0.0: ecc error, page=11461
spinand: ECC error

yachen, 2019-08-08:
有啊, mtdblock.
mtdblock([2-9]+) 0:0 660 */sbin/automount.sh $MDEV X${ACTION}

tongshi203, 2019-08-08:
删除 mtdblock([2-9]+) 0:0 660 */sbin/automount.sh $MDEV X${ACTION}，但reboot后还是出错，提示ECC error

[root@localhost etc]# cat mdev.conf
mmcblk([0-9]+)p([0-9]+) 0:0 660 */sbin/automount.sh $MDEV X${ACTION}
mmcblk([0-9]+)          0:0 660 */sbin/automount.sh $MDEV X${ACTION}
sd([a-z]+)([0-9]+)      0:0 660 */sbin/automount.sh $MDEV X${ACTION}
sd([a-z]+)              0:0 660 */sbin/automount.sh $MDEV X${ACTION}

reboot 出错信息：
ubi0: attaching mtd2
ubi0: scanning is finished
ubi0: attached mtd2 (name "user", size 106 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 848, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 1, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 2011198537
ubi0: available PEBs: 0, total reserved PEBs: 848, PEBs reserved for bad PEB handling: 20
ubi0: background thread "ubi_bgt0d" started, PID 1093
ALSA device list:
  #0: nuc980_IIS
mt29f spi0.0: ecc error, page=11461
UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 1095
spinand: ECC error
ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 126976 bytes from PEB 3:4096, read only 126976 bytes, retry
mt29f spi0.0: ecc error, page=11461
spinand: ECC error

tongshi203, 2019-08-08:
能帮忙提供一个spi nand 模式下ubifs根文件系统编译NUC980-linux-4.4.y的.config文件吗？

tongshi203, 2019-08-08:
[config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3479700/config.txt)
这是我的.config文件 麻烦帮忙看一下是否有问题，谢谢！谢谢！！！

yachen, 2019-08-12:
spinand.txt是我使用的 kernel config. env.txt 是 u-boot 環境變數.
重新開機並不會有 ubi rootfs 損毀問題. 測試硬件是 NK-980IOT 板.
[env.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3490990/env.txt)
[spinand.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3490987/spinand.txt)

tongshi203, 2019-08-12:
谢谢您的回答！
问题已解决.修改了两个地方：
1. 编译内核时取消YAFFS2 文件系統設置，添加UBIFS文件系統設置
2. mkfs.ubifs工具改用github上提供的1.5.1版本

closed 2019-08-12 by yachen

GuoZhiyong, 2021-01-27:
我也遇到同样的问题。128MB yaffs2挂载太慢。ubifs总是挂载不成功。mtd2 attach总是失败。

pjeutr, 2021-05-23:
Thank you orokin, mkfs.ubifs -F fixes the `ubi0 warning: ubi_io_read: error -74 (ECC error) while reading ` for me