OpenNuvoton/NUC980-linux-4.4.y #4: NUC980 IOT 配置gadget驱动实现USB大容量存储，模拟U盘

tongshi203, 2019-08-12:
你好：
     NUC980 IOT 配置gadget驱动实现USB大容量存储，模拟U盘功能，操作如下：（请问哪里做的不正确？）

1.编译NUC980-linux-4.4.y： make menuconfig 配置如下;
USB Device設置 – Mass Storage Device
Device Drivers --->
   [*] USB support --->
        <*> USB Gadget Support --->
                USB Peripheral Controller --->
                      <*> NUC980 USB Device Controller
        <M> USB Gadget Driver
        <M> Mass Storage Gadget
2.将生成的configfs.ko  g_mass_storage.ko  libcomposite.ko  usb_f_mass_storage.ko
在板子上执行 
insmod configfs.ko
insmod libcomposite.ko
insmod usb_f_mass_storage.ko
insmod g_mass_storage.ko file=/dev/mmcblk0p1 stall=0 removable=1

运行结果如下：
Mass Storage Function, version: 2009/09/11
LUN: removable file: (no medium)
LUN: removable file: /dev/mmcblk0p1
Number of LUNs=1
g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
g_mass_storage gadget: userspace failed to provide iSerialNumber
g_mass_storage gadget: g_mass_storage ready
enable: 0x620
plug-in
enable: 0x80000620
plug-out
disable: 0x720
plug-out
disable: 0x620
plug-out
disable: 0x620
plug-out

yachen, 2019-08-12:
Hi, 什麼板子上測試的? 能換條USB線嗎? 一直跳enable/disable 不太正常.

tongshi203, 2019-08-13:
NUC980 IOT 上测试的, 对于USB线串口都有打印应该没问题。
/dev/mmcblk0p1 是一个usb卡，这需要进行其他的什么操作吗？除了上面提法的

tongshi203, 2019-08-13:
需要在编译内核时加上loop驱动，现已正常工作！谢谢！

closed 2019-08-13 by tongshi203