OpenNuvoton/NUC980-linux-4.4.y #6: 当根文件系统是ubifs时，是否支持shttpd？

tongshi203, 2019-08-15:
你好：
       NUC980 IOT 上，当根文件系统为nfs模式时，开发板子上运行：shttpd -root /mnt/mmcblk0p1/website/tongshi -ports 80 ，运行http://ip/index.html 是正常的；
      相同的程序，在ubifs模式下，板子上运行：shttpd -root /mnt/mmcblk0p1/website/tongshi -ports 80，出现Segmentation fault？
      其中shttpd版本为shttpd-1.42

yachen, 2019-08-15:
原則上文件系統不會影響 http server. 能用 strace 先看一下發生什麼問題嗎?

tongshi203, 2019-08-16:
~ # strace -tt -e  trace=process -f shttpd -root /mnt/mmcblk0p1/website/tongshi -ports 80
00:47:28.830816 execve("/bin/shttpd", ["shttpd", "-root", "/mnt/mmcblk0p1/website/tongshi", "-ports", "80"], 0xbef7ce20 /* 8 vars */) = 0
shttpd 1.42 started on port(s) 80, serving /mnt/mmcblk0p1/website/tongshi
00:47:28.911232 --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xdef3cba8} ---
00:47:28.913216 +++ killed by SIGSEGV +++
Segmentation fault

tongshi203, 2019-08-16:
~ # strace -i shttpd -root /mnt/mmcblk0p1/website/tongshi -ports 80
[b6e92bac] execve("/bin/shttpd", ["shttpd", "-root", "/mnt/mmcblk0p1/website/tongshi", "-ports", "80"], 0xbef47e34 /* 8 vars */) = 0
[b6f1b28c] mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0xb6f26000
[b6f1c444] open("/lib/libpthread.so.0", O_RDONLY) = 3
[b6f1c488] fstat(3, {st_mode=S_IFREG|0755, st_size=103434, ...}) = 0
[b6f1c668] mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0xb6f25000
[b6f1c4d4] read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\260C\0\0004\0\0\0"..., 4096) = 4096
[b6f1c894] mmap2(NULL, 114688, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6efd000
[b6f1cbd4] mmap2(0xb6efd000, 67104, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0xb6efd000
[b6f1ca30] mmap2(0xb6f15000, 4124, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x10000) = 0xb6f15000
[b6f1cb34] mmap2(0xb6f17000, 4328, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f17000
[b6f1ccf8] close(3)                     = 0
[b6f1cf50] munmap(0xb6f25000, 4096)     = 0
[b6f1c444] open("/lib/libc.so.0", O_RDONLY) = 3
[b6f1c488] fstat(3, {st_mode=S_IFREG|0755, st_size=593962, ...}) = 0
[b6f1c668] mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0xb6f25000
[b6f1c4d4] read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0p\301\0\0004\0\0\0"..., 4096) = 4096
[b6f1c894] mmap2(NULL, 479232, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e88000
[b6f1cbd4] mmap2(0xb6e88000, 417584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0xb6e88000
[b6f1ca30] mmap2(0xb6ef6000, 4892, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x66000) = 0xb6ef6000
[b6f1cb34] mmap2(0xb6ef8000, 19312, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ef8000
[b6f1ccf8] close(3)                     = 0
[b6f1cf50] munmap(0xb6f25000, 4096)     = 0
[b6f1c20c] stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=32667, ...}) = 0
[b6f1c444] open("/lib/libdl.so.0", O_RDONLY) = 3
[b6f1c488] fstat(3, {st_mode=S_IFREG|0755, st_size=16038, ...}) = 0
[b6f1c668] mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0xb6f25000
[b6f1c4d4] read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\20\v\0\0004\0\0\0"..., 4096) = 4096
[b6f1c894] mmap2(NULL, 49152, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e7c000
[b6f1cbd4] mmap2(0xb6e7c000, 8468, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0xb6e7c000
[b6f1ca30] mmap2(0xb6e86000, 4120, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0xb6e86000
[b6f1ccf8] close(3)                     = 0
[b6f1cf50] munmap(0xb6f25000, 4096)     = 0
[b6f1c444] open("/lib/libc.so.0", O_RDONLY) = 3
[b6f1c488] fstat(3, {st_mode=S_IFREG|0755, st_size=593962, ...}) = 0
[b6f1c5fc] close(3)                     = 0
[b6f1c444] open("/lib/libc.so.0", O_RDONLY) = 3
[b6f1c488] fstat(3, {st_mode=S_IFREG|0755, st_size=593962, ...}) = 0
[b6f1c5fc] close(3)                     = 0
[b6f1b28c] mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0xb6f25000
[b6f1b91c] set_tls(0xb6f254a0, 0xb6f25b40, 0xb6f25b48, 0xb6f254a0, 0xb6f27f74) = 0
[b6f1bfac] mprotect(0xb6f15000, 4096, PROT_READ) = 0
[b6f1bfac] mprotect(0xb6ef6000, 4096, PROT_READ) = 0
[b6f1bfac] mprotect(0xb6f27000, 4096, PROT_READ) = 0
[b6f1bfac] mprotect(0xb6e86000, 4096, PROT_READ) = 0
[b6f06788] set_tid_address(0xb6f25068)  = 1283
[b6f067b8] set_robust_list(0xb6f25070, 12) = 0
[b6f0242c] rt_sigaction(SIGRTMIN, {sa_handler=0xb6f06660, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0xb6e953c4}, NULL, 8) = 0
[b6f0242c] rt_sigaction(SIGRT_1, {sa_handler=0xb6f06500, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0xb6e953c4}, NULL, 8) = 0
[b6f0689c] rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
[b6e9663c] ugetrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM_INFINITY}) = 0
[b6e968a8] ioctl(0, TCGETS, {B115200 opost isig icanon echo ...}) = 0
[b6e968a8] ioctl(1, TCGETS, {B115200 opost isig icanon echo ...}) = 0
[b6f0242c] rt_sigaction(SIGCHLD, {sa_handler=0x9644, sa_mask=[CHLD], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6e953b8}, {sa_handler=SIG_DFL, sa_mask=[0
[b6f0242c] rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6e953b8}, {sa_handler=SIG_DFL, sa_mask=0
[b6f0242c] rt_sigaction(SIGTERM, {sa_handler=0x9644, sa_mask=[TERM], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6e953b8}, {sa_handler=SIG_DFL, sa_mask=[0
[b6f0242c] rt_sigaction(SIGINT, {sa_handler=0x9644, sa_mask=[INT], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6e953b8}, {sa_handler=SIG_DFL, sa_mask=[],0
[b6e94c10] brk(NULL)                    = 0xe98000
[b6e94c10] brk(0xe99000)                = 0xe99000
[b6ee1adc] open("shttpd.conf", O_RDONLY) = -1 ENOENT (No such file or directory)
[b6ed98d4] socket(AF_INET, SOCK_STREAM, IPPROTO_TCP) = 3
[b6e98ed4] fcntl64(3, F_GETFL)          = 0x2 (flags O_RDWR)
[b6e98ed4] fcntl64(3, F_SETFL, O_RDWR|O_NONBLOCK) = 0
[b6ed9854] setsockopt(3, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
[b6ed91f4] bind(3, {sa_family=AF_INET, sin_port=htons(80), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
[b6ed93b4] listen(3, 128)               = 0
[b6e98ed4] fcntl64(3, F_SETFD, FD_CLOEXEC) = 0
[b6ed98d4] socket(AF_INET, SOCK_STREAM, IPPROTO_IP) = 4
[... cut ...]
[b6ee1adc] open("/etc/TZ", O_RDONLY)    = -1 ENOENT (No such file or directory)
[b6ee1adc] open("/etc/localtime", O_RDONLY) = -1 ENOENT (No such file or directory)
[b6ee1adc] open("/etc/TZ", O_RDONLY)    = -1 ENOENT (No such file or directory)
[b6ee1adc] open("/etc/localtime", O_RDONLY) = -1 ENOENT (No such file or directory)
[b6e966f0] gettimeofday({tv_sec=3199, tv_usec=449066}, NULL) = 0
[0000d398] --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xdec63ba8} ---
[????????] +++ killed by SIGSEGV +++
Segmentation fault

tongshi203, 2019-08-16:
解决方法：ifconfig lo 127.0.0.1 up 
谢谢！！

closed 2019-08-16 by tongshi203