OpenNuvoton/NUC980-linux-4.4.y #7: Warm boot hangs on NUC980 IoT

jaydcarlson, 2019-08-17:
Cold boots (e.g., where I hit the physical "reset" pushbutton) work fine, but if I issue a `reboot` command from software, the board begins to boot up Linux again but hangs on what looks to be the `nuc980-ehci` initialization during start-up:
```
Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.179 (jay@linux) (gcc version 4.8.5 (Buildroot 2016.11.1-g600a114-dirty) ) #24 PREEMPT Thu Aug 15 21:44:57 CDT 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 58044K/65536K available (2870K kernel code, 233K rwdata, 936K rodata, 2600K init, 195K bss, 7492K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03bff1c   (3808 kB)
      .init : 0xc03c0000 - 0xc064a000   (2600 kB)
      .data : 0xc064a000 - 0xc0684400   ( 233 kB)
       .bss : 0xc0684400 - 0xc06b5340   ( 196 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8400 - 0x843c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource nuc980-timer5
NET: Registered protocol family 2
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered (default)
nuc980_gpio_probe - pdev = nuc980-gpio
nuc980-uart.0: ttyS0 at I/O 0x0 (irq = 36, base_baud = 750000) is a NUC980
brd: module loaded
libphy: nuc980_rmii0: probed
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
nuc980-ehci nuc980-ehci: Nuvoton NUC980 EHCI Host Controller
nuc980-ehci nuc980-ehci: new USB bus registered, assigned bus number 1
nuc980-ehci nuc980-ehci: irq 23, io mem 0xb0015000
```

yachen, 2019-08-19:
Hi, 

I cannot reproduce this issue here. What hardware are you using, IoT board? Did you plug in any USB device during the test? Can you give me your kernel config file? Thanks.

Sincerely,

Yi-An Chen

jaydcarlson, 2019-09-11:
Hey Yi-An,
This issue seems to have gone away after reflashing / power cycling / setting the board on the back burner for a while. Not sure what it was, but it's gone now! (and no, I didn't have any USB devices plugged in, and yes, it was the IoT board)

closed 2019-09-11 by jaydcarlson