OpenNuvoton/NUC980-linux-4.4.y #9: NUC980开始摄像头不成功

1020165558, 2019-09-25:
按照手册配置好IIC和VCAP
开机打印信息如下：
i2c i2c-4: Failed to register i2c client cap0_nt99141 at 0x2a (-16)
i2c i2c-4: Can't create device at 0x2a
i2c-gpio i2c-gpio.4: using pins 66 (SDA) and 65 (SCL)
nuvoton_cap0_device_probe - pdev = nuc980-videoin0
nuvoton_cap0_device_probe video0, main sensor registeration fail.
注册失败
IIC的管脚设置正确，测量并无波形，从上面的打印信息来看，好像是先配置摄像头的寄存器，再分配的IIC管脚。不知是否和这个相关。
另外插入SD卡时
打印
nuvoton_cap0_device_probe - pdev = nuc980-videoin0
CAP0_PD_PIN failed ret=-16
nuvoton_cap0_device_probe video0, main sensor registeration fail.
是不是CAP0_PD_PIN引脚的缘故。CAP0_PD_PIN引脚又是再哪里设置的呢？

yachen, 2019-09-25:
Hi, 請問你是用哪一張板子測試 CAP 功能?

1020165558, 2019-09-25:
自己做的板子。参考NuMaker NUC980 IIoT开发板
管脚分配为

SIICSDA       PC2
SIICClk             PC1
S_nRst             PC7
S_Vsync      PC6
S_PwDn      PC0
S_Hsync      PC5
S_D0      PC8
S_D1      PC9
S_D2      PC10
S_D3      PC11
S_D4      PC12
S_D5      PC13
S_D6      PC14
S_D7      PC15
S_Clk      PC3
S_PClk      PC4

yachen, 2019-09-26:
Hi,

請問menuconfig有開GPIO-based bitbanging I2C 設定嗎?

Sincerely,

Yi-An Chen

The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

yachen, 2019-09-30:
Hi, 

請pull 最新驅動. 在drivers/media/i2c/nuvoton/sensor0_XXXXXX.c 裡面可以設定 I2C device id ,
I2c0 => 0
I2C1 =>1
I2C2 =>2
I2C3 =>3
I2C_GPIO =>4

Sincerely,

Yi-An Chen

closed 2019-10-21 by yachen

gdyshi, 2020-10-22:
> 按照手册配置好IIC和VCAP
> 开机打印信息如下：
> i2c i2c-4: Failed to register i2c client cap0_nt99141 at 0x2a (-16)
> i2c i2c-4: Can't create device at 0x2a
> i2c-gpio i2c-gpio.4: using pins 66 (SDA) and 65 (SCL) [...]

请问你是看的哪个手册？能否在附件中带一份。我看了很多手册，但没有找到关于video的说明？ @1020165558  @yachen

yachen, 2020-10-23:
應該是看 NUC980 Linux 4.4 BSP User Manual. 你在哪張板子上測試的?

gdyshi, 2020-10-23:
我也是参考iiot板子，我参考NUC980 Linux 4.4 BSP User Manual手册配置，发现启动时根本没有调用nt91141的probe函数。
附件中为内核配置文件，因为github附件格式要求的原因，我将配置文件重命名了
[my.config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/5427709/my.config.txt)

我发现另一个现象：

menuconfig中已经配置了CONFIG_VIDEO0_NUC980宏，但nuc980-videoin0似乎没有注册成功，因为没有这个platform设备，ls /sys/bus/platform/devices/ 没有设备的话，是否表示nuc980-videoin0设备没有注册？

[image]

yachen, 2020-10-23:
IoT板子上沒有 sensor 的接口, 請確認一下你目前使用的 I2C 是使用 PC1 及 PC2, 才有跟 console message 顯示的設置匹配.
`i2c-gpio i2c-gpio.4: using pins 66 (SDA) and 65 (SCL)`