OpenNuvoton/NUC980-linux-4.4.y #10: SPI Nand Flash (W25N01GV)问题：Bad eraseblock

zyj9527, 2019-10-11:
NUC980DK61Y，使用SPI NandFlash W25N01GV
kernel启动log报出很多"Bad eraseblock", 而且/dev/mtdblock2设备读写也异常，
具体log如下：
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
nand: Winbond W25N01GV 1G 3.3V
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
Scanning device for bad blocks
Bad eraseblock 9 at 0x000000120000
Bad eraseblock 10 at 0x000000140000
Bad eraseblock 11 at 0x000000160000
Bad eraseblock 12 at 0x000000180000
Bad eraseblock 13 at 0x0000001a0000
Bad eraseblock 14 at 0x0000001c0000
Bad eraseblock 15 at 0x0000001e0000
Bad eraseblock 16 at 0x000000200000
...
...
Bad eraseblock 1000 at 0x000007d00000
Bad eraseblock 1001 at 0x000007d20000
Bad eraseblock 1002 at 0x000007d40000
Creating 3 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000001600000 : "Kernel"
0x000001600000-0x000008000000 : "user"
nuc980-nadc: nuc980 Normal ADC adapter
NET: Registered protocol family 17

yachen, 2019-10-14:
請透過 NuWriter 把 SPI NAND flash 整個 erase 乾淨, 看問題還在不在.

yachen, 2019-10-25:
請問是在980IoT 板子上測試的嗎?

zyj9527, 2019-10-25:
Thank you！
问题解决了：W25N01GV的SIO2和SIO3接10K pull high

closed 2019-10-25 by zyj9527