OpenNuvoton/NUC980-linux-4.4.y #11: linux boot very slow

ifq, 2019-10-16:
Hi

I am using a 980uart2P board. I have changed the origin W25N  SPI-nand to a W25Q128 SPI FLASH.  
I'm using the buildroot from NUC. It can run, but the problem is it is very slow during the kernel initialization.

```
[    2.559516] usb_hcd_nuc980_probe, id = 4, name: nuc980-ohci.4, 0
[    2.566666] usb_hcd_nuc980_probe, id = 5, name: nuc980-ohci.5, 0
[    2.573658] usb_hcd_nuc980_probe, id = 6, name: nuc980-ohci.6, 0
[    2.581208] usbcore: registered new interface driver usb-storage
[    2.634166] NUC980 Crypto engine enabled.
[    2.657091] usbcore: registered new interface driver usbhid
[    2.664024] usbhid: USB HID core driver
[    2.669566] NET: Registered protocol family 17
[    2.736874] Freeing unused kernel memory: 2372K
[Mount JFFS2]: /dev/mtdblock0 --> /mnt/mtdblock0
[    3.926383] nuc980-emac0 nuc980-emac0: eth0 is OPENED
[    3.948041] nuc980-emac1 nuc980-emac1: eth1 is OPENED
[  184.851524] random: nonblocking pool is initialized             << it tooks 3 minutes before this line show up
# 
```
Is this normal for this system, or how could I solve this problem?

some basic info here:
```
~ # uname -a
Linux (none) 4.4.179 #8 PREEMPT Wed Oct 16 12:31:48 CST 2019 armv5tejl GNU/Linux
~ # df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/mtdblock0            2048       108      1940   5% /mnt/mtdblock0
~ # free
             total       used       free     shared    buffers     cached
Mem:         60468      12992      47476          0          0       7216
-/+ buffers/cache:       5776      54692
Swap:            0          0          0
```

thanks
JT

yachen, 2019-10-16:
We do not have 980uart2P  board. Do you mean 980eth2P?
3 min is way too long.  Please try nuc980_eth2uart_defconfig before e4b177f.
Storage changed from SPI NOR to SPI NAND from e4b177f.

ifq, 2019-10-16:
yes, it's 980eth2P.

ifq, 2019-10-16:
Can I specify kernel version with buildroot tool? or I have to checkout a separate kernel repo?

yachen, 2019-10-16:
Try 4abdfdb. The commit prior to e4b177f.

ifq, 2019-10-16:
I found buildroot downloaded the tar file, not git repo for kernel.  could you tell me what's the simplest way to switch kernel version in this setup?

ifq, 2019-10-17:
hi， I switched to 4abdfdb, it still cost 3 minutes.

```
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/OpenNuvoton/NUC980-linux-4.4.y.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="4abdfdb9417f4dcf5d3e659ef331d3ca49ac4b55"

```

To make sure I'm testing the right version, I confirmed one of the file, it matches commit 4abdfdb.
```
jt@jt-VirtualBox:~/src/NUC970_Buildroot-master/output/build/linux-4abdfdb9417f4dcf5d3e659ef331d3ca49ac4b55/arch/arm/configs$ head nuc980_eth2uart_defconfig 
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 4.4.115 Kernel Configuration
#

```

yachen, 2019-10-17:
Hi, How large is your rootfs? Are you using quad mode or 1-bit mode to access SPI flash?

ifq, 2019-10-17:
Hi, 
```
5.6M Oct 17 09:48 rootfs.tar

~ # df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/mtdblock0            2048       108      1940   5% /mnt/mtdblock0

```

I'm using quad mode.

yachen, 2019-10-17:
I tested with a 6.1MB rootfs. It only took few seconds to mount rootfs. Here's my kernel config:
[config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3738900/config.txt)

And below is console log:
> [2019-10-17 19:04:31.782] Linux version 4.4.179 (andy@YA-T470P) (gcc version 4.8.4 (GCC) ) #413 PREEMPT Thu Oct 17 19:03:22 CST 2019
> [2019-10-17 19:04:31.782] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
> [2019-10-17 19:04:31.782] CPU: VIVT data cache, VIVT instruction cache
> [2019-10-17 19:04:31.782] Machine: NUC980
> [2019-10-17 19:04:31.782] Warning: Neither atags nor dtb found
> [2019-10-17 19:04:31.782] Memory policy: Data cache writeback
> [2019-10-17 19:04:31.782] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
> [2019-10-17 19:04:31.782] Kernel command line: root=/dev/mtdblock0 ro rootfstype=jffs2 console=ttyS0,115200n8 mem=64M
> <snip...>
> [2019-10-17 19:04:36.171] Freeing unused kernel memory: 140K
> [2019-10-17 19:04:38.967] 
> [2019-10-17 19:04:38.967] 
> [2019-10-17 19:04:38.967] BusyBox v1.22.1 (2016-02-03 14:11:04 CST) built-in shell (ash)
> [2019-10-17 19:04:38.967] Enter 'help' for a list of built-in commands.
> [2019-10-17 19:04:38.967] 
> [2019-10-17 19:04:38.967] ~ #

closed 2020-03-09 by yachen