OpenNuvoton/NUC980-linux-4.4.y #12: sdio

mgkiller7, 2019-10-23:
Nuc980 Secure Digital Host Controller (SDH) connets to MicroChip's WILC1000-SD sdio wifi module as STATION mode, use the vendor driver https://github.com/linux4wilc/driver; 

wifi connect will fail after some Seconds as bellow:
64 bytes from 192.168.0.84: seq=1 ttl=64 time=12.175 ms
64 bytes from 192.168.0.84: seq=2 ttl=64 time=81.500 ms
64 bytes from 192.168.0.84: seq=3 ttl=64 time=60.933 ms

[15937.766197] SD time-out
[15940.766172] cfg_get Timed Out
[15940.769155] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1799] Get Timed out
[15940.778797] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters

The SDIO clk frequence is limited at nuc980_sd_probe() in drivers/mmc/host/nuc980_sd.c :
   mmc->f_max = 25000000; //for SDIO WIFI RTL8821AS
when I change SDIO clk from 25MHz to 20MHz,  wifi works well.

when i change SDIO clk to 12.5MHz or 50MHz, wifi can't works at these two frequences.
MicroChip say their WILC1000-SD module can work at the frequence upto 50MHz.

please help us to fix it?

yachen, 2019-10-24:
So there's no timeout error if SDIO clock is 20MHz. And SDIO WiFi cannot work at all if clock is 12.5MHz or 50MHz?

mgkiller7, 2019-10-24:
Yes. 
When sdio clk at 50MHz, the situation is the same as at 25MHz clk, also log:
[  483.429399] SD time-out
[  486.429399] cfg_get Timed Out
[  486.432383] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1799] Get Timed out
[  486.442041] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters

When at 12.5MHz, even fail to download the wifi firmware bin:
~ # wpa_supplicant -iwlan0 -Dnl80211 -c /nvm/wilc_wpa_supplicant.conf &
 Successfully initialized wpa_supplicant
[  147.705058] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.714658] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.725874] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.735491] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.746516] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.756091] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  147.767066] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
rfkill: Cannot o[  147.779058] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c39a2000] wlan0
pen RFKILL control device
[  147.789533] WILC POWER UP
[  147.792624] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[c39a2000][c39a3428]
[  147.802558] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  147.809741] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  147.817458] wilc_sdio mmc0:0001:1: SDIO speed: 12500000
[  147.825383] wilc_sdio mmc0:0001:1: chipid 001003a0
[  147.831391] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  147.841408] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  147.850233] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  147.861641] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  147.871374] wilc_sdio mmc0:0001:1: failed to get IRQ GPIO, load default
[  147.878191] wilc_sdio mmc0:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 347
[  147.888966] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  147.897666] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000000
[  147.905099] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[  147.915924] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[  147.930241] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  147.938824] wilc_wlan_firmware_downloadDownloading firmware size = 135060
[  147.947691] wilc_sdio mmc0:0001:1: wilc_sdio_cmd53..failed, err(-5)
[  147.954149] wilc_sdio mmc0:0001:1: Failed cmd53, read reg (00001400)...
[  147.967608] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  147.974166] wilc_sdio mmc0:0001:1: Failed cmd52, set 0x10c data...
[  147.987183] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  147.993741] wilc_sdio mmc0:0001:1: Failed cmd52, set 0x10c data...
[  148.006749] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  148.013316] wilc_sdio mmc0:0001:1: Failed cmd 52, read reg (000000fa) ...
[  148.026974] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  148.033541] wilc_sdio mmc0:0001:1: Failed cmd52, set 0x10c data...
[  148.046558] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  148.053133] wilc_sdio mmc0:0001:1: Failed cmd 52, read reg (000000fc) ...
[  148.061424] wilc_wlan_firmware_download Bus error
[  148.066274] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:938] Failed to download firmware
[  148.076908] wilc_sdio mmc0:0001:1: wilc_sdio_disable_interrupt
[  148.088358] wilc_sdio mmc0:0001:1: can't release sdio_irq, err(-110)
[  148.096391] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  148.105333] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  148.889641] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  148.902716] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:985] WLAN initialization FAILED
[  148.911908] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
Could not set interface wlan0 fl[  148.923708] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
ags (UP): Input/output error
nl80211: Could not[  148.937674] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
 set interface 'wlan0' UP
nl802[  148.948816] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
11: deinit ifnam[  148.959074] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
e=wlan0 disabled_11b_rates=0
wl[  148.971133] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
an0: Failed to i[  148.981149] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
nitialize driver interface
[  148.992974] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed

[1]+  Done(255)                  wpa_supplicant -iwlan0 -Dnl80211 -c /nvm/wilc_wpa_supplicant.conf

Anything needed please let me know.

schung1218, 2019-10-24:
Hi,
Please increase udelay time to test.
https://github.com/OpenNuvoton/NUC980-linux-4.4.y/blob/master/drivers/mmc/host/nuc980_sd.c
****

582 				/* To Avoid CMD53 reading data than less 64 bytes will be worng */ 
### 583 				udelay(10); 
584 				nuc980_sd_write(REG_DMACCSR, nuc980_sd_read(REG_DMACCSR) | DMACCSR_SW_RST);
****

mgkiller7, 2019-10-25:
Hi, schung1218
your patch do not work in WILC1000-SD 

AT 50MHz sd_clk:
~ # ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1): 56 data bytes
[  509.427633] SD time-out
[  512.427641] cfg_get Timed Out
[  512.430624] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1799] Get Timed out
[  512.440241] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[  709.427583] SD time-out
[  709.430074] cfg_indicate_rx: Receive unknown message 0-129-0-0-0-0-27-16
[  909.427641] SD time-out
[  909.514074] wilc_sdio mmc0:0001:1: wilc_sdio_cmd53..failed, err(-110)
[  909.520574] wilc_sdio mmc0:0001:1: Failed cmd53 [0], bytes read...
[  909.526699] wilc_wlan_handle_isr_ext: fail block rx

AT 25Hz sd_clk:
~ # ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1): 56 data bytes
64 bytes from 192.168.0.1: seq=1 ttl=64 time=9.233 ms
64 bytes from 192.168.0.1: seq=2 ttl=64 time=7.350 ms
64 bytes from 192.168.0.1: seq=3 ttl=64 time=6.783 ms
64 bytes from 192.168.0.1: seq=4 ttl=64 time=6.242 ms
64 bytes from 192.168.0.1: seq=5 ttl=64 time=5.708 ms
64 bytes from 192.168.0.1: seq=6 ttl=64 time=5.167 ms
64 bytes from 192.168.0.1: seq=7 ttl=64 time=14.592 ms
64 bytes from 192.168.0.1: seq=8 ttl=64 time=4.017 ms
64 bytes from 192.168.0.1: seq=9 ttl=64 time=13.500 ms
64 bytes from 192.168.0.1: seq=10 ttl=64 time=12.941 ms
64 bytes from 192.168.0.1: seq=11 ttl=64 time=12.392 ms
64 bytes from 192.168.0.1: seq=12 ttl=64 time=11.850 ms
64 bytes from 192.168.0.1: seq=13 ttl=64 time=11.275 ms
64 bytes from 192.168.0.1: seq=14 ttl=64 time=10.742 ms
64 bytes from 192.168.0.1: seq=15 ttl=64 time=10.108 ms
64 bytes from 192.168.0.1: seq=16 ttl=64 time=9.575 ms
64 bytes from 192.168.0.1: seq=17 ttl=64 time=9.000 ms
64 bytes from 192.168.0.1: seq=18 ttl=64 time=8.408 ms
64 bytes from 192.168.0.1: seq=19 ttl=64 time=7.867 ms
64 bytes from 192.168.0.1: seq=20 ttl=64 time=7.325 ms
64 bytes from 192.168.0.1: seq=21 ttl=64 time=16.792 ms
64 bytes from 192.168.0.1: seq=22 ttl=64 time=16.292 ms

[  589.339333] SD time-out
[  592.339366] cfg_get Timed Out
[  592.342349] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1799] Get Timed out
[  592.351966] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters

[  789.339341] SD time-out
[  789.341833] cfg_indicate_rx: Receive unknown message 255-255-255-255-255-255-255-255

mgkiller7, 2019-10-29:
In the top commit， i say  “when I change SDIO clk from 25MHz to 20MHz, wifi works well”  means i can ping the wifi AP last for 12+ hours without any dirver except lose 1% packets.

But do iperf test, the driver output lots of error:
~ # ping 192.168.0.100
PING 192.168.0.100 (192.168.0.100): 56 data bytes
64 bytes from 192.168.0.100: seq=0 ttl=128 time=21.900 ms
64 bytes from 192.168.0.100: seq=1 ttl=128 time=8.650 ms
64 bytes from 192.168.0.100: seq=2 ttl=128 time=8.075 ms
64 bytes from 192.168.0.100: seq=3 ttl=128 time=7.525 ms
64 bytes from 192.168.0.100: seq=4 ttl=128 time=6.942 ms
64 bytes from 192.168.0.100: seq=5 ttl=128 time=6.383 ms
64 bytes from 192.168.0.100: seq=6 ttl=128 time=5.917 ms
64 bytes from 192.168.0.100: seq=7 ttl=128 time=15.317 ms
64 bytes from 192.168.0.100: seq=8 ttl=128 time=4.725 ms
^C
--- 192.168.0.100 ping statistics ---
9 packets transmitted, 9 packets received, 0% packet loss
round-trip min/avg/max = 4.725/9.492/21.900 ms
~ #  
~ # 
~ # 
~ # /nvm/iperf3 -c 192.168.0.100 -p 5201
Connecting to host 192.168.0.100, port 5201
[  4] local 192.168.0.47 port 57644 connected to 192.168.0.100 port 5201
[ 1456.816974] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1456.828683] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec   182 KBytes  1.49 Mbits/sec    2   1.41 KBytes       
[  4]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[ 1463.289491] cfg_get Timed Out
[ 1463.292474] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1463.303841] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[ 1468.299649] cfg_get Timed Out
[ 1468.302633] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1468.329599] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[ 1473.319483] cfg_get Timed Out
[ 1473.322466] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1473.343541] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[ 1478.319491] cfg_get Timed Out
[ 1478.322474] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1478.342041] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[ 1483.339449] cfg_get Timed Out
[ 1483.342433] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1483.368533] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[ 1486.070208] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1486.791066] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1487.610674] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1488.030483] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1488.240199] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1488.349474] cfg_get Timed Out
[ 1488.352458] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1803] Get Timed out
[ 1488.389524] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1104] Failed to send scan parameters
[ 1490.109549] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.430266] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.460258] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.520008] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.551674] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.580249] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.610733] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.657824] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.700233] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.730258] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1490.940366] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1491.000149] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
[ 1491.048908] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000018
....

The same  WILC1000-SD driver work well on NXP's i.MX6 EVK board.
 (WILC1000-SD driver links:https://github.com/linux4wilc/driver)

So I really think there may be something issue on NUC980's sdio controller.

yachen, 2019-10-31:
We don't have WILC1000-SD so retest SDIO WiFi using RTL8821AS instead. The communication on SDIO  interface seems fine @50MHz. Are you testing on our NUC980 DEV board?

mgkiller7, 2019-11-01:
No, we have not a NUC980 DEV board, and just have a NK-NUC980 board.

closed 2021-07-30 by yachen