OpenNuvoton/NUC980-linux-4.4.y #13: USBD used as RNDIS, suddenly miss connect with PC

mgkiller7, 2019-10-24:
we use NUC980 USBD used as RNDIS，we correctly config kernel，and we can let the USBD work well as RNDIS connect to a PC（which also have RNDIS driver） after we run a script（named “config-usbd-as-rndis.sh”， see below）.
After this we can ping the Wifi AP's gateway successfully
~ # sh /nvm/usb0.sh 
[  127.568966] using random self ethernet address
[  127.573749] using random host ethernet address
[  127.733149] usb0: HOST MAC de:1e:e2:08:0a:e9
[  127.759783] usb0: MAC ee:bf:d8:93:38:a7
[  127.763658] enable: 0x80000620
[  127.766691] pullup
[  127.768649] enable: 0x80000720
 [  128.456291] configfs-gadget gadget: high-speed config #1: c
~ # 
~ # ping 192.168.100.2
PING 192.168.100.2 (192.168.100.2): 56 data bytes
64 bytes from 192.168.100.2: seq=0 ttl=128 time=1.850 ms
64 bytes from 192.168.100.2: seq=1 ttl=128 time=0.900 ms
64 bytes from 192.168.100.2: seq=2 ttl=128 time=0.975 ms
64 bytes from 192.168.100.2: seq=3 ttl=128 time=0.941 ms
64 bytes from 192.168.100.2: seq=4 ttl=128 time=0.975 ms

but after tens minutes or some hours, will fail to ping. but do not see any driver error at console.

after this the RNDIS on Windows PC side become a "unknown device",  even i plug-out and plug-in the USB line, the PC only can regnize the USBD as "unknown device".

at this time, i change to plug-in a Ubuntu PC, log out like this:
[  143.325771] usb 1-5: new high-speed USB device number 6 using xhci_hcd
[  148.578960] usb 1-5: device descriptor read/64, error -110
[  164.197036] usb 1-5: device descriptor read/64, error -110
[  164.425040] usb 1-5: new high-speed USB device number 7 using xhci_hcd
[  169.573172] usb 1-5: device descriptor read/64, error -110
[  185.189772] usb 1-5: device descriptor read/64, error -110
[  185.417820] usb 1-5: new high-speed USB device number 8 using xhci_hcd
[  190.445906] usb 1-5: device descriptor read/8, error -110
[  195.822026] usb 1-5: device descriptor read/8, error -110
[  196.050014] usb 1-5: new high-speed USB device number 9 using xhci_hcd
[  201.198127] usb 1-5: device descriptor read/8, error -110
[  206.574230] usb 1-5: device descriptor read/8, error -110
[  206.682203] usb usb1-port5: unable to enumerate USB device

yachen, 2019-10-28:
Do you have USB protocol analyser that can capture the USB traffic for us to check this issue?

mgkiller7, 2019-10-29:
Hi, yachen
we do RNDIS using NK-NUC980 V1.1 EVK Board.  And we have no USB protocol analyser. Can you do RNDIS test with the EVK Board.

below is my rndis script:
#!/bin/sh

cd /
mkdir cfg
mount none cfg -t configfs
mkdir cfg/usb_gadget/g1
cd cfg/usb_gadget/g1
mkdir configs/c.1
mkdir functions/rndis.usb0
mkdir strings/0x409
mkdir configs/c.1/strings/0x409
echo 0x9999 > idProduct
echo 0x9999 > idVendor
echo 123 > strings/0x409/serialnumber
echo manufacturer > strings/0x409/manufacturer
echo RNDIS Gadget > strings/0x409/product
echo "Conf 1" > configs/c.1/strings/0x409/configuration
echo 120 > configs/c.1/MaxPower

echo 1 > os_desc/use
echo 0xcd > os_desc/b_vendor_code
echo MSFT100 > os_desc/qw_sign

echo RNDIS > functions/rndis.usb0/os_desc/interface.rndis/compatible_id
echo 5162001 > functions/rndis.usb0/os_desc/interface.rndis/sub_compatible_id

ln -s configs/c.1 os_desc

mkdir functions/rndis.usb0/os_desc/interface.rndis/Icons
echo 2 > functions/rndis.usb0/os_desc/interface.rndis/Icons/type
echo "%SystemRoot%\\system32\\shell32.dll,-233" > functions/rndis.usb0/os_desc/interface.rndis/Icons/data

mkdir functions/rndis.usb0/os_desc/interface.rndis/Label
echo 1 > functions/rndis.usb0/os_desc/interface.rndis/Label/type
echo "XYZ Device" > functions/rndis.usb0/os_desc/interface.rndis/Label/data

ln -s functions/rndis.usb0 configs/c.1
echo b0016000.usbdev > UDC

if [ ! -n "$2" ] ;then
        ifconfig usb0 192.168.100.1
    ifconfig usb0 up
else
        ifconfig usb0 hw ether $2
    ifconfig usb0 192.168.100.1
        ifconfig usb0 up
fi

cd ~

yachen, 2019-10-31:
Hi, We've tested RNDIS and couldn't reproduce this problem. What Windows RNDIS driver are you using? And can you use bus hound free version to capture the USB traffic for us to analyse this issue? Thanks.

mgkiller7, 2019-10-31:
Hi, yechen
   这是在一次rndis异常时 抓到的
[image]
Bus Hound 6.01 capture on Windows Vista Service Pack 1 (x64). Complements of www.perisoft.net

rndis_error_log

  Device - Device ID (followed by the endpoint for USB devices)
            (11) 英特尔(R) USB 3.0 根集线器
            (29) Remote NDIS Compatible Device #5
            (30) Unknown Device
  Phase  - Phase Type
            CTL   USB control transfer       
            IN    Data in transfer           
            OUT   Data out transfer          
            USTS  USB status                 
  Data   - Hex dump of the data transferred
  Descr  - Description of the phase
  Cmd... - Position in the captured data

Device  Phase  Data                                                Description       Cmd.Phase.Ofs(rep)
------  -----  --------------------------------------------------  ----------------  ------------------
  11.0  CTL    23 03 1b 00  01 00 00 00                            CLASS                    1.1.0        
  11.0  CTL    23 03 1b 00  02 00 00 00                            CLASS                    2.1.0        
  11.0  CTL    23 03 1b 00  03 00 00 00                            CLASS                    3.1.0        
  11.0  CTL    23 03 1b 00  04 00 00 00                            CLASS                    4.1.0        
  11.0  CTL    23 03 1b 00  05 00 00 00                            CLASS                    5.1.0        
  11.0  CTL    23 03 1b 00  06 00 00 00                            CLASS                    6.1.0        
  11.0  CTL    23 03 1b 00  07 00 00 00                            CLASS                    7.1.0        
  11.0  CTL    23 03 1b 00  08 00 00 00                            CLASS                    8.1.0        
  11.0  CTL    23 03 1b 00  09 00 00 00                            CLASS                    9.1.0        
  11.0  CTL    23 03 1b 00  0a 00 00 00                            CLASS                   10.1.0        
  11.0  CTL    23 03 1b 00  0b 00 00 00                            CLASS                   11.1.0        
  11.0  CTL    23 03 1b 00  0c 00 00 00                            CLASS                   12.1.0        
  11.0  CTL    23 03 1b 00  0d 00 00 00                            CLASS                   13.1.0        
  11.0  USTS   80000300                                            bad parameter           13.2.0        
  11.0  CTL    23 03 1b 00  0e 00 00 00                            CLASS                   14.1.0        
  11.0  USTS   80000300                                            bad parameter           14.2.0        
  11.0  CTL    23 03 1b 00  0f 00 00 00                            CLASS                   15.1.0        
  11.0  USTS   80000300                                            bad parameter           15.2.0        
  11.0  CTL    23 03 1b 00  10 00 00 00                            CLASS                   16.1.0        
  11.0  CTL    23 03 1b 00  11 00 00 00                            CLASS                   17.1.0        
  11.0  CTL    23 03 1b 00  12 00 00 00                            CLASS                   18.1.0        
  11.0  CTL    23 03 1b 00  13 00 00 00                            CLASS                   19.1.0        
  11.0  CTL    23 03 1b 00  14 00 00 00                            CLASS                   20.1.0        
  11.0  CTL    23 03 1b 00  15 00 00 00                            CLASS                   21.1.0        
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                   22.1.0(2)     
  11.0  IN     07 25 00 00                                         .%..                    22.2.0        
  11.0  CTL    23 01 02 00  04 00 00 00                            CLASS                   24.1.0        
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                   25.1.0        
  11.0  IN     07 25 00 00                                         .%..                    25.2.0        
  11    IN     10 00 00                                            ...                     26.1.0        
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                   27.1.0        
  11.0  IN     03 05 04 00                                         ....                    27.2.0        
  11.0  CTL    23 01 12 00  04 00 00 00                            CLASS                   28.1.0        
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                   29.1.0(2)     
  11.0  IN     03 05 00 00                                         ....                    29.2.0        
  11.0  CTL    23 03 1b 00  01 00 00 00                            CLASS                   31.1.0        
  11.0  CTL    23 03 1b 00  02 00 00 00                            CLASS                   32.1.0        
  11.0  CTL    23 03 1b 00  03 00 00 00                            CLASS                   33.1.0        
  11.0  CTL    23 03 1b 00  04 00 00 00                            CLASS                   34.1.0        
  11.0  CTL    23 03 1b 00  05 00 00 00                            CLASS                   35.1.0        
  11.0  CTL    23 03 1b 00  06 00 00 00                            CLASS                   36.1.0        
  11.0  CTL    23 03 1b 00  07 00 00 00                            CLASS                   37.1.0        
  11.0  CTL    23 03 1b 00  08 00 00 00                            CLASS                   38.1.0        
[... cut ...]
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                  174.1.0        
  11.0  IN     03 05 00 00                                         ....                   174.2.0        
  11.0  CTL    23 03 02 00  04 00 00 00                            CLASS                  175.1.0        
  11.0  CTL    a3 00 00 00  04 00 04 00                            CLASS                  176.1.0        
  11.0  IN     07 25 00 00                                         .%..                   176.2.0

closed 2021-12-29 by ychuang3