OpenNuvoton/NUC980-linux-4.4.y #14: 发现I2C Read问题，并提供修改方案

zyj9527, 2019-10-31:
i2c驱动中iicstat == M_RECE_ADDR_ACK后，没有使能ACK，导致read第一个byte数据后，master没有发ACK，按照如下修改就OK了：

--- a/drivers/i2c/busses/i2c-nuc980-p0.c
+++ b/drivers/i2c/busses/i2c-nuc980-p0.c
@@ -362,7 +362,7 @@ static void i2c_nuc980_irq_master_TRx(struct nuc980_i2c *i2c, unsigned long iics
                        //return;
                }
                else
-                       writel(((readl(i2c->regs+CTL0) &~ (0x3C))|I2C_CTL_SI), (i2c->regs+CTL0));
+                       writel(((readl(i2c->regs+CTL0) &~ (0x3C))|I2C_CTL_SI_AA), (i2c->regs+CTL0));
        }
        else if ((iicstat == M_RECE_DATA_ACK) || (iicstat == M_RECE_DATA_NACK))
        { /* DATA has been transmitted and ACK has been received */

yachen, 2019-10-31:
Hi, Would you like to send a pull request, or do you prefer I modify our tree directly?

zyj9527, 2019-10-31:
Hi，you may modify your tree directly

yachen, 2019-10-31:
Done. Thanks a lot.

closed 2019-10-31 by yachen