OpenNuvoton/NUC980-linux-4.4.y #15: nuc980 I2C can't reading pcf8563 value using i2c-tools(i2cdetect and i2cdump)

xiebluelike, 2019-11-18:
we using NUC980 I2C-0 to reading rtc from pcf8563, but can't get normal value
(we try first using i2c-tools to get register value from pcf8653,  command just like that);
[image]
[image]
[image]

    It is normal when we using other plantform, just like this:
    
[image]
[image]
[image]

mjchen1, 2019-11-21:
[i2c-nuc980-p0.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3873533/i2c-nuc980-p0.zip)

I modify driver to fix some problem.
Could you update and test again...??

xiebluelike, 2019-11-21:
Thanks！ I have try the driver and it work fine now, can you update i2c-nuc980-p1 as well ?

mjchen1, 2019-11-25:
[i2c.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/3884487/i2c.zip)

I update i2c-0 ~ i2c-3 driver.
If any problem, please let me know.
thanks a lot.

xiebluelike, 2019-11-26:
Thanks!  I will take more test; there have another question：
can I used the slave-eeprom support in the kernel to test the slave support？？
I want to testing I2c slave mode at user space， and there have an article mentioned：
https://patchwork.ozlabs.org/patch/822238/
[image]

and I know NUC980 can support slave eeprom mode， just like as NUC980 BSP support doc said:
[image]

So i open slave support in kernel：
[image]
[image]

And I try using the cmd to add device:
echo slave-24c02 0x1064 > /sys/bus/i2c/devices/i2c-1/new_device

but when I using i2cdetect to view it， system dump！！
`# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          [  136.280858] INFO: rcu_preempt self-detected stall on CPU
[  136.286158]  0-...: (2100 ticks this GP) idle=3bf/140000000000002/0 softirq=12463/12463 fqs=0 
[  136.294699]   (t=2100 jiffies g=7601 c=7600 q=17)
[  136.299399] rcu_preempt kthread starved for 2100 jiffies! g7601 c7600 f0x0 s3 ->state=0x1
[  136.307533] Task dump for CPU 0:
[  136.310733] i2cdetect       D c04a3a54     0  2995   2976 0x00000002
[  136.317058] Backtrace: 
[  136.319566] [<c00131ac>] (dump_backtrace) from [<c00133a4>] (show_stack+0x18/0x1c)
[  136.327074]  r7:c0bde578 r6:80000093 r5:00000ba0 r4:c2a18740
[  136.332749] [<c001338c>] (show_stack) from [<c00410e4>] (sched_show_task+0xac/0x108)
[  136.340449] [<c0041038>] (sched_show_task) from [<c0041948>] (dump_cpu_task+0x24/0x30)
[  136.348299]  r5:c0bde578 r4:00000000
[  136.351891] [<c0041924>] (dump_cpu_task) from [<c0054868>] (rcu_dump_cpu_stacks+0xa8/0x10c)
[  136.360216] [<c00547c0>] (rcu_dump_cpu_stacks) from [<c0058524>] (rcu_check_callbacks+0x430/0x858)
[  136.369099]  r9:c0bc2094 r8:00001db1 r7:c0bde578 r6:00000000 r5:c0bde578 r4:c0bde9e4
[  136.376858] [<c00580f4>] (rcu_check_callbacks) from [<c005b4fc>] (update_process_times+0x40/0x6c)
[  136.385649]  r10:c0bdf500 r9:00000000 r8:c2926000 r7:0000001f r6:00000000 r5:c2a18740
[  136.393449]  r4:ffffe000
[  136.396024] [<c005b4bc>] (update_process_times) from [<c006a6f0>] (tick_sched_handle+0x50/0x54)
[  136.404658]  r7:0000001f r6:b92222f6 r5:c2927ba8 r4:c0bdfba0
[  136.410341] [<c006a6a0>] (tick_sched_handle) from [<c006a750>] (tick_sched_timer+0x5c/0xb8)
[  136.418674] [<c006a6f4>] (tick_sched_timer) from [<c005c240>] (__hrtimer_run_queues.constprop.6+0x124/0x1c4)
[  136.428424]  r7:c006a6f4 r6:c0bdf50c r5:c0bdfba0 r4:c0bdf4c0
[  136.434108] [<c005c11c>] (__hrtimer_run_queues.constprop.6) from [<c005c50c>] (hrtimer_interrupt+0xa8/0x248)
[  136.443849]  r10:c0bdf578 r9:c2927a98 r8:ffffffff r7:7fffffff r6:00000003 r5:c2926000
[  136.451649]  r4:c0bdf4c0
[  136.454224] [<c005c464>] (hrtimer_interrupt) from [<c001ba28>] (nuc980_timer4_interrupt+0x24/0x34)
[  136.463116]  r10:c0c0627c r9:c0bce338 r8:00000020 r7:00000000 r6:00000000 r5:c0bce348
[  136.470908]  r4:00000001
[  136.473483] [<c001ba04>] (nuc980_timer4_interrupt) from [<c004d86c>] (handle_irq_event_percpu+0x8c/0x160)
[  136.482974]  r5:c0bce348 r4:c0bc8900
[  136.486566] [<c004d7e0>] (handle_irq_event_percpu) from [<c004d9a4>] (handle_irq_event+0x64/0x90)
[  136.495366]  r10:c0bf10a8 r9:c0bdec80 r8:00000000 r7:00000000 r6:c2927c98 r5:c0bce348
[  136.503166]  r4:c0bce338
[  136.505724] [<c004d940>] (handle_irq_event) from [<c005085c>] (handle_level_irq+0xb4/0x178)
[  136.513999]  r5:c0bce348 r4:c0bce338
[  136.517591] [<c00507a8>] (handle_level_irq) from [<c004d1b4>] (__handle_domain_irq+0x6c/0xd0)
[  136.526049]  r5:c0bf10c8 r4:00000020
[  136.529633] [<c004d148>] (__handle_domain_irq) from [<c0009020>] (asm_do_IRQ+0x20/0x24)
[  136.537574]  r9:c0bdec80 r8:00000000 r7:c2927bdc r6:f0000000 r5:60000013 r4:c0021ee4
[  136.545333] [<c0009000>] (asm_do_IRQ) from [<c0013d70>] (__irq_svc+0x30/0x70)
[  136.552399] Exception stack(0xc2927ba8 to 0xc2927bf0)
[  136.557449] 7ba0:                   c0c06b40 c0c06b60 c0c06b40 60000013 00000000 00000202
[  136.565599] 7bc0: 00000000 c2926000 00000000 c0bdec80 c0bf10a8 c2927c44 c2927c48 c2927bf8
[  136.573733] 7be0: c00223a0 c0021ee4 60000013 ffffffff
[  136.578774] [<c0021e60>] (__do_softirq) from [<c00223a0>] (irq_exit+0xd8/0x120)
[  136.586024]  r10:c0bf10a8 r9:c0bdec80 r8:00000000 r7:00000000 r6:00000000 r5:c0bf10c8
[  136.593816]  r4:00000000
[  136.596374] [<c00222c8>] (irq_exit) from [<c004d1bc>] (__handle_domain_irq+0x74/0xd0)
[  136.604133]  r5:c0bf10c8 r4:00000000
[  136.607716] [<c004d148>] (__handle_domain_irq) from [<c0009020>] (asm_do_IRQ+0x20/0x24)
[... 5 lines cut ...]
[  136.651833] 7cc0: c0bf10a8 c2927cfc c2927d00 c2927ce8 c04a6d98 c04a3db0 60000013 ffffffff
[  136.660008] [<c04a3d38>] (schedule) from [<c04a6d98>] (schedule_timeout+0x124/0x1cc)
[  136.667683]  r5:c2927d04 r4:ffffb9c9
[  136.671266] [<c04a6c74>] (schedule_timeout) from [<c02e6860>] (nuc980_i2c1_xfer+0x12c/0x350)
[  136.679641]  r9:c280dc30 r8:000001f4 r7:00000002 r6:00000000 r5:c2926000 r4:c280dc00
[  136.687408] [<c02e6734>] (nuc980_i2c1_xfer) from [<c02e2d08>] (__i2c_transfer+0x134/0x278)
[  136.695599]  r10:00000000 r9:c0c38694 r8:ffffb7d5 r7:c0bdec60 r6:c2927e10 r5:00000001
[  136.703399]  r4:c280dc30
[  136.705958] [<c02e2bd4>] (__i2c_transfer) from [<c02e2ebc>] (i2c_transfer+0x70/0xa4)
[  136.713624]  r10:00000000 r9:00000003 r8:00000000 r7:00000001 r6:c2927e10 r5:c280dc40
[  136.721424]  r4:c280dc30 r3:00000000
[  136.725016] [<c02e2e4c>] (i2c_transfer) from [<c02e3194>] (i2c_smbus_xfer+0x1f8/0x5dc)
[  136.732866]  r7:00000000 r6:c29ed3c0 r5:c2927e2b r4:c280dc30
[... cut ...]
[  136.768958] [<c02e50b0>] (i2cdev_ioctl) from [<c00cbc24>] (do_vfs_ioctl+0x47c/0x660)
[  136.776633]  r5:c28dd540 r4:bed02cf0
[  136.780224] [<c00cb7a8>] (do_vfs_ioctl) from [<c00cbe44>] (SyS_ioctl+0x3c/0x64)
[  136.787466]  r10:00000000 r9:c2926000 r8:bed02cf0 r7:00000720 r6:c29ed3c0 r5:00000003
[  136.795266]  r4:c29ed3c0
[  136.797824] [<c00cbe08>] (SyS_ioctl) from [<c000ff60>] (ret_fast_syscall+0x0/0x3c)
[  136.805324]  r9:c2926000 r8:c0010148 r7:00000036 r6:00000003 r5:00000003 r4:00000003`

mjchen1, 2019-11-26:
I think, i2c-tool only support master mode.

xiebluelike, 2019-11-26:
Could you show me how to testing I2c slave mode at user space？
And can I test it with another master device？

mjchen1, 2019-11-26:
You can reference ...../linux-4.4.y/Documentation/i2c/slave-interface.
[image]

xiebluelike, 2019-11-26:
I have try this before, Connect master device's I2C1 and slave device's I2C1 SCL and SDA lines, using   i2c-tool to write/read, just like that:
In slave device:
echo slave-24c02 0x1064 > /sys/bus/i2c/devices/i2c-1/new_device

In master device:
i2cset -f -y 1 0x64 1 0x3
and then:
i2cget -f -y 1 0x64 1

But can't get true value!

mjchen1, 2019-11-26:
your master i2c need contain device(ex:slave-24c02) driver.

closed 2020-03-09 by yachen