OpenNuvoton/NUC980-linux-4.4.y #16: Is there any way to set some pin to specified status at linux boot?

tonyho, 2019-11-19:
I have designed core board using the NUC980DK61YC SoC, I use the IP101 as the PHY IC, and connect it to the PF port (EMAC1 port), except the RMII pins, I also use the PB3 as the nRESET pin to connect with Pin1(Reset) of IP101GR. 

And now, I have a question: How can I output the PB3 as high after boot or reset the PHY in kernel code?

yachen, 2019-11-19:
You can use gpio_direction_output() to enable GPIO output and gpio_set_value() to set the output level.
Please refer to drivers/media/i2c/nuvoton/nuv980_cap[01].c, these drivers control GPIO to reset image sensor.

tonyho, 2019-11-20:
> You can use gpio_direction_output() to enable GPIO output and gpio_set_value() to set the output level.
> Please refer to drivers/media/i2c/nuvoton/nuv980_cap[01].c, these drivers control GPIO to reset image sensor.

This is exactly what I want, thanks.

closed 2019-11-20 by tonyho