OpenNuvoton/NUC980-linux-4.4.y #18: in nuc980-spi1-slave.c  spi slave mode error

Ryaner1994, 2020-01-02:
在spi slave 中模式设置错误，如下：
if ((spimode == SPI_MODE_0) || (spimode == SPI_MODE_2)) {
		hw->pdata->txneg = 1;
		hw->pdata->rxneg = 0;
	} else {
		hw->pdata->txneg = 0;
		hw->pdata->rxneg = 1;
	}
应为：
if ((spimode == SPI_MODE_0) || (spimode == SPI_MODE_3)) {
                hw->pdata->txneg = 1;
		hw->pdata->rxneg = 0;
	} else {
		hw->pdata->txneg = 0;
		hw->pdata->rxneg = 1;
	}
另外，
	hw->bitbang.setup_transfer = nuc980_spi1_setupxfer;
	hw->bitbang.chipselect     = nuc980_spi1_chipsel;
	hw->bitbang.master->setup  = nuc980_spi1_setup;
此三个函数未加载，导致用户无法根据自己需求来设置spi slave 模式！

cwweng-nuvoton, 2020-01-03:
Hi Ryaner,

SPI slave mode 模式設置的確錯誤，我們會修正。
但是不清楚你說的 "此三个函数未加载" 是什麼意思 ?
這三個函數我們都有註冊，定義。

感謝

Ryaner1994, 2020-01-03:
在“nuc980_spi1_slave_probe”函数中：
有定义：“master->mode_bits          = (SPI_MODE_0 | SPI_CS_HIGH | SPI_LSB_FIRST | SPI_CPHA | SPI_CPOL);”
在SPI初始化完成后，读取SPI1_CTL寄存器的值为“0x00040805”
此时根据datasheet，为MSB，CLKPOL = 0，与上述值不符。而此寄存器的值恰好为“arch/arm/mach-nuc980/dev.c/”中“ nuc980_spi1_slave_platform_data”中的初始值。
之后我利用用户层面程序通过打开设备“spidev0.1"来修改SPI模式，仍然无法修改。

cwweng-nuvoton, 2020-01-06:
SPI1 slave 初始化時，的確是根據 arch/arm/mach-nuc980/dev.c 中的 nuc980_spi1_slave_platform_data 來設定相關寄存器。
如果要修改 SPI 模式，就必須修改 nuc980_spi1_slave_platform_data 中的設定值。

closed 2020-03-09 by yachen