OpenNuvoton/NUC980-linux-4.4.y #19: "RFC: remove Nuvoton w90x900/nuc900 platform" on kernel master

remuscl, 2020-01-05:
Hi :-)

What is going on here? Does that mean NUC980's bsp support will be dropped soon either?

https://lkml.org/lkml/2019/7/30/480

```
As the mach-netx and mach-8695 platforms are being removed now,
I wonder whether we should do the same with w90x00: Here is what
I found after looking at the git history and external material for it.

    - The supported chips (nuc910/950/960) are no longer marketed
      by the manufacturer

    - Newer chips from the same family (nuc97x, nuc980, n329x)
      that are still marketed have Linux BSPs but those were never
      submitted for upstream inclusion.

    - Wan ZongShun is listed as maintainer, but the last patch he wrote
      was in 2011.

    - All patches to w90x900 platform specific files afterwards
      are cleanups that were apparently done without access to
      test hardware.

    - The http://www.mcuos.com/ website listed in the MAINTAINERS
       file is no longer reachable.

We do support the newer NPCM platform from Nuvoton. I don't think
there are any shared drivers between the two, but I've added its
maintainers to Cc anyway, in case they still (plan to) use one of
those drivers.

If we decide that it's time to let go, I'll would the patches below.

      watchdog: remove w90x900 driver
      spi: remove w90x900 driver
      ASoC: remove w90x900/nuc900 platform drivers
      fbdev: remove w90x900/nuc900 platform drivers
      Input: remove w90x900 keyboard driver
      Input: remove w90x900 touchscreen driver
      mtd: rawnand: remove w90x900 driver
      net: remove w90p910-ether driver
      rtc: remove w90x900/nuc900 driver
      usb: remove ehci-w90x900 driver
      ARM: remove w90x900 platform
```

yachen, 2020-01-06:
No. We'll keep support 980 on 4.4

remuscl, 2020-01-06:
That is awesome, thanks a lot for your contribution!

But the LTS 4.4 should be EOL after Feb 2022 according to [kernel.org](https://www.kernel.org/category/releases.html). So does it means the BSP support for NUC980 would end in about two years?

yachen, 2020-01-06:
No, we'll continue support the NUC980 BSP.

closed 2020-03-09 by yachen