OpenNuvoton/NUC980-linux-4.4.y #21: NAND Flash not found during boot on NUC980ETH2UART board

NUC980ETH2UART, 2020-01-23:
I am using nuc980 bsp, and writing image from linux NuWriter. I am using nuc980_eth2uart_defconfigs to build rootfs, u-boot, and the kernel.
Chip is NUC980DK61YC
My .ini file for NuWriter

`image_num = 4
image0_path= /home/notpi/WORKING_DIR/nuc980bsp/ZZZZZZ/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x000200
image0_start_offset = 0x00000

image1_type = data
image1_path= /home/notpi/WORKING_DIR/nuc980bsp/ZZZZZZ/u-boot.bin
image1_exe_addr = 0x0000
image1_start_offset = 0x100000

image2_type = data
image2_path= /home/notpi/WORKING_DIR/nuc980bsp/ZZZZZZ/980uimage
image2_exe_addr = 0x0000
image2_start_offset = 0x200000

image3_type = env
image3_path= /home/notpi/WORKING_DIR/nuc980bsp/ZZZZZZ/env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000`

My env.txt

`baudrate=115200
bootcmd=run setspi;run loadkernel;bootm 0x7fc0
bootdelay=1
loadkernel=sf read 0x7fc0 0x200000 0x300000
setspi=sf probe 0 40000000
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel`

NUC980's log after writing image

`NUC980 IBR 20180813
Boot from USB
DDR-OK
=======================================
Run firmware code
CPU: 300MHz, DDR: 300MHz, SYS: 300MHz, PCLK: 150MHz
Parse NuWriter command line
=======================================
SPI NOR ID=0x0000ffff  _spi_type =0
SPI NOR ID=0x0000ffff  _spi_type =0

SPI NAND: Auto Detect
ID=[0xefaa21]
BlockPerFlash = 1024, PagePerBlock = 64
PageSize = 2048, SpareArea = 64
QuadReadCmd    = 0x6b
ReadStatusCmd  = 0xff
WriteStatusCmd = 0xff
StatusValue    = 0xff
Dummybyte      = 0x1

NAND ID=[f7][f7][f7][f7]
PowerOn setting 0x7f6001d4
BlockPerFlash=1023, PagePerBlock=64, PageSize=2048

eMMC1/SD1 eMMCBlockSize=0xffff0115(-65259)

Finish get INFO!!
`

U-boot log file

`NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow▒c{
                 #S
 SPL load main U-Boot from SPI NAND Flash! (Jan 23 2020 13:48:45)
initcall: 00e389d8

U-Boot 2016.11-g60b7bee-dirty (Jan 23 2020 - 13:48:26 +0300)

initcall: 00e0c730
U-Boot code: 00E00000 -> 00E50AA4  BSS: -> 00E5D1EC
initcall: 00e00364
CPU: NUC980
initcall: 00e0ccc0
Board: NUC980
initcall: 00e0c75c
DRAM:  initcall: 00e00d84
initcall: 00e0ca50
Monitor len: 0005D1EC
Ram size: 04000000
Ram top: 04000000
initcall: 00e0c640
initcall: 00e0c658
initcall: 00e0c6e4
Reserving 372k for U-Boot at: 03fa2000
initcall: 00e0c6b8
Reserving 1088k for malloc() at: 03e92000
initcall: 00e0c8ec
Reserving 88 Bytes for Board Info at: 03e91fa8
initcall: 00e0c950
initcall: 00e0c684
Reserving 160 Bytes for Global Data at: 03e91f08
initcall: 00e0c7b4
initcall: 00e0c940
initcall: 00e0cacc
initcall: 00e0ca30
initcall: 00e0c974

RAM Configuration:
Bank #0: 0 64 MiB
Bank #1: 0 0 Bytes

DRAM:  64 MiB
initcall: 00e0c668
New Stack Pointer is: 03e91ee0
initcall: 00e0c8a8
initcall: 00e0c83c
Relocation Offset is: 031a2000
Relocating to 03fa2000, new gd at 03e91f08, sp at 03e91ee0
initcall: 03faeb18
initcall: 03faeb20
initcall: 00e0cc70 (relocated to 03faec70)
initcall: 00e0cb38 (relocated to 03faeb38)
initcall: 00e0cc88 (relocated to 03faec88)
initcall: 00e0cc54 (relocated to 03faec54)
using memory 0x3e92000-0x3fa2000 for malloc()
initcall: 00e0cc90 (relocated to 03faec90)
initcall: 00e0cb10 (relocated to 03faeb10)
initcall: 00e0cc40 (relocated to 03faec40)
initcall: 00e00e0c (relocated to 03fa2e0c)
initcall: 00e11218 (relocated to 03fb3218)
initcall: 00e0cc30 (relocated to 03faec30)
initcall: 00e0cc14 (relocated to 03faec14)
Now running in RAM - U-Boot at: 03fa2000
initcall: 00e0cc80 (relocated to 03faec80)
initcall: 00e0cbf8 (relocated to 03faebf8)
NAND:  NAND Flash not found !
NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
0 MiB
initcall: 00e0cbc8 (relocated to 03faebc8)
SF: Got idcodes
00000000: ff ef aa 21 00                                     ...!.
SF: Unsupported flash IDs: manuf ff, jedec efaa, ext_jedec 2100
*** Warning - spi_flash_probe() failed, using default environment

Destroy Hash Table: 03fee1d0 table = 00000000
Create Hash Table: N=67
INSERT: table 03fee1d0, filled 1/67 rv 03ea29f8 ==> name="bootdelay" value="3"
INSERT: table 03fee1d0, filled 2/67 rv 03ea2a0c ==> name="baudrate" value="115200"
INSERT: free(data = 03e92618)
INSERT: done
initcall: 00e0cc9c (relocated to 03faec9c)
initcall: 00e11234 (relocated to 03fb3234)
initcall: 00e0cbb8 (relocated to 03faebb8)
initcall: 00e0f094 (relocated to 03fb1094)
In:    serial
Out:   serial
Err:   serial
Initial value for argc=3
Final value for argc=3
Initial value for argc=3
Final value for argc=3
Initial value for argc=3
Final value for argc=3
initcall: 00e009cc (relocated to 03fa29cc)
initcall: 00e0cba8 (relocated to 03faeba8)
initcall: 00e0cb88 (relocated to 03faeb88)
initcall: 00e00e3c (relocated to 03fa2e3c)
initcall: 00e0cb6c (relocated to 03faeb6c)
Net:   Net Initialization Skipped
No ethernet found.
initcall: 00e0cb60 (relocated to 03faeb60)
### main_loop entered: bootdelay=3

### main_loop: bootcmd="<UNDEFINED>"
`
There are no reported mistakes in any steps beside booting.
Mistake might be by my side, and might be something basic. I've been trying to run this board for a week now.

yachen, 2020-01-24:
Hi, Are you using latest U-Boot source code? Can't tell what goes wrong from the log you provided. We're on Chinese New Year vacation. Will check this issue after office open after vacation's over.

NUC980ETH2UART, 2020-01-24:
I am using 
NUC970_U-Boot_v2016.11 which comes with nuc980 package. If there is one for nuc980 besides this, i couldn't find it. And enjoy your vacation.

NUC980ETH2UART, 2020-01-29:
Progress. I used nuc980_iot_defconfig for u-boot, and it sees kernel image, and loads it. But, i can't get to the console yet.

Here is boot log,

> NUC980 IBR 20180813
> Boot from SPI-NAND
> DDR-OK
> finish SPI dow▒load
> 
>  SPL load main U-Boot from SPI NAND Flash! (Jan 28 2020 16:39:00)
> initcall: 00e3a354
> 
> 
> U-Boot 2016.11-g60b7bee-dirty (Jan 28 2020 - 16:38:42 +0300)
> 
> initcall: 00e0c7f8
> U-Boot code: 00E00000 -> 00E52614  BSS: -> 00E5F268
> initcall: 00e0036c
> CPU: NUC980
> initcall: 00e0cd8c
> Board: NUC980
> initcall: 00e0c824
> DRAM:  initcall: 00e00df0
> initcall: 00e0cb18
> Monitor len: 0005F268
> Ram size: 04000000
> Ram top: 04000000
> initcall: 00e0c6b0
> initcall: 00e0c7a0
> TLB table from 03ff0000 to 03ff4000
> initcall: 00e0c6c8
> initcall: 00e0c754
> Reserving 380k for U-Boot at: 03f90000
> initcall: 00e0c728
> Reserving 1152k for malloc() at: 03e70000
> initcall: 00e0c9b4
> Reserving 88 Bytes for Board Info at: 03e6ffa8
> initcall: 00e0ca18
> initcall: 00e0c6f4
> Reserving 168 Bytes for Global Data at: 03e6ff00
> initcall: 00e0c87c
> initcall: 00e0ca08
> initcall: 00e0cb94
> initcall: 00e0caf8
> initcall: 00e0ca3c
> 
> RAM Configuration:
> Bank #0: 0 64 MiB
> Bank # 1: 0 0 Bytes
> 
> DRAM:  64 MiB
> initcall: 00e0c6d8
> New Stack Pointer is: 03e6fee0
> initcall: 00e0c970
> initcall: 00e0c904
> Relocation Offset is: 03190000
> Relocating to 03f90000, new gd at 03e6ff00, sp at 03e6fee0
> initcall: 03f9cbe0
> initcall: 03f9cbe8
> initcall: 00e0cd3c (relocated to 03f9cd3c)
> initcall: 00e0cc00 (relocated to 03f9cc00)
> initcall: 00e0cd54 (relocated to 03f9cd54)
> initcall: 00e0cd20 (relocated to 03f9cd20)
> using memory 0x3e70000-0x3f90000 for malloc()
> initcall: 00e0cd5c (relocated to 03f9cd5c)
> initcall: 00e0cbd8 (relocated to 03f9cbd8)
> initcall: 00e0cd0c (relocated to 03f9cd0c)
> initcall: 00e00e6c (relocated to 03f90e6c)
> initcall: 00e11368 (relocated to 03fa1368)
> initcall: 00e0ccfc (relocated to 03f9ccfc)
[... 164 lines cut ...]
> usb_hcd_nuc980_probe, id = 0, name: nuc980-ohci.0, 0
> nuc980-ohci nuc980-ohci.0: Nuvoton NUC980 OHCI Host Controller
> nuc980-ohci nuc980-ohci.0: new USB bus registered, assigned bus number 2
> nuc980-ohci nuc980-ohci.0: irq 24, io mem 0xb0017000
> hub 2-0:1.0: USB hub found
> hub 2-0:1.0: 8 ports detected
> usb_hcd_nuc980_probe, id = 1, name: nuc980-ohci.1, 0
> usb_hcd_nuc980_probe, id = 2, name: nuc980-ohci.2, 0
> usb_hcd_nuc980_probe, id = 3, name: nuc980-ohci.3, 0
> usb_hcd_nuc980_probe, id = 4, name: nuc980-ohci.4, 0
> usb_hcd_nuc980_probe, id = 5, name: nuc980-ohci.5, 0
> usb_hcd_nuc980_probe, id = 6, name: nuc980-ohci.6, 0
> usbcore: registered new interface driver usb-storage
> usbcore: registered new interface driver usbhid
> usbhid: USB HID core driver
> NET: Registered protocol family 17
> Freeing unused kernel memory: 2356K
> nuc980-emac0 nuc980-emac0: eth0 is OPENED
> nuc980-emac1 nuc980-emac1: eth1 is OPENED
> random: nonblocking pool is initialized

Can you give me a working 980uimage, env.txt, u-boot.bin and u-boot-spl.bin for NUC980ETH2UART card so i can test if its actually working, or im doing something wrong.

And i don't know if it does matter but, product's box says NUC980DK61Y,
on the chip it says NUC980DK61YC. I checked ini files, they are same. I tried both anyway.

NUC980ETH2UART, 2020-01-31:
I found something. nuc980_eth2uart_defconfig u-boot includes below settings which causes error for me;

> SF: Got idcodes
> 00000000: ff ef aa 21 00                                     ...!.
> SF: Unsupported flash IDs: manuf ff, jedec efaa, ext_jedec 2100
> SF: Lock ops not supported for ff flash
> SF: Detected <NULL> with page size 256 Bytes, erase size 0 Bytes, total 0 Bytes
> *** Warning - bad CRC, using default environment

-DeviceDriver->SPIFlashSupport->Winbond SPI flash support   ----> So i disable this option,
And i enable below option
-DeviceDriver->SPIFlashSupport->SPI NAND flash support->Select SPI NAND Flash (Winbond SPI NAND flash support)

This makes chip detected, linux loadable, and logs out this;
> SF: Got idcodes
> 00000000: ff ef aa 21 00                                     ...!.
> SF: Lock ops not supported for ef flash
> SF: Detected W25N01GV with page size 2 KiB, erase size 4 KiB, total 128 MiB
> initcall: 00e0cbc8 (relocated to 03facbc8)
> SF: Got idcodes
> 00000000: 00 ef aa 21 00                                     ...!.
> SF: Lock ops not supported for ef flash
> SF: Detected W25N01GV with page size 2 KiB, erase size 4 KiB, total 128 MiB
> *** Warning - bad CRC, using default environment

Is this supposed to happen? I discovered this while comparing iot and eth2p configs.
And boot takes around 5 minutes.

yachen, 2020-02-03:
Hi,

Here's my step to execute image on ETH2UART board.
1. Load default ETH2UART U-Boot config.
2. make menuconfig->Device Drivers->SPI Flahs Support
2.1 Uncheck Winbond SPI flahs support
2.2 Check SPI NAND flash support.
2.3 Build U-Boot
3. Load default ETH2UART kernel config and build kernel.

We used to have SPI NOR on this board but replaced it with SPI NAND. I'll update the U-Boot config file for SPI NAND. Thanks.

Sincerely,

Yi-An Chen

yachen, 2020-02-03:
By the way, please use nand commad to read data from SPI NAND instead of spi command.
nand read 0x7FC0 0x200000 <image length>
bootm 0x7fC0

closed 2020-02-03 by NUC980ETH2UART