OpenNuvoton/NUC980-linux-4.4.y #22: UART9 can not select the PB2 as RTS Pin

tonyho, 2020-04-19:
the  Used IC is the NUC980DK61YC. From the Tech reference manual(EN_R1.0.0 pdf),  the PB2 can be used as the UART9 RTS, but there's no option in `menuconfig` for this:
[image]

In DS:
[image]

Is that possible to select the  "PB2 --> UART9_RTS,  PB1--> UART9_TXD, PB3--> UART9_RXD"  as UART9 Pins, as we have designed and Product the boards using this pinMap?

yachen, 2020-04-20:
Hi,
Modify line 257 in pinctrl-nuc980.c from
`static const unsigned uart9_0_pins[] = {0x11, 0x13};`
to
`static const unsigned uart9_0_pins[] = {0x11, 0x12, 0x13};`
Then you'll be able to control PB2 as RTS pin.

tonyho, 2020-04-21:
> 
> 
> Hi,
> Modify line 257 in pinctrl-nuc980.c from
> `static const unsigned uart9_0_pins[] = {0x11, 0x13};`
> to
> `static const unsigned uart9_0_pins[] = {0x11, 0x12, 0x13};`
> Then you'll be able to control PB2 as RTS pin.

Thanks for your response and this do solve my problem !
Another related question: 
 This will not influence the PE group as EMAC1 function, right? Since the UART9 CTS would use the PE pin, but this setting `static const unsigned uart9_0_pins[] = {0x11, 0x12, 0x13};`` wouldn't change the PE pins, so I can use the PE pins as EMAC1 without problem?

yachen, 2020-04-21:
EMAC0 uses PE while EMAC1 uses PF. You can still use EMAC0 without problem becasue uart9_0_pins[] doesn't contain PE pins.

tonyho, 2020-04-21:
> 
> 
> EMAC0 uses PE while EMAC1 uses PF. You can still use EMAC0 without problem becasue uart9_0_pins[] doesn't contain PE pins.

Thanks for your quick reply, I get it.

closed 2020-04-21 by tonyho