OpenNuvoton/NUC980-linux-4.4.y #23: mmc not working: nuc980-fmi: no support for card's volts

tonyho, 2020-05-31:
I have design a board refering the NUC980 reference board shematic, but the TF/MMC is not working.
My schematic is:
[image]

When I insert the mmc/TF, the log is:
```
mmc0: error -110 whilst initialising SD card
```

Then I added a 5.1KΩ Resistor to pull up the `CLK` , then the log is:
```
[    7.220000] mmc0: error -95 whilst initialising SD card
[    7.230000] nuc980-fmi nuc980-fmi: no support for card's volts
[    7.230000] mmc0: error -22 whilst initialising MMC card
[    7.280000] nuc980-fmi nuc980-fmi: no support for card's volts
[    7.300000] mmc0: error -22 whilst initialising SDIO card
[    7.300000] nuc980-fmi nuc980-fmi: no support for card's volts
[    7.310000] mmc0: error -22 whilst initialising MMC card
```

I tried 2 kind card:
- Panasonic MicroSD HC I 4GB
- XiaKe  MicroSD HC I  32GB

Both of them are not working with above errors.

What is the problem, how to fix it?

yachen, 2020-06-01:
Hi,

Try remove 100R and short circuit.
No need to pull up CLK.

Sincerely,

Yi-An Chen

tonyho, 2020-06-05:
> 
> 
> Hi,
> 
> Try remove 100R and short circuit.
> No need to pull up CLK.
> 
> Sincerely,
> 
> Yi-An Chen

Thanks for the quickly reply. I tried to remove the 100R, and short circuit. And add some extra log information :
```
[   81.050000] mmc0: card aaaa removed

[   83.510000] mmc0: SD Status: Invalid Allocation Unit size
[   83.540000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   83.560000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   83.590000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   83.600000] mmc0: new SDHC card at address aaaa
[   83.600000] mmcblk0: mmc0:aaaa SU08G 7.40 GiB 
```

The kernel not detect the mmcblk partition with `mmc0: SD Status: Invalid Allocation Unit size`, but it does detect the size of SD card correctly, is any further suggestion?  
Should I remove the pull up  5.1K resistor of the CLK line?

yachen, 2020-06-05:
Yes, please test again with the 5.1k removed. Thanks.

tonyho, 2020-06-05:
> 
> 
> Yes, please test again with the 5.1k removed. Thanks.

I removed the 5.1K but stille not working:
```
# [   21.680000] mmc0: card aaaa removed
[   23.890000] mmc0: SD Status: Invalid Allocation Unit size
[   23.920000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   23.940000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   23.970000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   23.980000] mmc0: new SDHC card at address aaaa
[   23.980000] mmcblk0: mmc0:aaaa SU08G 7.40 GiB 

# 
# [   39.840000] mmc0: card aaaa removed
[   56.770000] mmc0: problem reading SD Status register
[   56.800000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   56.830000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   56.850000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   56.860000] mmc0: new SDHC card at address aaaa
[   56.870000] mmcblk0: mmc0:aaaa SC16G 14.8 GiB 

# 
# [   77.370000] mmc0: card aaaa removed
[   85.190000] mmc0: SD Status: Invalid Allocation Unit size
[   85.220000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   85.240000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   85.270000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   85.280000] mmc0: new SDHC card at address 26da
[   85.280000] mmcblk0: mmc0:26da B04GK 3.69 GiB 

```

yachen, 2020-06-08:
Does the SD PC pins share with other function on your board?
Can you replace C21 with a 10uF capacitor and test again?

tonyho, 2020-06-08:
> 
> 
> Does the SD PC pins share with other function on your board?
> Can you replace C21 with a 10uF capacitor and test again?

2. Can you replace C21 with a 10uF capacitor and test again?
--> I tried, same result. Except adding a 10uF, I also tried to add a 220uF capacitor, but still the same result.
```
[   23.960000] mmc0: SD Status: Invalid Allocation Unit size
[   23.990000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   24.010000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   24.040000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   24.050000] mmc0: new SDHC card at address 0ad9
[   24.050000] mmcblk0: mmc0:0ad9 B04GK 3.69 GiB 

[   27.950000] mmc0: card 0ad9 removed

[   34.150000] mmc0: SD Status: Invalid Allocation Unit size
[   34.180000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   34.200000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   34.230000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   34.240000] mmc0: new SDHC card at address aaaa
[   34.240000] mmcblk0: mmc0:aaaa SU08G 7.40 GiB
```

1. Does the SD PC pins share with other function on your board?
--> You mean the CD pin?  No.
```
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: NUC980
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat c094615c, node_mem_map c3f77000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: mem=64M console=ttyS0,115200 root=/dev/sda5 rootfstype=ext4 rootwait ignore_loglevel debug printk.time=1
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 55160K/65536K available (6818K kernel code, 354K rwdata, 2100K rodata, 192K init, 250K bss, 10376K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc08bdf64   (8920 kB)
[    0.000000]       .init : 0xc08be000 - 0xc08ee000   ( 192 kB)
[    0.000000]       .data : 0xc08ee000 - 0xc0946a20   ( 355 kB)
[    0.000000]        .bss : 0xc0946a20 - 0xc09854ac   ( 251 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:545
[    0.000000] clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
[    0.000000] Console: colour dummy device 80x30
[    0.160000] console [ttyS0] enabled
[    0.170000] Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
[    0.240000] pid_max: default: 32768 minimum: 301
[    0.240000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.250000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.260000] CPU: Testing write buffer coherency: ok
[    0.260000] Setting up static identity map for 0x8400 - 0x843c
[... 266 lines cut ...]
OK
Starting NFS services: OK
Starting NFS daemon: rpc.nfsd: Unable to access /proc/fs/nfsd errno 2 (No such file or directory).
Please try, as root, 'mount -t nfsd nfsd /proc/fs/nfsd' and then restart rpc.nfsd to correct the problem
FAIL
Starting NFS mountd: rpc.mountd: svc_tli_create: could not open connection for udp6
rpc.mountd: svc_tli_create: could not open connection for tcp6
rpc.mountd: svc_tli_create: could not open connection for udp6
rpc.mountd: svc_tli_create: could not open connection for tcp6
rpc.mountd: svc_tli_create: could not open connection for udp6
rpc.mountd: svc_tli_create: could not open connection for tcp6
OK
atd: created missing .SEQ file (atjobs will be lost on reboot)
Starting atd: OK

Welcome to Buildroot
buildroot login: root
# [   24.080000] mmc0: card aaaa removed
[   30.680000] mmc0: SD Status: Invalid Allocation Unit size
[   30.710000] mmc_sd_setup_card: mdelay 20 before mmc_read_switch
[   30.730000] mmc_sd_init_card: mdelay 20 before mmc_sd_switch_hs
[   30.760000] mmc_sd_init_card: mdelay 20 before mmc_sd_get_max_clock
[   30.770000] mmc0: new SDHC card at address e773
[   30.770000] mmcblk0: mmc0:e773 B04GK 3.69 GiB 
```

yachen, 2020-06-10:
Hi, Can you check the voltage level of PC12? Wonder why there's always a card remove message before kernel trying to detect the card type.

tonyho, 2020-06-12:
> 
> 
> Hi, Can you check the voltage level of PC12? Wonder why there's always a card remove message before kernel trying to detect the card type.

The card remove message is printed when I remove the mmc card.

yachen, 2020-06-15:
What about other Port C pins (such as SD DATx/CMD/CLK), are they shared with other function on your board? 
It's difficult for us to check this issue remotely without your hardware. Who's your distributor in ShangHai? Can you ask them to check your hardware?

Does the SD PC pins share with other function on your board?
--> You mean the CD pin? No.

tonyho, 2020-06-16:
> What about other Port C pins (such as SD DATx/CMD/CLK), are they shared with other function on your board?
> It's difficult for us to check this issue remotely without your hardware. Who's your [...]

Thanks for such a long support in Github issue, I think the root case for this is caused by the hardware. We tried to manufact 2 more board, and both of them without this issue.
So thanks all your reply, and I will close this issue.

We buy the NUC980 for a remote distributor in ShenZhen not in Shanghai. 
Also hope that the NUC would have a wiki or discussion forum for customers.

yachen, 2020-06-16:
Below are URLs for our forums. Post issues here are always welcome.
http://forum.nuvoton.com/
http://www.nuvoton-mcu.com/forum.php?mod=forumdisplay&fid=77
https://bbs.21ic.com/iclist-187-1.html

Sincerely,

Yi-An Chen

closed 2020-06-16 by tonyho

redworlf007, 2020-09-01:
> 以下是我们论坛的URL。始终欢迎在此处发布问题。
> [http://forum.nuvoton.com/ ](http://forum.nuvoton.com/)
> [http://www.nuvoton-mcu.com/forum.php?mod=forumdisplay&fid=77 ](http://www.nuvoton-mcu.com/forum.php?mod=forumdisplay&fid=77)
> https://bbs.21ic.com/iclist-187-1.html
> 
> 真诚的
> 
> 陈宜安

nuc980  SD有两个SD0_nCD，一个PB8 一个PC12，怎么配置用哪一个引脚呢？