OpenNuvoton/NUC980-linux-4.4.y #24: use rtc-3232 driver，the system resters

heater, 2020-06-04:
my board uses the nuc980 chip,i2c-0 and i2c-1are used, the ds3232M is attached to i2c-0,the driver is compiled into the kernel. According to the trace printing information,the system restarts after the initialization of ds3232 without any error print infomation.   Do not compile ds3232 driver system is noemal, re-clone down the kernel,and only compile rtc-ds3232 will restart.

heater, 2020-06-05:
[image] This picture is the console print result, I added a few prints myself

heater, 2020-06-05:
[image]

yachen, 2020-06-05:
Did you enable watch dog timer by power on setting?

heater, 2020-06-05:
Both the kernel and uboot watchdog are off. Also，one time i2c_smbus_read/write should wait about 6-8 seconds

yachen, 2020-06-05:
Please dump SYS_RSTSTS register, its value indicates the chip reset source.

heater, 2020-06-05:
Thanks for the answers.  It should be a hardware cause of low voltage causing the system to restart

closed 2020-09-10 by yachen