OpenNuvoton/NUC980-linux-4.4.y #25: Ethernet failed in NUC980 with LAN8720

tonyho, 2020-06-18:
Our new design using 2 LAN8720 and HR913550A, connected to PE and PF of NUC980, but both of them are not working.  Only one ethernet Interface of one board(one of three) can dhcp sometimes, and even this port is drop a lots of packets:
```
# ping baidu.com
PING baidu.com (220.181.38.148) 56(84) bytes of data.
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=2 ttl=46 time=81.6 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=3 ttl=46 time=81.4 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=5 ttl=46 time=81.1 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=6 ttl=46 time=80.9 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=10 ttl=46 time=84.7 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=16 ttl=46 time=84.3 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=30 ttl=46 time=81.2 ms
64 bytes from 220.181.38.148 (220.181.38.148): icmp_seq=36 ttl=46 time=80.8 ms

^C
--- baidu.com ping statistics ---
42 packets transmitted, 8 received, 80.9524% packet loss, time 45148ms
rtt min/avg/max/mdev = 80.775/81.987/84.683/1.458 ms
# 
```

But in the most case, it cannot dhcp successful.  I have confirm that the Router and RJ45 wire are OK using other devices.

The Schematic of the LAN is:

[image]

[image]

[image]

[image]

Except the LN5052 as the Power supply for the Core board, the base board has a LM1117-3.3V for the HR91355, the layout:

[image]

[image]

[image]

For the another Ethernet:

[image]

[image]

[image]

```
# ifconfig 
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:1368 (1.3 KiB)

eth1      Link encap:Ethernet  HWaddr 08:00:27:00:01:93  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:91 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:31122 (30.3 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1482 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1482 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:109588 (107.0 KiB)  TX bytes:109588 (107.0 KiB)
```

All the magnetic bead (L2~L5) have been replace with 0Ω, R25 and R5 are not mounted.  And I have notice that: the `LINK LED`(Yellow) in RJ45 socket is on, while the `Active LED`(Green) is  blinking very fast.
Hope you can give some information for futher information, thanks.

yachen, 2020-06-18:
Hi,

There's comparability issue between NUC980 and LAN8720 due to hold time violation. Please replace with pin-2-pin compatible part LAN8742.

Sincerely,

Yi-An Chen

tonyho, 2020-06-18:
> 
> 
> Hi,
> 
> There's comparability issue between NUC980 and LAN8720 due to hold time violation. Please replace with pin-2-pin compatible part LAN8742.
> 
> Sincerely,
> 
> Yi-An Chen

Thanks for quickly response,  I will try that. But there're many NUC980 board using LAN8720 in Taobao.com, I guess they should work?

yachen, 2020-06-18:
Well, the timing of NUC980 and LAN8720 are incompatible. And you're not the 1st customer report such issue. Not sure how those products on TaoBao works.
By the way we use IP101 on our demo boards.
https://nuvoton.world.tmall.com/shop/view_shop.htm?_lang=zh_CN:TB-GBK

tonyho, 2020-06-23:
> 
> 
> Well, the timing of NUC980 and LAN8720 are incompatible. And you're not the 1st customer report such issue. Not sure how those products on TaoBao works.
> By the way we use IP101 on our demo boards.
> https://nuvoton.world.tmall.com/shop/view_shop.htm?_lang=zh_CN:TB-GBK

I have replace 4 LAN8720 with LAN8742, 3 of them are working now, the rest one still have packet lossing problem( 88.89% packet lost), I guess this one is cause by SMT/soldering  problem, thanks. and I will open a new issue about using IP101GR can not working.

closed 2020-06-23 by tonyho

ct991218620, 2023-03-12:
我也遇到了同样的问题，不知道有没有解决