OpenNuvoton/NUC980-linux-4.4.y #26: USB host lite ports don't support hotplug detection

jaydcarlson, 2020-06-21:


yachen, 2020-06-22:
Hi, Which board are you working on?

jaydcarlson, 2020-06-22:
Just a custom little break-out board I built for the 64-pin NUC980 package. And now that you mention it, I'm looking now and noticing the hardware design guide shows 15k pull-down resistors on the host lite ports. I assume I need those? lol that would explain it...

jaydcarlson, 2020-06-22:
I added 15k pull-downs and that fixes the issue! Sorry for the bother!

closed 2020-06-22 by jaydcarlson

AnterCreeper, 2024-02-06:
> Just a custom little break-out board I built for the 64-pin NUC980 package. And now that you mention it, I'm looking now and noticing the hardware design guide shows 15k pull-down resistors on the host lite ports. I assume I need those? lol that would explain it...

Does NUC980 USB 1.1 FS Host Lite supported by 64pin product(DR63YC)?
From datasheet, it doesn't support(from PB_NUC980_Series_EN_1.0.pdf and TRM_NUC980_Series_EN_Rev1.01.pdf), but seems does support(from DS_NUC980_Series_EN_Rev1.24.pdf)......

ychuang3, 2024-02-06:
NUC980 USB Host Lite is supported throughout the entire NUC980 series.