OpenNuvoton/NUC980-linux-4.4.y #27: YAFFS2 support on NUC980 IoT SPI NAND Flash

jaydcarlson, 2020-06-26:
I'm trying to get YAFFS2 working on my NUC980 IoT dev board.

Here's my `run.ini` file for use with `nuwriter`:
```
image_num = 5
image0_path= /home/jay/embedded-linux/nuc980nand/buildroot/output/build/uboot-master/spl/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x000200
image0_start_offset = 0x00000

image1_type = data
image1_path= /home/jay/embedded-linux/nuc980nand/buildroot/output/images/u-boot.bin
image1_exe_addr = 0x00
image1_start_offset = 0x100000

image2_type = data
image2_path= /home/jay/embedded-linux/nuc980nand/buildroot/output/images/uImage
image2_exe_addr = 0x0000
image2_start_offset = 0x200000

image3_type = env
image3_path= /home/jay/embedded-linux/nuc980nand/buildroot/nuc980_mtd_env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= /home/jay/embedded-linux/nuc980nand/buildroot/output/images/rootfs.yaffs2
image4_exe_addr = 0x00000
image4_start_offset = 0x1600000
```

Here's my `env.txt` file:
```
baudrate=115200
bootcmd=sf probe 0 75000000;sf read 0x7FC0 0x200000 0x800000;bootm 0x7FC0
bootdelay=5
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
```

Here's my kernel config:
```
 <*>   yaffs2 file system support
    -*-     512 byte / page devices
    [ ]       Use older-style on-NAND data format with pageStatus text
    [ ]         Lets yaffs do its own ECC
    -*-     2048 byte (or larger) / page devices
    [*]       Autoselect yaffs2 format
    [ ]       Disable yaffs from doing ECC on tags by default
    [ ]     Force chunk erase check
    [ ]     Empty lost and found on boot
    [ ]     Disable yaffs2 block refreshing
    [ ]     Disable yaffs2 background processing
    [ ]     Disable yaffs2 bad block marking              
    [*]     Enable yaffs2 xattr support
```

Unfortunately, when I boot my system, I get ECC error messages:
```
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
nand: Winbond W25N01GV 1G 3.3V
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
Scanning device for bad blocks
mt29f spi0.0: ecc error, page=26688
mt29f spi0.0: ecc error, page=26752
mt29f spi0.0: ecc error, page=26816
mt29f spi0.0: ecc error, page=26880
mt29f spi0.0: ecc error, page=26944
...
mt29f spi0.0: ecc error, page=29440
spinand: ECC error
mt29f spi0.0: ecc error, page=29504
spinand: ECC error
mt29f spi0.0: ecc error, page=29568
...
```
Eventually, the system boots up properly, and all my files are there, but I can't create any new files (though I can edit them, it seems).
```
# cp mdev.conf mdev.conf.old
cp: can't create 'mdev.conf.old': Cannot allocate memory
# touch testFile.txt
touch: testFile.txt: Cannot allocate memory
```

Any ideas?

yachen, 2020-06-29:
Hi, I think buildroot doesn't use inband-tags option to generate yaffs2 image. Please try again without "rootflags=inband-tags" in bootargs.

jaydcarlson, 2020-06-29:
Actually, according to my build output, yaffs2 is doing inband-tags. Here's what buildroot prints during the image creation process:
```
/mnt/storage/embedded-linux/nuc980nand/buildroot/output/host/usr/bin/mkyaffs2 --inband-tags --all-root /mnt/storage/embedded-linux/nuc980nand/buildroot/output/target /mnt/storage/embedded-linux/nuc980nand/buildroot/output/images/rootfs.yaffs2\n' >> /mnt/storage/embedded-linux/nuc980nand/buildroot/output/build/_fakeroot.fs
```
Anyway, I tried taking out the inband-tags from the bootargs and now it can't mount the filesystem at all. 

What's going on with the zillions of ECC errors I'm getting in my boot log? It looks like sometimes I'm able to modify files or even create them, but it's rare. Is there some sort of block size / erase size misconfiguration somewhere?

yachen, 2020-06-30:
Hi. Can you set "erase_all = yes,  run = erase" in run.ini to erase whole SPI NAND. And then program again with your setting but change "run = program_verify". Thanks.

jaydcarlson, 2020-07-09:
That didn't work either, unfortunately. 

Have you tried getting YAFFS2 working on your end? Do you have a different block file system you recommend for the NUC980? JFFS works, but it's very slow to mount on start-up. I haven't tried UBI, but I thought that was more optimized for bigger systems.

By default, your SDK is set up to just run everything out of an initrd image, but it'd be nice to be able to persist data across reboot cycles. What do you recommend using?

yachen, 2020-07-09:
Yes. We've tested this feature (yaffs2 as rootfs on SPI NAND). Attached is our buildroot/kernel's config and env used by u-boot.

[config.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/4896187/config.zip)

bernardlee99, 2020-07-22:
I have tried out your buildroot configuration, but I got this error while creating the image. 

```
/usr/bin/make -j1 O=/home/bbd/Desktop/NUC970_Buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/bbd/Desktop/NUC970_Buildroot/output/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/bbd/Desktop/NUC970_Buildroot/output/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/bbd/Desktop/NUC970_Buildroot/output/build/buildroot-config/tristate.config BR2_CONFIG=/home/bbd/Desktop/NUC970_Buildroot/.config HOST_GCC_VERSION="5" BUILD_DIR=/home/bbd/Desktop/NUC970_Buildroot/output/build SKIP_LEGACY= /home/bbd/Desktop/NUC970_Buildroot/output/build/buildroot-config/conf --silentoldconfig Config.in
#
# configuration written to /home/bbd/Desktop/NUC970_Buildroot/.config
#
>>> linux master Building
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     scripts/mod/devicetable-offsets.h
make[3]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  ./scripts/gen_initramfs_list.sh: Cannot open '../../images/rootfs.cpio'
usr/Makefile:73: recipe for target 'usr/initramfs_data.cpio.gz' failed
make[3]: *** [usr/initramfs_data.cpio.gz] Error 1
  CHK     include/generated/compile.h
Makefile:1001: recipe for target 'usr' failed
make[2]: *** [usr] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: *** wait: No child processes.  Stop.
package/pkg-generic.mk:216: recipe for target '/home/bbd/Desktop/NUC970_Buildroot/output/build/linux-master/.stamp_built' failed
make[1]: *** [/home/bbd/Desktop/NUC970_Buildroot/output/build/linux-master/.stamp_built] Error 2
Makefile:76: recipe for target '_all' failed
make: *** [_all] Error 2

```

yachen, 2020-07-23:
Oops, please try un-select  menuconfig->filesystem images->"initial ram filesystem linked into linux kernel" and "cpio the root filesystem".

bernardlee99, 2020-07-23:
For the configuration that you gave, only "tar the root filesystem" and yaffs2 root filesystem" is checked. I have no idea why it is attempting to get rootfs.cpio

yachen, 2020-07-28:
Can you give me the .config file in your linux-master direcotry? Thanks.

bernardlee99, 2020-09-08:
Sorry for the late response. I have switched my platform to the NUC980 Chilli Board. Can you send me a config file for Buildroot for that board so that I can try it out? I am also getting similar file system issues here. 

File changes gets discarded after power off. Attempting to create a file within mounted volume gives the following error.
```
/mnt/mtdblock1 # touch test.txt
Division by zero in kernel.
CPU: 0 PID: 1163 Comm: touch Tainted: G         C      4.4.207 #22
Hardware name: NUC980
Backtrace:
[<c0012ec0>] (dump_backtrace) from [<c00130ac>] (show_stack+0x18/0x1c)
 r6:00000002 r5:c385a000 r4:c2b19000 r3:00000000
[<c0013094>] (show_stack) from [<c01a2c18>] (dump_stack+0x20/0x28)
[<c01a2bf8>] (dump_stack) from [<c0013014>] (__div0+0x18/0x20)
[<c0012ffc>] (__div0) from [<c01a18f8>] (Ldiv0+0x8/0x10)
[<c014fb14>] (jffs2_reserve_space) from [<c0153370>] (jffs2_do_create+0x3c/0x2ec)
 r10:c36fef68 r9:c3701000 r8:c36fede8 r7:c3701148 r6:c2b19000 r5:c2abe480
 r4:c3701170
[<c0153334>] (jffs2_do_create) from [<c014cdd0>] (jffs2_create+0x9c/0x138)
 r10:c36fef68 r9:c2b19000 r8:c36fedd0 r7:c3701148 r6:c3701028 r5:c2abe480
 r4:c3701170
[<c014cd34>] (jffs2_create) from [<c00c0450>] (vfs_create+0x88/0x98)
 r9:00000000 r8:c36fef68 r7:c385bf64 r6:c385beb8 r5:c3701028 r4:c36fedd0
[<c00c03c8>] (vfs_create) from [<c00c20a0>] (path_openat+0x7cc/0xfa8)
 r5:000081a4 r4:00020042
[<c00c18d4>] (path_openat) from [<c00c3fa0>] (do_filp_open+0x68/0xbc)
 r10:00000000 r9:c385a000 r8:c000ff28 r7:00000001 r6:c385beb8 r5:c385bf64
 r4:00000003
[<c00c3f38>] (do_filp_open) from [<c00b4f34>] (do_sys_open+0x108/0x190)
 r7:00000005 r6:ffffff9c r5:c38ad000 r4:00000003
[<c00b4e2c>] (do_sys_open) from [<c00b4fe4>] (SyS_open+0x28/0x2c)
 r6:bec8af77 r5:01b60000 r4:00020042
[<c00b4fbc>] (SyS_open) from [<c000fd40>] (ret_fast_syscall+0x0/0x3c)
touch: test.txt: No space left on device
/mnt/mtdblock1 #

```

Attached is the boot log for the default nuc980_chilli defconfig file.
```
jffs2: version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
io scheduler noop registered (default)
nuc980_gpio_probe - pdev = nuc980-gpio
nuc980_dma_probe - pdev = nuc980-dma
nuc980-dma nuc980-dma: NUC980 DMA ready
nuc980-uart.0: ttyS0 at I/O 0x0 (irq = 36, base_baud = 750000) is a NUC980
nuc980-uart.4: ttyS4 at I/O 0x0 (irq = 39, base_baud = 9375000) is a NUC980
nuc980-uart.8: ttyS8 at I/O 0x0 (irq = 41, base_baud = 9375000) is a NUC980
nuvoton-rng nuvoton-rng: nuvoton PRNG active
brd: module loaded
RX nuc980_qspi0_probe: dma0chan0 module removed
TX nuc980_qspi0_probe: dma0chan1 module removed
m25p80 spi0.0: found w25q128, expected mx66l51235l
m25p80 spi0.0: w25q128 (16384 Kbytes)
Creating 2 MTD partitions on "m25p80":
0x000000000000-0x000001400000 : "kernel"
mtd: partition "kernel" extends beyond the end of device "m25p80" -- size trunca                                                                                                                                                                                                                                             ted to 0x1000000
0x000001400000-0x000002000000 : "rootfs"
mtd: partition "rootfs" is out of reach -- disabled
CAN device driver interface
nuc980-can3 nuc980-can3: nuc980_can3_platform device registered (regs=f00a3000,                                                                                                                                                                                                                                              irq=42)
PPP generic driver version 2.4.2
PPP BSD Compression module registered
PPP Deflate Compression module registered
PPP MPPE Compression module registered
NET: Registered protocol family 24
usbcore: registered new interface driver rtl8192cu
usbcore: registered new interface driver qmi_wwan
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
nuc980-ehci nuc980-ehci: Nuvoton NUC980 EHCI Host Controller
[... 122 lines cut ...]
 r10:00000000 r9:c0a492a8 r8:00000000 r7:c2ae5800 r6:00000000 r5:00000000
 r4:c2b19000
[<c015a070>] (jffs2_do_fill_super) from [<c015a8c8>] (jffs2_fill_super+0xc0/0xf4                                                                                                                                                                                                                                             )
 r8:00000000 r7:00000001 r6:00000000 r5:c2b19000 r4:c2ae5800 r3:c046b9c4
[<c015a808>] (jffs2_fill_super) from [<c0222f40>] (mount_mtd_aux+0x84/0xcc)
 r8:c0a492a8 r7:00000000 r6:00008000 r5:c2a5aa00 r4:c2ae5800 r3:c015a808
[<c0222ebc>] (mount_mtd_aux) from [<c02230ac>] (mount_mtd+0xfc/0x140)
 r7:00000000 r6:01f00001 r5:00008000 r4:c2ad91a0
[<c0222fb0>] (mount_mtd) from [<c015a410>] (jffs2_mount+0x1c/0x28)
 r8:00000000 r7:c0a492a8 r6:c0a492a8 r5:c2ad91a0 r4:c2a9ff00
[<c015a3f4>] (jffs2_mount) from [<c00b903c>] (mount_fs+0x1c/0xb0)
[<c00b9020>] (mount_fs) from [<c00d38bc>] (vfs_kern_mount+0x50/0x11c)
 r6:00008000 r5:c2ad91a0 r4:c2a9ff00
[<c00d386c>] (vfs_kern_mount) from [<c00d6ea0>] (do_mount+0x1a0/0xcc4)
 r8:c0a32a48 r7:c2ad9260 r6:c2ad91a0 r5:00000020 r4:00000000 r3:00000000
[<c00d6d00>] (do_mount) from [<c00d7d1c>] (SyS_mount+0x7c/0xa8)
 r10:00000000 r9:c2b0c000 r8:beae6f5a r7:00008000 r6:00000000 r5:c2ad91a0
 r4:c2ad9260
[<c00d7ca0>] (SyS_mount) from [<c000fd40>] (ret_fast_syscall+0x0/0x3c)
 r8:c000ff28 r7:00000015 r6:b6f884ac r5:beae6c80 r4:00000000
[Mount JFFS2]: /dev/mtdblock1 --> /mnt/mtdblock1
ifconfig: SIOCSIFADDR: No such device
route: SIOCADDRT: Network is unreachable

```

yachen, 2020-09-08:
Hi,

The chili board config is already included in buildroot. You can use `make nuvoton_nuc980_chili_defconfig` to select it.

Sincerely,

Yi-An Chen

jaydcarlson, 2020-09-09:
I think Bernard is talking about a chili config (SPI NOR) that uses YAFFS.

yachen, 2020-09-10:
Hi @bernardlee99,

What board are you using? This is your boot log.
```
m25p80 spi0.0: found w25q128, expected mx66l51235l
m25p80 spi0.0: w25q128 (16384 Kbytes)
Creating 2 MTD partitions on "m25p80":
0x000000000000-0x000001400000 : "kernel"
mtd: partition "kernel" extends beyond the end of device "m25p80" -- size truncated to 0x1000000
0x000001400000-0x000002000000 : "rootfs"
mtd: partition "rootfs" is out of reach -- disabled
```

And here is mine using chili config on chili board. Seems the flash on your board is only half the size.
The partition doesn't fit with the SPI flash your're using. I believe that's the reason JFFS2 triggers those error message.
```
m25p80 spi0.0: found w25q256, expected mx66l51235l
m25p80 spi0.0: w25q256 (32768 Kbytes)
Creating 2 MTD partitions on "m25p80":
0x000000000000-0x000001400000 : "kernel"
0x000001400000-0x000002000000 : "rootfs"
```

Also there're some pinctrl conflict warning I'm unable to find in my boot log. You might want to make sure the kernel config matches with your board.

Sincerely,

Yi-An Chen

bernardlee99, 2021-03-09:
Hello @yachen ,

I apologize for the late response here. I have switched back to the NUC980-IoT board. Attached is my config file. 

[config.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6105922/config.zip)

Here's what I have attempted:
- Re-cloned the repo for a fresh start.
- Overwriting my *output/build/linux-master/.config* file with your *nuc980_iot_linux_config* file(renamed). 
- Overwriting my  buildroot's directory *.config* file on with the *nuc980_iot_buildroot_config* file(renamed). 
- Built the Linux FS with the .yaffs2 image generated in the image directory. 
- Erase ran with *erase_all* set to yes before the flash.

This is the parameters in my run.ini file:
```
run = program_verify

#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 5
image0_path= /home/bbd/Desktop/new/NUC970_Buildroot/output/build/uboot-master/spl/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x200
image0_start_offset = 0x00000

image1_type = data
image1_path= /home/bbd/Desktop/new/NUC970_Buildroot/output/images/u-boot.bin
image1_exe_addr = 0x00
image1_start_offset = 0x100000

image2_type = data
image2_path= /home/bbd/Desktop/new/NUC970_Buildroot/output/images/uImage
image2_exe_addr = 0x0000
image2_start_offset = 0x200000

image3_type = env
image3_path= ./env-2.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= /home/bbd/Desktop/new/NUC970_Buildroot/output/images/rootfs.yaffs2
image4_exe_addr = 0x0000
image4_start_offset = 0x1600000
```

And this is the output for the given flash:
```
Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.207 (bbd@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #1 PREEMPT Mon Mar 8 21:29:58 CST 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 59296K/65536K available (3734K kernel code, 277K rwdata, 1212K rodata, 140K init, 216K bss, 6240K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc04dce24   (4948 kB)
[... 164 lines cut ...]
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c03a7d08 r4:c046eafc r3:00000000
[<c0013014>] (show_stack) from [<c017e298>] (dump_stack+0x20/0x28)
[<c017e278>] (dump_stack) from [<c0075f28>] (panic+0xb0/0x240)
[<c0075e7c>] (panic) from [<c03a7dd8>] (kernel_init+0xd0/0xf4)
 r3:c3840000 r2:c3f6e0b0 r1:60000013 r0:c046eafc
 r7:00000000
[<c03a7d08>] (kernel_init) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.

```

Can you perform this build's configuration on your development board to see whether similar issue persists? Thank you. 

For convenience, attached is my env.txt file:
```
baudrate=115200
bootcmd=sf probe 0 75000000;sf read 0x7FC0 0x200000 0x800000;bootm 0x7FC0
bootdelay=5
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
```

yachen, 2021-03-09:
Hi,

Did you generate the rootfs with inband-tag option? You should set the rootflags as below in boot options.
rootflags=inband-tags

Sincerely,

Yi-An Chen

________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

bernardlee99, 2021-03-09:
Thank you for the prompt response @yachen ,

That helped! The file write is persistent now, and the boot log is short and simple with no ECC errors anymore!

In case if others are running into this issue, with the config provided by @yachen, the *run.ini* file I provided above and my env.txt file(I did change *sf probe* to *nand read* too here):
```
baudrate=115200
bootcmd=nand read 0 75000000;nand read 0x7FC0 0x200000 0x800000;bootm 0x7FC0
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M rootflags=inband-tags
```
This should get you up and running with persistent file storage. If you run into ECC errors, try running:
- *nand bad* too see bad blocks
- *nand erase.chip* to erase the NAND flash.
- *nand scrub.chip* to scrub the NAND flash(this should be final resort and should not be done all the time).

bernardlee99, 2021-03-29:
It seem that the suggested solution worked for one iteration. Upon changing things in menuconfig, I got myself into a Kernel Panic once again. However, this time even after re-cloning and using the exact same config and envs, it comes into this issue. I tried using _erase_all_ before the reflash. Anything I am missing here? 

Here's my config file:
[config.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6218994/config.zip)

And here's my parameters in my run.ini file:
```
run = program_verify

#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 5
image0_path= /home/bbd/Desktop/NUC970_Buildroot/output/build/uboot-master/spl/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x200
image0_start_offset = 0x00000

image1_type = data
image1_path= /home/bbd/Desktop/NUC970_Buildroot/output/images/u-boot.bin
image1_exe_addr = 0x00
image1_start_offset = 0x100000

image2_type = data
image2_path= /home/bbd/Desktop/NUC970_Buildroot/output/images/uImage
image2_exe_addr = 0x0000
image2_start_offset = 0x200000

image3_type = env
image3_path= ./env-2.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= /home/bbd/Desktop/NUC970_Buildroot/output/images/rootfs.yaffs2
image4_exe_addr = 0x0000
image4_start_offset = 0x1600000
```

```
U-Boot 2016.11 (Mar 28 2021 - 17:43:09 -0500)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

NAND read: Offset exceeds device limit

NAND read: device 0 offset 0x200000, size 0x800000
 8388608 bytes read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2635424 Bytes = 2.5 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
[... 149 lines cut ...]
lib80211: common routines for IEEE802.11 drivers
ALSA device list:
  #0: nuc980_IIS
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags "inband-tags"
random: nonblocking pool is initialized
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: mounted
Freeing unused kernel memory: 140K
Kernel panic - not syncing: No working init found.  Try passing init= option to                                                                                                                                                                                                                                              kernel. See Linux Documentation/init.txt for guidance.
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.207 #1
Hardware name: NUC980
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c03a7d08 r4:c046eafc r3:00000000
[<c0013014>] (show_stack) from [<c017e298>] (dump_stack+0x20/0x28)
[<c017e278>] (dump_stack) from [<c0075f28>] (panic+0xb0/0x240)
[<c0075e7c>] (panic) from [<c03a7dd8>] (kernel_init+0xd0/0xf4)
 r3:c3840000 r2:c3f6e0b0 r1:60000013 r0:c046eafc
 r7:00000000
[<c03a7d08>] (kernel_init) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: No working init found.  Try passing init= o                                                                                                                                                                                                                                             ption to kernel. See Linux Documentation/init.txt for guidance.

```

yachen, 2021-04-02:
Hi @bernardlee99, not sure what goes wrong yet, but can you erase whole flash before program the images and see if the problem still exist?

bernardlee99, 2021-04-03:
Yes, I have used the _erase_all_ option, and I have scrubbed the flash using U-boot options. As a final resort, I have replaced the SPI flash IC and the problem still persists. Can you send me the image that works with your setup(images and the _run.ini_ file for _nuwriter_)?

closed 2021-07-30 by yachen