OpenNuvoton/NUC980-linux-4.4.y #34: NK-980IOT development board USB1 HOST NOT Working

ifdreams, 2020-09-22:
<*> SCSI disk support
<*> USB Mass Storage support

Insert the U disk (FAT32) DEVICE,Nothing happens.

<*> EHCI HCD (USB 2.0) support
<*> OHCI HCD (USB 1.1) support 
<*> ChipIdea Highspeed Dual Role Controller
[*] ChipIdea device controller
[*] ChipIdea host controller
<*> IEEE 802.11 for Host AP (Prism2/2.5/3 and WEP/TKIP/CCMP)
[*] Support downloading firmware images with Host AP driver
[*] Support for non-volatile firmware download
[*] cfg80211 wireless extensions compatibility
<*> Generic IEEE 802.11 Networking Stack (mac80211)
<*>   Realtek RTL8188EU Wireless LAN NIC driver                                   
[*]     Realtek RTL8188EU AP mode 

Insert the RTL8188EU DEVICE in USB1,Nothing happens.

Compiling module RTL8188EU.ko
insmod RTL8188EU.ko

Insert the RTL8188EU DEVICE in USB1,Nothing happens.

Can you tell me how to make RTL8188EU DEVICE work on USB1,
or make U disk (FAT32) work on USB1, 
I think USB  DRIVERS is not working now.

yachen, 2020-09-22:
Please use nuc980_iot_defconfig to build kernel for  IoT board.

ifdreams, 2020-09-22:
Thank you very much, it works fine! there is one more problem, can you tell me about the description file of nuc980-iot-v1.0.dts? in Documentation?

closed 2020-09-22 by ifdreams