OpenNuvoton/NUC980-linux-4.4.y #35: NUC980  SPI NAND 启动问题。

redworlf007, 2020-09-22:
1、NUC980 从 SPI NAND启动，SD0位置上有SD NAND。
2、可以从usb启动，并且可以烧录系统，串口控制台输出正常。
3、烧录完系统，从spi nand启动后，串口控制台一直打印乱码。
4、按理说，能烧录成功，启动就不应该乱码，求助。
[image]
[image]

redworlf007, 2020-09-22:
SPL load main U-Boot from SPI NAND Flash 

感觉是这里执行失败。

redworlf007, 2020-09-22:
[image]
烧录成功

yachen, 2020-09-23:
Hi,

你在哪一張板子上測試? IoT, LoraG...?

Sincerely,

Yi-An Chen

redworlf007, 2020-10-06:
谢谢，我找到问题了，我把uboot的烧录地址不对。

redworlf007, 2020-10-06:
> 你好
> 
> 你在哪一张板子上测试？物联网，LoraG ...？
> 
> 真诚的
> 
> 陈宜安

谢谢，我找到问题了，是我uboot烧录错地址了。

closed 2020-10-06 by redworlf007