OpenNuvoton/NUC980-linux-4.4.y #36: 请教：128引脚的nuc980，如何配置PG[15-11]为普通io口。

redworlf007, 2020-10-06:
1、128引脚的nuc980，没有PG4引脚，PG[15-11]默认为jtag引脚。
2、如何配置为普通的io口，内核里面没有发现可以配置的地方。
谢谢。

yachen, 2020-10-07:
Hi,

目前沒有放入將 multi function pin 切換回 GPIO 的功能.
請在 pinctrl-nuc980.c 的 probe function 加上幾行 code 讓 kernel 一開始就將這幾跟 pin 設置成 GPIO.

```
static int nuc980_pinctrl_probe(struct platform_device *pdev)
{
	struct pinctrl_dev *pctl;
	u32 reg;

	pctl = pinctrl_register(&nuc980_pinctrl_desc, &pdev->dev, NULL);
	if (IS_ERR(pctl)) {
		pr_err("could not register nuc980 pin driver\n");
	}

	platform_set_drvdata(pdev, pctl);

	// Restore JTAG pins @ PG11~15 to GPIO
	reg = __raw_readl(REG_MFP_GPG_H);
	reg &= ~0xFFFFF000;
	__raw_writel(reg, REG_MFP_GPG_H);

	return pinctrl_register_mappings(nuc980_pinmap, ARRAY_SIZE(nuc980_pinmap));

}
```

Sincerely,

Yi-An Chen

redworlf007, 2020-10-07:
> 你好
> 
> 目前没有引入将多功能引脚转换回GPIO的功能。
> 请在pinctrl-nuc980.c的探针功能加上几行代码让内核一开始就将这几跟pin设置成GPIO。
> 
> ```
> static int nuc980_pinctrl_probe(struct platform_device *pdev)
> {
> 	struct pinctrl_dev *pctl;
> 	u32 reg;
> 
> 	pctl = pinctrl_register(&nuc980_pinctrl_desc, &pdev->dev, NULL);
> 	if (IS_ERR(pctl)) {
> 		pr_err("could not register nuc980 pin driver\n");
> 	}
> 
> 	platform_set_drvdata(pdev, pctl);
> 
> 	// Restore JTAG pins @ PG11~15 to GPIO
> 	reg = __raw_readl(REG_MFP_GPG_H);
> 	reg &= ~0xFFFFF000;
> 	__raw_writel(reg, REG_MFP_GPG_H);
> 
> 	return pinctrl_register_mappings(nuc980_pinmap, ARRAY_SIZE(nuc980_pinmap));
> 
> }
> ```
> 
> 真诚的
> 
> 陈宜安

好的，谢谢，等下我试试看行不行，还有个问题，我把PG口配置为SPI口、内核开启spi user model，用ss1做片选信号，外接spi的屏幕，nuc980发数据给spi的屏幕，用的是新塘BSP包里面的spi  demo程序，会死机，发送0xd5  这个字节死机的概率最大，我示波器测了，死机的时候，clk无信号了，屏幕没有问题，我用stm32f103的io口驱动，没问题。

yachen, 2020-10-08:
Hi,

死機是應用沒反應, 還是 shell 也停了? 內核有崩潰嗎?
SPI驅動不會去解析應用層傳進來的參數, 應跟跟字節內容無關才對.

Sincerely,

Yi-An Chen

redworlf007, 2020-10-08:
> Hi,
> 
> 死機是應用沒反應, 還是 shell 也停了? 內核有崩潰嗎?
> SPI驅動不會去解析應用層傳進來的參數, 應跟跟字節內容無關才對.
> 
> Sincerely,
> 
> Yi-An Chen

对，死机的时候，应用没反应，shell也停了，无法操作，串口控制台什么都不打印，无法操作，只能复位重启，发送0xd5死机概率改，发送其他的死机概率低，我朋友他们用980 应用层操作spi也会死机，他们修改了spi驱动，加了CLK超时处理。

yachen, 2020-10-12:
Hi,

是使用BSP 包 aapplications/demo/spi 裡的範例測試嗎? 還是另外寫代碼?

Sincerely,

Yi-An Chen

closed 2021-07-30 by yachen

QHxiedong, 2022-03-17:
我的QSPI0也出现了这样的问题，QSPI0接的Nor flash，挂载文件系统并且存放sqlite3的数据库文件。程序运行大概1周左右时间有一定概率出现卡死现象，数据库文件概率被破坏，敲ls指令shell卡死，无法操作只能重启 @yachen