OpenNuvoton/NUC980-linux-4.4.y #37: usb 设备应用退出之后，无法识别

styfstr, 2020-10-15:
两块980板子，板A使用usb0做为DEVICE，板B使用usb1作为HOST，B之间A通过USB1和USB0连接，用于两个板子间通信。

板A上电，usb device mount、写入usb descriptors和string之后，板B可以正常识别A的usb接口，并可以使用libusb与板A进行数据通信。

当A unmount usb 之后，板B无法识别板A usb 消失信息。重新mount，A的usb设备也无法正常使用（可以BIND但无法ENABLE）。只有给A重新上电，才可以被B识别到。

在imx6U上同样的操作。A设备的usb 重复mount、unmount都可以被识别。

usb数据收发使用：
    device 程序参考：http://git.toradex.com/cgit/linux-toradex.git/tree/tools/usb/ffs-aio-example/simple/device_app/aio_simple.c?h=toradex_4.1-2.0.x-imx
   host 程序参考：libusb-1.0.23

styfstr, 2020-10-15:
这个是内核打印：
[image]

只有给980重新上电，才能重新注册usb设备。

yachen, 2020-10-15:
Hi,

NUC980 USB gadget 的設計無法將 D+ 信號拉低. 所以 host 端無法得知 device umount.

Sincerely,

Yi-An Chen

styfstr, 2020-10-15:
我该如何解决呢？有没有什么好的建议？

yachen, 2020-10-15:
Hi,

請問你系統的實際應用情境是怎麼樣?

Sincerely,

Yi-An Chen

styfstr, 2020-10-15:
> Hi,
> 
> NUC980 USB gadget 的設計無法將 D+ 信號拉低. 所以 host 端無法得知 device umount.
> 
> Sincerely,
> 
> Yi-An Chen
Hi,
我这边的一个应用程序注册usb设备，与另外模块的host进行通信。当这个应用程序重新启动时，需要重新注册usb设备。但是现在无法重新注册。

yachen, 2020-10-19:
enable/disable 是由上層呼叫下來的.  請問目前是使用哪個 class?

styfstr, 2020-10-19:
非常感谢您的回复！

我使用的class为0xff.
下边是我usb的描述：

/*-------------slave param start------------------*/
#define EP_PATH "/root/ep/"
#define AIO_BLKSIZE 1024 * 10
#define IOCB_FLAG_RESFD         (1 << 0)
static const struct {
	struct usb_functionfs_descs_head_v2 header;
	__le32 fs_count;
	__le32 hs_count;
	struct {
		struct usb_interface_descriptor intf;
		struct usb_endpoint_descriptor_no_audio bulk_sink;
		struct usb_endpoint_descriptor_no_audio bulk_source;
	} __attribute__((__packed__)) fs_descs, hs_descs;
} __attribute__((__packed__)) descriptors = {
	.header = {
		.magic = htole32(FUNCTIONFS_DESCRIPTORS_MAGIC_V2),
		.flags = htole32(FUNCTIONFS_HAS_FS_DESC |
					 FUNCTIONFS_HAS_HS_DESC),
		.length = htole32(sizeof(descriptors)),
	},
	.fs_count = htole32(3),
	.fs_descs = {
		.intf = {
			.bLength = sizeof(descriptors.fs_descs.intf),
			.bDescriptorType = USB_DT_INTERFACE,
			.bNumEndpoints = 2,
			.bInterfaceClass = USB_CLASS_VENDOR_SPEC,
			.iInterface = 1,
		},
		.bulk_sink = {
			.bLength = sizeof(descriptors.fs_descs.bulk_sink),
			.bDescriptorType = USB_DT_ENDPOINT,
			.bEndpointAddress = 1 | USB_DIR_IN,
			.bmAttributes = USB_ENDPOINT_XFER_BULK,
		},
		.bulk_source = {
			.bLength = sizeof(descriptors.fs_descs.bulk_source),
			.bDescriptorType = USB_DT_ENDPOINT,
			.bEndpointAddress = 2 | USB_DIR_OUT,
			.bmAttributes = USB_ENDPOINT_XFER_BULK,
		},
	},
	.hs_count = htole32(3),
	.hs_descs = {
		.intf = {
			.bLength = sizeof(descriptors.hs_descs.intf),
			.bDescriptorType = USB_DT_INTERFACE,
			.bNumEndpoints = 2,
			.bInterfaceClass = USB_CLASS_VENDOR_SPEC,
			.iInterface = 1,
		},
		.bulk_sink = {
			.bLength = sizeof(descriptors.hs_descs.bulk_sink),
			.bDescriptorType = USB_DT_ENDPOINT,
			.bEndpointAddress = 1 | USB_DIR_IN,
			.bmAttributes = USB_ENDPOINT_XFER_BULK,
			.wMaxPacketSize = htole16(512),
		},
		.bulk_source = {
			.bLength = sizeof(descriptors.hs_descs.bulk_source),
			.bDescriptorType = USB_DT_ENDPOINT,
			.bEndpointAddress = 2 | USB_DIR_OUT,
			.bmAttributes = USB_ENDPOINT_XFER_BULK,
			.wMaxPacketSize = htole16(512),
		},
	},
};

#define STR_INTERFACE "AIO Test"

static const struct {
	struct usb_functionfs_strings_head header;
	struct {
		__le16 code;
		const char str1[sizeof(STR_INTERFACE)];
	} __attribute__((__packed__)) lang0;
} __attribute__((__packed__)) strings = {
	.header = {
		.magic = htole32(FUNCTIONFS_STRINGS_MAGIC),
		.length = htole32(sizeof(strings)),
		.str_count = htole32(1),
		.lang_count = htole32(1),
	},
	.lang0 = {
		htole16(0x0409), /* en-us */
		STR_INTERFACE,
	},
};

closed 2021-12-08 by ychuang3