OpenNuvoton/NUC980-linux-4.4.y #39: Segmentation fault when use cap0

gdyshi, 2020-10-28:
I used the demo program in BSP/applications/demos/cap to get pictures from /dev/video0, but a segmentation error occurred.
To get more information, I opened the VDEBUG macro in the nuc980_cap.h file, and then ran the program cap2bmp to get the following log. I am using the latest version in current git
/test/cap # ./cap2bmp
[nuvoton_vdi0_open   ] : Enter...
[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
Video device video0 is open
[nuvoton_vdi0_open   ] : Leave...
[nuvoton_vidioc0_querycap] : Enter...
[nuvoton_vidioc0_querycap] : Leave...
[nuvoton_vidioc0_s_crop] : Enter...
[nuvoton_vidioc0_s_fmt_vid_cap] : Enter...
[nuvoton_vidioc0_try_fmt_vid_cap] : Enter...
Packet
pix->height=240, s->cropcap.bounds.height = 480
pix->width=320, s->cropcap.bounds.width = 640
[nuvoton_vidioc0_try_fmt_vid_cap] : Leave...
[nuvoton_vidioc0_s_fmt_vid_cap] : Leave...
[nuvoton_vidioc0_reqbufs] : Enter...
[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
[nuvoton_vin0_release_buffers] : Enter...
[nuvoton_vin0_release_buffers] : Leave...
[nuvoton_vin0_request_buffers] : Enter...
cam->frame[0].bufmem=0xc49a7000
cam->frame[0].pbuf=0x03200000
cam->frame[1].bufmem=0xc4a64000
cam->frame[1].pbuf=0x03300000
cam->frame[2].bufmem=0xc4b21000
cam->frame[2].pbuf=0x02c00000
[nuvoton_vin0_request_buffers] : Leave...
[nuvoton_vidioc0_reqbufs] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #0 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #1 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #2 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_streamon] : Enter...
cam->nbuffers=3
cam->frame_current->bufmem=0xc49a7000
cam->frame_current->pbuf=0x03200000
cam->vpe.PacketEnable=1
cam->vpe.PlanarEnable=0
[nuvoton_vdi0_enable ] : Enter...
[nuvoton_vdi0_enable ] : Leave...
[nuvoton_vidioc0_streamon] : Leave...
Capture image to[nuvoton_vidioc0_dqbuf] : Enter...
 image00.bmp
enter xioctl
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
Frame0 #0 dequeued[nuvoton_vidioc0_dqbuf] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
Unable to handle kernel paging request at virtual address 0005b000
pgd = c3bd8000
[0005b000] *pgd=032c4831, *pte=00000000[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
, *ppte=00000000
Internal error: Oops: 817 [#1] PREEMPT ARM
Modules linked in:
CPU: 0 PID: 1160 Comm: cap2bmp Not tainted 4.4.207 #87
Hardware name: NUC980
task: c3a1ba00 ti: c32c2000 task.ti: c32c2000
PC is at 0x1070c
LR is at 0x8770
pc : [<0001070c>]    lr : [<00008770>]    psr: 20000010
sp : bebc6b30  ip : 00000000  fp : bebc6b74
r10: 00000000  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000001  r5 : 00024000  r4 : 00000001
r3 : 00000000  r2 : 00000003  r1 : bebc6b53  r0 : 0005b000
Flags: nzCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Control: 0005317f  Table: 03bd8000  DAC: 00000055
Process cap2bmp (pid: 1160, stack limit = 0xc32c2190)
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
---[ end trace 2bed31f70429bc01 ]---
note: cap2bmp[1160] exited with preempt_count 3
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vdi0_close  ] : Enter...
[nuvoton_vdi0_disable] : Enter...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_disable] : Leave...
[nuvoton_vin0_release_buffers] : Enter...
[nuvoton_vin0_release_buffers] : Leave...
Video device video0 closed[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
[nuvoton_vdi0_close  ] : Leave...
Segmentation fault

yachen, 2020-10-29:
Hi, Can you provide your kernel config file for us to setup the same test scenario? Thanks.

gdyshi, 2020-10-29:
I am using TVP5150. The driver is modified based on sensor0_ov5640.c. The main modifications are:
1 Modify client->addr to the address of TVP5150
2 Modify the initialization register
From the printed information, it is an addressing error, which should not be caused by sensor changes.
[kernel_config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/5455592/kernel_config.txt)

yachen, 2020-10-29:
Yes, it's an invalid address. But the trace message also confused me. There're two consecutive "Leave" messages in your log. Still can't see the possibility an "Enter" could miss.

```
[nuvoton_vdi0_isr    ] : Leave...
Unable to handle kernel paging request at virtual address 0005b000
pgd = c3bd8000
[0005b000] *pgd=032c4831, *pte=00000000[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
```

gdyshi, 2020-10-29:
um, yes. You have discovered all such details. I think the possible reason is that the log here conflicts with coredump. Because if I close the VDEBUG macro, a lot of coredump records will be output, and when I open the VDEBUG macro, the coredump records are only a few lines.
And there is one more detail: there are two consecutive leaves and two consecutive enters at the coredump
```
[nuvoton_vdi0_isr    ] : Leave...
Unable to handle kernel paging request at virtual address 0005b000
pgd = c3bd8000
[0005b000] *pgd=032c4831, *pte=00000000[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
, *ppte=00000000
Internal error: Oops: 817 [#1] PREEMPT ARM
Modules linked in:
CPU: 0 PID: 1160 Comm: cap2bmp Not tainted 4.4.207 #87
Hardware name: NUC980
task: c3a1ba00 ti: c32c2000 task.ti: c32c2000
PC is at 0x1070c
LR is at 0x8770
pc : [<0001070c>]    lr : [<00008770>]    psr: 20000010
sp : bebc6b30  ip : 00000000  fp : bebc6b74
r10: 00000000  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000001  r5 : 00024000  r4 : 00000001
r3 : 00000000  r2 : 00000003  r1 : bebc6b53  r0 : 0005b000
Flags: nzCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Control: 0005317f  Table: 03bd8000  DAC: 00000055
Process cap2bmp (pid: 1160, stack limit = 0xc32c2190)
[nuvoton_vdi0_isr    ] : Enter...
```
Information output when VDEBUG macro is closed:
```
/test/cap # ./cap2bmp
Capture image to image00.bmp
enter xioctl
BUG: workqueue leaked lock or atomic: kworker/0:1/0x00000001/239
     last function: rpc_async_release
CPU: 0 PID: 239 Comm: kworker/0:1 Not tainted 4.4.207 #88
Hardware name: NUC980
Workqueue: nfsiod rpc_async_release
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:c05649e4 r5:c3a64e1c r4:c38c5b40 r3:c38fa000
[<c0013014>] (show_stack) from [<c01a2afc>] (dump_stack+0x20/0x28)
[<c01a2adc>] (dump_stack) from [<c00337d4>] (process_one_work+0x2a4/0x388)
[<c0033530>] (process_one_work) from [<c0033924>] (worker_thread+0x6c/0x5c8)
 r10:c38c5b40 r9:c05649e4 r8:00000008 r7:c38c5b58 r6:c05a7452 r5:c05649f4
 r4:c05649e4
[<c00338b8>] (worker_thread) from [<c0038de4>] (kthread+0xc8/0xe4)
 r10:00000000 r9:00000000 r8:00000000 r7:c00338b8 r6:c38c5b40 r5:00000000
 r4:c3a036e0
[<c0038d1c>] (kthread) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r7:00000000 r6:00000000 r5:c0038d1c r4:c3a036e0
BUG: scheduling while atomic: kworker/0:1/239/0x00000006
Modules linked in:
CPU: 0 PID: 239 Comm: kworker/0:1 Not tainted 4.4.207 #88
Hardware name: NUC980
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c05652e8 r4:00000000 r3:00000000
[<c0013014>] (show_stack) from [<c01a2afc>] (dump_stack+0x20/0x28)
[<c01a2adc>] (dump_stack) from [<c00767ec>] (__schedule_bug+0x4c/0x64)
[<c00767a0>] (__schedule_bug) from [<c03e8e10>] (__schedule+0x3c0/0x470)
 r4:c38a5220 r3:00000006
[<c03e8a50>] (__schedule) from [<c03e8f14>] (schedule+0x54/0xb8)
 r8:00000008 r7:c05649e4 r6:c05a7452 r5:c05649f4 r4:c38fa000 r3:00000006
[<c03e8ec0>] (schedule) from [<c00339f0>] (worker_thread+0x138/0x5c8)
 r4:c05649e4 r3:c38fa000
[<c00338b8>] (worker_thread) from [<c0038de4>] (kthread+0xc8/0xe4)
 r10:00000000 r9:00000000 r8:00000000 r7:c00338b8 r6:c38c5b40 r5:00000000
 r4:c3a036e0
[<c0038d1c>] (kthread) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r7:00000000 r6:00000000 r5:c0038d1c r4:c3a036e0
BUG: workqueue leaked lock or atomic: kworker/0:1H/0x00000001/1127
     last function: rpc_async_schedule
CPU: 0 PID: 1127 Comm: kworker/0:1H Tainted: G        W       4.4.207 #88
Hardware name: NUC980
Workqueue: rpciod rpc_async_schedule
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
```

yachen, 2020-10-29:
Hi,

We cannot reproduce this problem using your kernel configuration so far. Can you modify the cap2bmp application a bit and paste the console log here? Thanks.

```
diff --git a/demos/cap/cap2bmp.c b/demos/cap/cap2bmp.c
index 923c998..cf1dbc2 100755
--- a/demos/cap/cap2bmp.c
+++ b/demos/cap/cap2bmp.c
@@ -281,7 +281,11 @@ static void mainloop(void)
        while(1) {
                sprintf(filename,"image%02d.bmp",cnt++);
                printf("Capture image to %s \n", filename);
+#if 0
                read_frame(filename);
+#else
+                read_frame(0);
+#endif
                if(cnt>10) break;
        }
}

```

Sincerely,

Yi-An Chen

gdyshi, 2020-10-29:
Thanks for your reply.
According to your suggestion, no errors were reported after the modification. The log information is as follows

```
/test/cap # ./cap2bmp
[nuvoton_vdi0_open   ] : Enter...
[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
Video device video0 is open
[nuvoton_vdi0_open   ] : Leave...
[nuvoton_vidioc0_querycap] : Enter...
[nuvoton_vidioc0_querycap] : Leave...
[nuvoton_vidioc0_s_crop] : Enter...
[nuvoton_vidioc0_s_fmt_vid_cap] : Enter...
[nuvoton_vidioc0_try_fmt_vid_cap] : Enter...
Packet
pix->height=240, s->cropcap.bounds.height = 480
pix->width=320, s->cropcap.bounds.width = 640
[nuvoton_vidioc0_try_fmt_vid_cap] : Leave...
[nuvoton_vidioc0_s_fmt_vid_cap] : Leave...
[nuvoton_vidioc0_reqbufs] : Enter...
[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
[nuvoton_vin0_release_buffers] : Enter...
[nuvoton_vin0_release_buffers] : Leave...
[nuvoton_vin0_request_buffers] : Enter...
cam->frame[0].bufmem=0xc49a7000
cam->frame[0].pbuf=0x03200000
cam->frame[1].bufmem=0xc4a64000
cam->frame[1].pbuf=0x03300000
cam->frame[2].bufmem=0xc4b21000
cam->frame[2].pbuf=0x02c00000
[nuvoton_vin0_request_buffers] : Leave...
[nuvoton_vidioc0_reqbufs] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_querybuf] : Enter...
[nuvoton_vidioc0_querybuf] : Leave...
[nuvoton_vdi0_mmap   ] : Enter...
[nuvoton_vin0_vm_open] : Enter...
[nuvoton_vdi0_mmap   ] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #0 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #1 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_qbuf] : Enter...
Frame #2 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vidioc0_streamon] : Enter...
cam->nbuffers=3
cam->frame_current->bufmem=0xc49a7000
cam->frame_current->pbuf=0x03200000
cam->vpe.PacketEnable=1
cam->vpe.PlanarEnable=0
[nuvoton_vdi0_enable ] : Enter...
[nuvoton_vdi0_enable ] : Leave...
[nuvoton_vidioc0_streamon] : Leave...
Capture image to[nuvoton_vidioc0_dqbuf] : Enter...
 image00.bmp
enter xioctl
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
Frame0 #0 dequeued[nuvoton_vidioc0_dqbuf] : Leave...
[... 101 lines cut ...]
Frame #1 queued[nuvoton_vidioc0_qbuf] : Leave...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vidioc0_streamoff] : Enter...
[nuvoton_vdi0_disable] : Enter...
[nuvoton_vdi0_isr    ] : Enter...
[nuvoton_vdi0_isr    ] : Leave...
[nuvoton_vdi0_disable] : Leave...
[nuvoton_vidioc0_streamoff] : Leave...
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vin0_vm_close] : Enter...
[nuvoton_vin0_vm_close] : Leave...
[nuvoton_vdi0_close  ] : Enter...
[nuvoton_vdi0_disable] : Enter...
[nuvoton_vdi0_disable] : Leave...
[nuvoton_vin0_release_buffers] : Enter...
[nuvoton_vin0_release_buffers] : Leave...
Video device video0 closed[nuvoton_vin0_empty_framequeues] : Enter...
[nuvoton_vin0_empty_framequeues] : Leave...
[nuvoton_vdi0_close  ] : Leave...

```

yachen, 2020-10-29:
Probably some setting issue of the nuvoton_vin_sensor structure in the TVP5150 driver your port. Can you paste the structure in your driver here? Thanks.

gdyshi, 2020-10-29:
```
static struct nuvoton_vin_sensor cap0_tvp5150 = {
	.i2c_id = 4,
	.name = "cap0_tvp5150",
	.init = &cap0_tvp5150_init,
	.infmtord = (INORD_YUYV | INFMT_YCbCr | INTYPE_CCIR601),
	.polarity = (VSP_HI | HSP_LO | PCLKP_HI),
	.cropstart = ( 1 | 0<<16 ), /*( Vertical | Horizontal<<16 ) */
	.cropcap = {
		.bounds = {
			.left = 0,
			.top = 0,
			.width = 640,
			.height = 480,
		},
		.defrect = {
			.left = 0,
			.top = 0,
			.width = 800,
			.height = 480,
		},
	},
	.pix_format  = {
		.width = 640,
		.height = 480,
		.pixelformat = V4L2_PIX_FMT_YUYV,
		.priv = 16,
		.colorspace = V4L2_COLORSPACE_JPEG,
	},
};
```
I just copied it from sensor0_ov5640

yachen, 2020-10-30:
Hi, Please update driver and test again with original application. Thanks.
```
diff --git a/drivers/media/i2c/nuvoton/nuc980_cap0.c b/drivers/media/i2c/nuvoton/nuc980_cap0.c
index ce72b90..b2a4df0 100644
--- a/drivers/media/i2c/nuvoton/nuc980_cap0.c
+++ b/drivers/media/i2c/nuvoton/nuc980_cap0.c
@@ -1431,7 +1431,7 @@ static irqreturn_t nuvoton_vdi0_isr(int irq, void *priv)
        }

        spin_lock(&cam->queue_lock);
-       if((*f)->state == F_QUEUED) {
+       if((*f)->state == F_QUEUED && !list_is_last(&(*f)->frame,&cam->inqueue) ) {
                list_move_tail(&(*f)->frame, &cam->outqueue);
        }
        if (!list_empty(&cam->inqueue)) {
```

gdyshi, 2020-11-02:
I updated the driver according to your advice, and now there are no errors. Thank you for your timely support, now I am more confident to use nuc980

closed 2020-11-30 by yachen