OpenNuvoton/NUC980-linux-4.4.y #42: SDIO WIFI如何使用SD1接口驱动

qiubin2016, 2020-11-15:
开发板：NuMaker-IIoT-NUC980，pcb丝印型号：NK-980IOT V1.0
使用引脚：
序号  名称             IO          CON11引脚丝印
49    SD0_CMD      PC.5           6
50    SD0_CLK        PC.6           7
51    SD0_DATA0    PC.7           8
52    SD0_DATA1    PC.8           9
53    SD0_DATA2    PC.9           10
54    SD0_DATA3    PC.10         11
56    SD0_nCD        PC.12         13
想使用印上引脚驱动SDIO接口的WIFI，接线后，上电加载驱动未成功出现wlan0网卡。内核中没看到MMC/SD/SDIO配置选项可以配置使用SD0还是SD1。

yachen, 2020-11-16:
另一個 port 跟 NAND 共用 controller, 要進 menuconfig->Generic Driver Options->Nuvoton NUC980 FMI function selection->Select FMI device to support (Support SD0 Device) 打開.

qiubin2016, 2020-11-16:
感谢！开启后通过SD0接口已经可以正常加载SDIO WIFI驱动。

closed 2020-11-30 by yachen