OpenNuvoton/NUC980-linux-4.4.y #44: 内核是否没启用设备树功能

qiubin2016, 2020-11-17:
开发板：NuMaker-IIoT-NUC980，pcb丝印型号：NK-980IOT V1.0
配置：make  nuc980_iot_defconfig
内核是否没启用设备树功能?

yachen, 2020-11-17:
這個config default 沒開. 要開的話 menuconfig->Boot options->[*] Flattened Device Tree support

qiubin2016, 2020-11-17:
ok

qiubin2016, 2020-11-28:
启用内核设备树功能，开发板开机无法挂载文件系统。开启方法：menuconfig->Boot options->[*] Flattened Device Tree support

开发板：NuMaker-IIoT-NUC980，pcb丝印型号：NK-980IOT V1.0
配置：make nuc980_iot_defconfig
设备树文件：arch/arm/boot/dts/nuc980-iot-v1.0.dtb
--------u-boot env--------:
baudrate=115200
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
bootcmd=run setspi;run loadkernel;run loaddtb;run bootmcmd
bootdelay=1
bootmcmd=bootm 0x7fc0 - 0x1400000
ethact=emac
loaddtb=sf read 0x1400000 0xB00000 0x80000
loadkernel=sf read 0x7fc0 0x200000 0x800000
setspi=sf probe 0 50000000
stderr=serial
stdin=serial
stdout=serial
--------开发板启动报错日志---------：
Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: Device[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.4.179 (qiub@ubuntu1604) (gcc version 4.8.4 (GCC) ) #18 PREEMPT Fri Nov 27 14:54:41 CST 2020
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Nuvoton NUC980 IOT V1.0
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 56488K/65536K available (5658K kernel code, 372K rwdata, 1812K rodata, 176K init, 258K bss, 9048K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0753f9c   (7472 kB)
[    0.000000]       .init : 0xc0754000 - 0xc0780000   ( 176 kB)
[    0.000000]       .data : 0xc0780000 - 0xc07dd240   ( 373 kB)
[    0.000000]        .bss : 0xc07dd240 - 0xc081da94   ( 259 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:545
[    0.000000] clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
[    0.000033] sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
[    0.000724] Console: colour dummy device 80x30
[    0.187883] console [ttyS0] enabled
[    0.191416] Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
[    0.262783] pid_max: default: 32768 minimum: 301
[    0.267624] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.274308] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.283316] CPU: Testing write buffer coherency: ok
[    0.289233] Setting up static identity map for 0x8400 - 0x843c
[    0.299974] devtmpfs: initialized
[    0.341649] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.352316] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.358824] pinctrl core: initialized pinctrl subsystem
[    0.366199] NET: Registered protocol family 16
[    0.373541] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.386816] <DT> nuc980_dt_device_init +
[    0.434441] <DT> nuc980_dt_device_init -
[    0.441774] pinctrl-nuc980 apb:pinctrl@b0000000: initialized NUC980 pinctrl driver
[    0.808424] SCSI subsystem initialized
[    0.818433] usbcore: registered new interface driver usbfs
[    0.826683] usbcore: registered new interface driver hub
[... 178 lines cut ...]
[   13.294283] mt29f spi0.0: read from cache failed!!
[   13.299383] mt29f spi0.0: read from cache failed!!

......此处省略n行

[   17.238949] mt29f spi0.0: read from cache failed!!
[   17.244158] mt29f spi0.0: read from cache failed!!
[   17.273016] VFS: Mounted root (yaffs2 filesystem) readonly on device 31:2.
[   17.279933] devtmpfs: error mounting -2
[   17.285899] Freeing unused kernel memory: 176K
[   17.290566] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[   17.303666] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.4.179 #18
[   17.310749] Hardware name: Nuvoton NUC980 (Device Tree)
[   17.315941] Backtrace: 
[   17.318449] [<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
[   17.325966]  r6:00000000 r5:c0588afc r4:c06bdacc r3:00000000
[   17.331658] [<c0013000>] (show_stack) from [<c01a769c>] (dump_stack+0x20/0x28)
[   17.338858] [<c01a767c>] (dump_stack) from [<c00785d8>] (panic+0xa4/0x234)
[   17.345699] [<c0078538>] (panic) from [<c0588bcc>] (kernel_init+0xd0/0xf4)
[   17.352508]  r3:c3840000 r2:c3f580b0 r1:60000013 r0:c06bdacc
[   17.358141]  r7:00000000
[   17.360691] [<c0588afc>] (kernel_init) from [<c000fd58>] (ret_from_fork+0x14/0x3c)
[   17.368191]  r4:00000000 r3:c383e000
[   17.371774] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[   81.972066] random: nonblocking pool is initialized

yachen, 2020-11-30:
Hi, 請更新附件的 dts 試試. 謝謝.
[nuc980-iot-v1.0.dts.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/5614422/nuc980-iot-v1.0.dts.zip)

qiubin2016, 2020-11-30:
可以正常开机了，rootfs挂载成功。但是文件系统是只读。修改bootargs=noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init后，文件系统可读写。

yachen, 2020-12-01:
等一下會把更新推出來. 謝謝.

closed 2020-12-01 by yachen

swapnilpanwal, 2021-08-05:
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow�load

 SPL load main U-Boot from SPI NAND Flash! (Aug 05 2021 14:38:02)

U-Boot 2016.11 (Aug 05 2021 - 14:38:02 +0530)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x800000
SF: 8388608 bytes @ 0x200000 Read: OK
device 0 offset 0x40000, size 0x10000
SF: 65536 bytes @ 0x40000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2675304 Bytes = 2.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
=> 
**this error display**

u-boot env :
baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
setspi=sf probe 0 30000000
loadkernel=sf read 0x7fc0 0x200000 0x800000
loaddtb=sf read 1400000 0x40000 0x10000
bootcmd=run setspi;run loadkernel;run loaddtb;bootm 0x7fc0 - 0x1400000