OpenNuvoton/NUC980-linux-4.4.y #47: SPI NAND FLASH W25M02GVZEIG(256MB)启动时挂载文件系统失败

qiubin2016, 2020-12-03:
开发板：NuMaker-IIoT-NUC980，pcb丝印型号：NK-980IOT V1.0

默认使用SPI NAND FLASH是W25N01GVZE1G(128MB)，现在我换成W25M02GVZEIG(256MB)，启动内核挂载文件系统失败。从内核启动日志对比W25N01G和W25M02G的内核启动日志，发现“[    2.610374] nand: No NAND device found”,初步怀疑是内核没有识别W25M02G的flash。失败日志如下所示：
------------
SF: Detected W25M02GV with page size 2 KiB, erase size 128 KiB, total 256 MiB
SF: Device[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.4.179+ (qiub@ubuntu1604) (gcc version 4.8.4 (GCC) ) #29 PREEMPT Wed Dec 2 15:13:27 CST 2020
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: NUC980
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 56560K/65536K available (5670K kernel code, 374K rwdata, 1840K rodata, 176K init, 255K bss, 8976K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc075df9c   (7512 kB)
[    0.000000]       .init : 0xc075e000 - 0xc078a000   ( 176 kB)
[    0.000000]       .data : 0xc078a000 - 0xc07e7840   ( 375 kB)
[    0.000000]        .bss : 0xc07e7840 - 0xc0827654   ( 256 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:545
[    0.000000] clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
[    0.000033] sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
[    0.000716] Console: colour dummy device 80x30
[    0.185874] console [ttyS0] enabled
[    0.189358] Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
[    0.262758] pid_max: default: 32768 minimum: 301
[    0.267599] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.274266] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.283258] CPU: Testing write buffer coherency: ok
[    0.289133] Setting up static identity map for 0x8400 - 0x843c
[    0.299883] devtmpfs: initialized
[    0.323366] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.334066] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.340791] pinctrl core: initialized pinctrl subsystem
[    0.347799] NET: Registered protocol family 16
[    0.355116] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.725099] SCSI subsystem initialized
[    0.735458] usbcore: registered new interface driver usbfs
[    0.742041] usbcore: registered new interface driver hub
[    0.748266] usbcore: registered new device driver usb
[    0.758033] Advanced Linux Sound Architecture Driver Initialized.
[    0.775891] clocksource: Switched to clocksource nuc980-timer5
[    0.880049] NET: Registered protocol family 2
[    0.888008] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.894991] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.901824] TCP: Hash tables configured (established 1024 bind 1024)
[    0.908799] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.914666] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.921783] NET: Registered protocol family 1
[    0.927499] RPC: Registered named UNIX socket transport module.
[    0.933383] RPC: Registered udp transport module.
[    0.938258] RPC: Registered tcp transport module.
[    0.942924] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.950749] NetWinder Floating Point Emulator V0.97 (double precision)
[    1.110083] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 253)
[    1.118899] io scheduler noop registered (default)
[    1.124099] nuc980_gpio_probe - pdev = nuc980-gpio
[    1.141158] nuc980_dma_probe - pdev = nuc980-dma
[    1.166508] nuc980-dma nuc980-dma: NUC980 DMA ready
[... 110 lines cut ...]
[    7.852258] 010f           16384 ram15  (driver?)
[    7.856958] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    7.865191] CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.179+ #29
[    7.871149] Hardware name: NUC980
[    7.874433] Backtrace: 
[    7.876941] [<c0012e2c>] (dump_backtrace) from [<c0013018>] (show_stack+0x18/0x1c)
[    7.884449]  r6:c3ad5000 r5:c3ad5000 r4:c06c6d8c r3:00000000
[    7.890124] [<c0013000>] (show_stack) from [<c01a653c>] (dump_stack+0x20/0x28)
[    7.897308] [<c01a651c>] (dump_stack) from [<c0077480>] (panic+0xa4/0x234)
[    7.904174] [<c00773e0>] (panic) from [<c075f240>] (mount_block_root+0x1b8/0x270)
[    7.911591]  r3:00000001 r2:00000000 r1:c383fea0 r0:c06c6d8c
[    7.917224]  r7:c3fecaa0
[    7.919783] [<c075f088>] (mount_block_root) from [<c075f4f4>] (mount_root+0x100/0x12c)
[    7.927624]  r10:000000fe r9:c0780838 r8:c0788a9c r7:c0780830 r6:c0780858 r5:c078de88
[    7.935424]  r4:00000000
[    7.937974] [<c075f3f4>] (mount_root) from [<c075f680>] (prepare_namespace+0x160/0x1c0)
[    7.945908]  r8:c0788a9c r7:c0780830 r6:c0780858 r5:c0780858 r4:c07e7864
[    7.952624] [<c075f520>] (prepare_namespace) from [<c075ee40>] (kernel_init_freeable+0x184/0x1cc)
[    7.961424]  r6:c07e7840 r5:c07e7840 r4:00000007
[    7.966066] [<c075ecbc>] (kernel_init_freeable) from [<c058bb00>] (kernel_init+0x10/0xf4)
[    7.974166]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c058baf0
[    7.981966]  r4:00000000
[    7.984516] [<c058baf0>] (kernel_init) from [<c000fd58>] (ret_from_fork+0x14/0x3c)
[    7.992016]  r4:00000000 r3:c383e000
[    7.995591] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

yachen, 2020-12-03:
@w SPI NAND是 multi-die 的. 打_ menuconfig->device drivers->staging drivers->Winbind Multi-Die SPI NAND Support .

________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

qiubin2016, 2020-12-03:
我的内核版本4.4.179，没有这个选项“Winbind Multi-Die SPI NAND Support”，我看4.4.207有这个选项，看来我得更新内核版本

qiubin2016, 2020-12-07:
内核更新为：4.4.207可以挂载文件系统，但是遇到另外一个问题，NuWriterv1.16 Earse All，串口打印“bad_block:247
”，重新烧录u-boot、kernel、env、ubi文件系统后，第一次启动会报错，报错如下：
----------------
SPI NAND: Auto Detect
ID=[0xefab21]
BlockPerFlash = 1024, PagePerBlock = 64
PageSize = 2048, SpareArea = 64
QuadReadCmd    = 0x6b
ReadStatusCmd  = 0xff
WriteStatusCmd = 0xff
StatusValue    = 0xff
Dummybyte      = 0x1
Multi-Die      = 1

NAND ID=[0][0][0][0]
PowerOn setting 0x7f6000d7
BlockPerFlash=1023, PagePerBlock=64, PageSize=2048

eMMC1/SD1 Init Fail

Finish get INFO!!

bad_block:247
bad block = 247

NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
4-bit
finish SPI dow󋱡d

 SPL load main U-Boot from SPI NAND Flash! (Feb 20 2019 09:12:29)

U-Boot 2016.11 (Jun 20 2020 - 15:25:49 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  NAND Flash not found !
NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
0 MiB
SF: Detected W25M02GV with page size 2 KiB, erase size 128 KiB, total 256 MiB
In:    serial
Out:   serial
Err:   serial
Net:   emac
Hit any key to stop autoboot:  0 
SF: Detected W25M02GV with page size 2 KiB, erase size 128 KiB, total 256 MiB
device 0 offset 0x200000, size 0x800000
SF: 8388608 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2827288 Bytes = 2.7 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25M02GV with page size 2 KiB, erase size 128 KiB, total 256 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.207 (root@ubuntu) (gcc version 4.8.3 20140320 (prerelease) (Sourcery CodeBench Lite 2014.05-29) ) #69 PREEMPT Wed Dec 2 17:21:31 CST 2020
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: mem=64M noinitrd ubi.mtd=2 root=ubi0_0 rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mtdparts=nand0:0x200000@0x0(u-boot)ro,0x1400000@0x200000(kernel)ro,-(user) coherent_pool=2M
[... 636 lines cut ...]
[<c0012e60>] (dump_backtrace) from [<c001304c>] (show_stack+0x18/0x1c)
 r6:c31df038 r5:c3847df4 r4:c04c52d0 r3:00000000
[<c0013034>] (show_stack) from [<c01da61c>] (dump_stack+0x20/0x28)
[<c01da5fc>] (dump_stack) from [<c00767c8>] (panic+0xbc/0x24c)
[<c0076710>] (panic) from [<c0020164>] (do_exit+0x9a8/0x9dc)
 r3:c3848000 r2:00000000 r1:00000007 r0:c04c52d0
 r7:c383fe60
[<c001f7bc>] (do_exit) from [<c002116c>] (do_group_exit+0x44/0x108)
 r7:c3848000
[<c0021128>] (do_group_exit) from [<c002adac>] (get_signal+0x140/0x7cc)
 r4:00000007 r3:00002710
[<c002ac6c>] (get_signal) from [<c00126f0>] (do_signal+0x19c/0x3b4)
 r10:00112868 r9:c383e000 r8:00000000 r7:c383fed0 r6:00000000 r5:00000000
 r4:c383ffb0
[<c0012554>] (do_signal) from [<c0012adc>] (do_work_pending+0xb4/0xc4)
 r9:c383e000 r8:00000000 r7:0005317f r6:c383ffb0 r5:00000000 r4:c383e000
[<c0012a28>] (do_work_pending) from [<c000fd14>] (slow_work_pending+0xc/0x20)
 r6:ffffffff r5:60000010 r4:b6e043b4 r3:00000001
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007
------------------
此时断电重启，串口打印：
NUC980 IBR 20180813
Boot from SPI-NAND
No image in SPI, USB boot
这是W25M02G有坏块引起的？该如何处理?

qiubin2016, 2020-12-08:
------接上面打印信息----
/etc/init.d/rcS: line 35: /etc/init.d/rc.local: UBIFS error (ubi0:0 pid 1): ubifs_read_node: bad node type (255 but expected 2)
Invalid argument
UBIFS error (ubi0:0 pid 1): ubifs_read_node: bad node at LEB 78:11424, LEB mapping status 1
Not a node, first 24 bytes:
00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                          ........................
CPU: 0 PID: 1 Comm: init Not tainted 4.4.207 #69
Hardware name: NUC980
Backtrace: 
[<c0012e60>] (dump_backtrace) from [<c001304c>] (show_stack+0x18/0x1c)
 r6:0000004e r5:c38aa000 r4:00002ca0 r3:00000000
[<c0013034>] (show_stack) from [<c01da61c>] (dump_stack+0x20/0x28)
[<c01da5fc>] (dump_stack) from [<c015bbb8>] (ubifs_read_node+0x1d4/0x32c)
[<c015b9e4>] (ubifs_read_node) from [<c017abc8>] (ubifs_tnc_read_node+0x144/0x154)
 r10:00000000 r9:00000000 r8:c3189400 r7:00000002 r6:c3189400 r5:c38aa000
 r4:c30cc260
[<c017aa84>] (ubifs_tnc_read_node) from [<c015d0f0>] (tnc_read_node_nm+0x8c/0x220)
 r7:00000000 r6:c38aa000 r5:c3189400 r4:c30cc260
[<c015d064>] (tnc_read_node_nm) from [<c015f418>] (ubifs_tnc_locate+0x16c/0x19c)
 r7:00000000 r6:c38aa1e8 r5:c383fd70 r4:c38aa000
[<c015f2ac>] (ubifs_tnc_locate) from [<c015fad8>] (ubifs_tnc_lookup_nm+0x30/0x158)
 r10:00000076 r9:00000001 r8:4029ab98 r7:c34bc128 r6:c383fd70 r5:c3189400
 r4:c38aa000
[<c015faa8>] (ubifs_tnc_lookup_nm) from [<c015296c>] (ubifs_lookup+0x94/0x1f4)
 r9:00000001 r8:4029ab98 r7:c38aa000 r6:c3189400 r5:c34ba000 r4:c34bc110
[<c01528d8>] (ubifs_lookup) from [<c00bf1c0>] (lookup_real+0x28/0x54)
 r9:00000000 r8:c3814015 r7:c34b5880 r6:00000001 r5:c34b5880 r4:c34bc110
[<c00bf198>] (lookup_real) from [<c00c033c>] (__lookup_hash+0x40/0x48)
 r5:c34b5880 r4:00000011
[<c00c02fc>] (__lookup_hash) from [<c00c0520>] (walk_component+0x1dc/0x2bc)
 r5:00000000 r4:c383fe78
[<c00c0344>] (walk_component) from [<c00c1848>] (link_path_walk+0x11c/0x494)
 r8:c3814015 r7:c383fe78 r6:c383ff60 r5:ffffff9c r4:c383fe78
[<c00c172c>] (link_path_walk) from [<c00c1cac>] (path_lookupat+0x88/0x10c)
 r10:c31be880 r9:c383e000 r8:c383ff60 r7:00000001 r6:c383ff60 r5:ffffff9c
 r4:c383fe78
[<c00c1c24>] (path_lookupat) from [<c00c3fa0>] (filename_lookup+0x94/0xf0)
 r6:c383fe78 r5:ffffff9c r4:c3814000 r3:00000000
[<c00c3f0c>] (filename_lookup) from [<c00c40d0>] (user_path_at_empty+0x48/0x50)
 r8:c000fea8 r7:00000000 r6:c383ff60 r5:ffffff9c r4:00000001
[<c00c4088>] (user_path_at_empty) from [<c00b4cec>] (SyS_faccessat+0xa8/0x1ec)
 r6:000fdd44 r5:ffffff9c r4:00000001
[<c00b4c44>] (SyS_faccessat) from [<c00b4e4c>] (SyS_access+0x1c/0x20)
 r10:00000000 r9:c383e000 r8:c000fea8 r7:00000021 r6:00000000 r5:00000008
 r4:0000043c
[<c00b4e30>] (SyS_access) from [<c000fcc0>] (ret_fast_syscall+0x0/0x3c)
UBIFS error (ubi0:0 pid 1): ubifs_read_node: bad node type (255 but expected 2)
UBIFS error (ubi0:0 pid 1): ubifs_read_node: bad node at LEB 78:11424, LEB mapping status 1
Not a node, first 24 bytes:
00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                          ........................
CPU: 0 PID: 1 Comm: init Not tainted 4.4.207 #69
Hardware name: NUC980
Backtrace: 
[<c0012e60>] (dump_backtrace) from [<c001304c>] (show_stack+0x18/0x1c)
 r6:0000004e r5:c38aa000 r4:00002ca0 r3:00000000
[<c0013034>] (show_stack) from [<c01da61c>] (dump_stack+0x20/0x28)
[<c01da5fc>] (dump_stack) from [<c015bbb8>] (ubifs_read_node+0x1d4/0x32c)
[<c015b9e4>] (ubifs_read_node) from [<c017abc8>] (ubifs_tnc_read_node+0x144/0x154)
 r10:00000000 r9:00000000 r8:c3189400 r7:00000002 r6:c3189400 r5:c38aa000
 r4:c30cc260
[<c017aa84>] (ubifs_tnc_read_node) from [<c015d0f0>] (tnc_read_node_nm+0x8c/0x220)
 r7:00000000 r6:c38aa000 r5:c3189400 r4:c30cc260
[<c015d064>] (tnc_read_node_nm) from [<c015f418>] (ubifs_tnc_locate+0x16c/0x19c)
 r7:00000000 r6:c38aa1e8 r5:c383fd50 r4:c38aa000
[<c015f2ac>] (ubifs_tnc_locate) from [<c015fad8>] (ubifs_tnc_lookup_nm+0x30/0x158)
 r10:00000076 r9:00000001 r8:4029ab98 r7:c34bc238 r6:c383fd50 r5:c3189400
 r4:c38aa000
[<c015faa8>] (ubifs_tnc_lookup_nm) from [<c015296c>] (ubifs_lookup+0x94/0x1f4)
 r9:00000001 r8:4029ab98 r7:c38aa000 r6:c3189400 r5:c34ba000 r4:c34bc220
[<c01528d8>] (ubifs_lookup) from [<c00bf1c0>] (lookup_real+0x28/0x54)
[... 108 lines cut ...]
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007

CPU: 0 PID: 1 Comm: init Not tainted 4.4.207 #69
Hardware name: NUC980
Backtrace: 
[<c0012e60>] (dump_backtrace) from [<c001304c>] (show_stack+0x18/0x1c)
 r6:c31df038 r5:c3847df4 r4:c04c52d0 r3:00000000
[<c0013034>] (show_stack) from [<c01da61c>] (dump_stack+0x20/0x28)
[<c01da5fc>] (dump_stack) from [<c00767c8>] (panic+0xbc/0x24c)
[<c0076710>] (panic) from [<c0020164>] (do_exit+0x9a8/0x9dc)
 r3:c3848000 r2:00000000 r1:00000007 r0:c04c52d0
 r7:c383fe60
[<c001f7bc>] (do_exit) from [<c002116c>] (do_group_exit+0x44/0x108)
 r7:c3848000
[<c0021128>] (do_group_exit) from [<c002adac>] (get_signal+0x140/0x7cc)
 r4:00000007 r3:00002710
[<c002ac6c>] (get_signal) from [<c00126f0>] (do_signal+0x19c/0x3b4)
 r10:00112868 r9:c383e000 r8:00000000 r7:c383fed0 r6:00000000 r5:00000000
 r4:c383ffb0
[<c0012554>] (do_signal) from [<c0012adc>] (do_work_pending+0xb4/0xc4)
 r9:c383e000 r8:00000000 r7:0005317f r6:c383ffb0 r5:00000000 r4:c383e000
[<c0012a28>] (do_work_pending) from [<c000fd14>] (slow_work_pending+0xc/0x20)
 r6:ffffffff r5:60000010 r4:b6e043b4 r3:00000001
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007
------------------

yachen, 2020-12-09:
不會是壞塊影響. MTD 層遇到壞塊會略過不用.
我們現在有遇到使用 W25M02GV 不穩的情況, 可能是驅動還需要調整.
請問你手邊是否有 W25N02GV 這個型號的 SPI NAND? 這顆不像W25M02GV需要控制 die select. 使用上目前沒問題.
________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

qiubin2016, 2020-12-09:
好的  感谢  我们换W25N02GV试试看

closed 2021-07-30 by yachen