OpenNuvoton/NUC980-linux-4.4.y #48: NUC980-RTC Not working

ladecadence, 2020-12-12:
The RTC in my NUC980 board is not working. The RTC receives 3.3V (3.1V from a battery/diode circuit), and has a 32.768KHz crystal on the X32 pins with load capacitors.

The kernel is built with CONFIG_RTC_DRV_NUC980=y and the boot messages are:

```
...
nuc980-rtc nuc980-rtc: rtc core: registered nuc980-rtc as rtc0 
...
nuc980-rtc nuc980-rtc: setting system clock to 2005-01-01 00:00:00 UTC (1104537600) 
...
```

but the time is always the same, hwclock always returns:

```
~ # hwclock -r
Sat Jan  1 00:00:00 2005  0.000000 seconds
~ # 

```
and if I try to change it:
```
~ # date
Sat Jan  1 00:20:36 UTC 2005
~ # hwclock -w
~ # hwclock -r
Sat Jan  1 00:00:00 2005  0.000000 seconds
~ # 
```

Any Idea why this is happening?
Thanks.

yachen, 2020-12-17:
We can't reproduce this issue. Which board are you using?

Sincerely,

Yi-An Chen

________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

ladecadence, 2020-12-17:
Sorry, it looks like a hardware problem. I built another board and this one works perfectly.
Thanks.

closed 2020-12-17 by ladecadence