OpenNuvoton/NUC980-linux-4.4.y #50: NUC980-linux-4.4.207播放合成语音无声音

qiubin2016, 2021-01-14:
内核NUC980-linux-4.4.207：
1. mplayer播放音乐mp3声音正常。播放合成语音，无声音。（命令：mplayer -ac mad -ao alsa ./1.mp3）
2. aplay播放音乐wav声音正常。播放合成语音，无声音。（命令：aplay -c 2 -f S16_LE ./1.wav）
3. aplay播放101.pcm声音正常（命令：aplay -c 2 -f S16_LE ./101.pcm）
内核NUC980-linux-4.4.179：
1. mplayer播放音乐mp3声音正常。播放合成语音，正常。（命令：mplayer -ac mad -ao alsa ./1.mp3）
2. aplay播放音乐wav声音正常。播放合成语音，正常。（命令：aplay -c 2 -f S16_LE ./1.wav）
3. aplay播放101.pcm声音正常（命令：aplay -c 2 -f S16_LE ./101.pcm）

qiubin2016, 2021-01-17:
[image]

yachen, 2021-01-18:
Hi, 

> mplayer播放音乐mp3声音正常。播放合成语音，无声音

請問以這個例子來說, 播放 mp3 及合成语音的命令都一樣嗎?

Sincerely,

Yi-An Chen

qiubin2016, 2021-01-18:
你好，一样的命令，仅MP3文件不一样。如上面图片所示，我发现20200325的提交对NAU88C22驱动进行修改，验证发现修改前版本是播放歌曲MP3和合成语音MP3都正常。修改后，即从20200325开始的后续所有版本，播放歌曲MP3正常，播放合成语音MP3声音非常小，几乎听不到。
并且我把20200325修改NAU88C22驱动前一个版本（即20200323）的nau8822.c、nau8822.h、nuc980-audio.c覆盖目前最新版内核的这3个文件后，播放歌曲MP3和合成语音MP3都正常。

yachen, 2021-01-22:
Hi, 我們沒遇到這樣的情況. 你的設定是980 是走 master mode?
是否可以幫忙 dump 出有問題及沒問題時的 NAU8822寄存器讓我們比對? 謝謝.

qiubin2016, 2021-01-25:
1. 980内核设置是master mode
2. dump寄存器需要等一段时间
3. 如果方便可以提供email，我把正常和不正常音频文件发过去。

closed 2021-07-30 by yachen