OpenNuvoton/NUC980-linux-4.4.y #53: I2c clock speed setting

piovemas, 2021-02-23:
Hi,

Is it possible to change the I2c clock speed (SCL3 signal) ?
If yes, how ?

Thanks

yachen, 2021-02-24:
Hi,

Either modify the bus_freq in arch/arm/mach-nuc980/dev.c or in nuc980.dtsi if you enabled device tree support.

piovemas, 2021-02-24:
Hy,

I set dev.c with:
".bus_freq = 400000"

and I obtain about 330 Khz. Why this difference ? Is it possible to increase this value ?
Thanks

yachen, 2021-02-25:
Consider the fraction being omitted, you should get a I2C clock rate slightly faster than 400k. Can you check the value been set to CLKDIV register?
```
ret = clk_get_rate(i2c->clk)/(busfreq * 4) - 1;
writel(ret & 0xffff, i2c->regs + CLKDIV);
```

piovemas, 2021-02-25:
Hi,
ret=92
and reading CLKDIV register, I get 0x5C.

yachen, 2021-02-26:
There's some problem with this line, it's now how the divider works is this I2C.
`ret = clk_get_rate(i2c->clk)/(busfreq * 4) - 1;`
And should be update to, that will get a 93 instead of 92.
`ret = (clk_get_rate(i2c->clk) * 10) / (busfreq * 4) + 5) / 10 - 1);`
 But ret = 92 still get a 403kHz I2C clock instead of 330kHz.

piovemas, 2021-02-26:
Hi,
Inside NUC980_TechnicalReferenceManual_EN_Rev1.00.pdf about I2C_CLKDIV register there is:
Data Baud Rate of I2C = (system clock) / (4x (I2C_CLKDIV+1))
and inside i2c-nuc980-p3.c the inverse formula is correct.

yachen, 2021-03-02:
Not sure what do you mean by "inverse formula is correct".
Following code can round ret to 93 which makes the output closer to target clock rate 400kHz.
`ret = (clk_get_rate(i2c->clk) * 10) / (busfreq * 4) + 5) / 10 - 1);`

piovemas, 2021-03-02:
Hi,
also with ret = 93 I get about 330Khz on SCL. Could you verify with oscilloscope ?

closed 2021-07-30 by yachen