OpenNuvoton/NUC980-linux-4.4.y #54: PG15 GPIO status at power on

piovemas, 2021-02-24:
Hi,
I'm using NK-980IOT V 1.0 and I try to set PG15 (LED4) GPIO without success.
Instead I can set PB8 ( (LED3) and PB13 (LED5).
I see that after power on, when linux shows the prompt, the PG15 is at 3,3V. Instead PB8 and PB13 are about 2V.

Who force PB13 GPIO ?
How can I set it ?

Thanks

yachen, 2021-02-25:
PG.15 is JTAG pin by default. You can add following code in an init function in arch/arm/mach-nuc980/mach-nuc980.c to change JTAG pins back to GPIO function while booting up Linux kernel

`__raw_writel((__raw_readl(REG_MFP_GPG_H) & ~0xFFFFF000, REG_MFP_GPG_H);`

piovemas, 2021-02-25:
Hi,

I inserted your line inside "void __init nuc980_map_io(void)" and it works.
Thanks

piovemas, 2021-02-25:
Hi,

I seen that there is a nuc980_mfp_set_port_g() function to set alternate function. Is it possible in some way to call it from user space ?
Thanks

yachen, 2021-02-26:
Currently there's no calling path that allows user space to call this function.

piovemas, 2021-02-26:
Hi,
It could be usefull to access these functions from user space also because they use the function that lock interrupt between read and write of registers.

closed 2021-12-08 by ychuang3