OpenNuvoton/NUC980-linux-4.4.y #55: Slow I2c

piovemas, 2021-02-24:
Hi,
I'm using NK-980IOT V 1.0 and I'm testing I2c on /dev/i2c-3.
For our future board with NUC980, I2c is fundamental and it must works very fast.
I send an image where some timing is not explanble.
I can't expand more because the last I2c stop condition is to later.

[image]

Linux is running only my application. My application do a I2c write command and then when write is end, an I2c read command.
Starting from the left of image at time 0uSec there is a I2c write and then at time 600uSec an  I2c read. 
T1 is a trigger that I set before calling the I2c write function and reset before calling the I2c read. Could you explain me the time from the first stop (P) and when T1 is reset ? Is it the driver that wait for something ?

Then after the I2c write command, there is a lot of time (about 1000uSec) between the NACK (N) from slave and the stop (P). Could you explain me why so much time ?

Other things that I can't explain are where in the oscillogram there is no I2c bus activity.

Thanks

yachen, 2021-02-25:
Consider the fraction being omitted, you should get a I2C clock rate slightly faster than 400k. Can you check the value been set to CLKDIV register?

```
ret = clk_get_rate(i2c->clk)/(busfreq * 4) - 1;
writel(ret & 0xffff, i2c->regs + CLKDIV);
```

piovemas, 2021-02-26:
Hi,
I want to add a comparation to I2c bit banged. Here the oscillogram:
[image]

As you can see the entire command write+read is executed in only 474 uSec also if SCL speed is only about 115 Khz.
Thanks