OpenNuvoton/NUC980-linux-4.4.y #56: SPI problem

piovemas, 2021-02-26:
Hi,
I'm trying SPI, so I configured buildroot with:
<*>Nuvoton NUC980 Series SPI Port 0
      SPI0 IO port selection (Port D)  ---> (X) Port D
      SPI0 TX/RX by PDMA or not (No PDMA)  ---> (X) No PDMA

but nothing appear in /dev.
I have these errors:
nuc980-spi0 nuc980-spi0.0: No platform data supplied
nuc980-spi0: probe of nuc980-spi0.0 failed with error -2

Could you help me ?
Thanks

yachen, 2021-03-02:
You need to specify device connected on SPI0 interface in arch/arm/mach-nuc980/dev.c

piovemas, 2021-03-02:
Hi,
I need to control a custom device from user space using function like ioctl and structure "struct spi_ioc_transfer" like in this example https://github.com/sparkfun/pcDuino/blob/master/examples/SPI/spi_test.cpp.
Could you help me ?

Thanks

yachen, 2021-03-03:
Hi,

There's a sample code utilize spidev to access SPI interface in our BSP as well. https://github.com/OpenNuvoton/NUC980_Linux_Applications/tree/master/demos/spi

The usage of spidev is documented in Linux kernel.
https://github.com/OpenNuvoton/NUC980-linux-4.4.y/blob/master/Documentation/spi/spidev

Sincerely,

Yi-An Chen

piovemas, 2021-03-03:
Hi,
I solved the problem but only setting:
SPI0 TX/RX by PDMA or not (No PDMA) ---> (X) No PDMA

Is it true that PDMA is not supported ?

Another question: the SPI chip select is only activable from nuc980_slave_select function ? Is it not possible to hardware  auto activate it near sending SCK ?
Thanks

yachen, 2021-03-04:
> Is it true that PDMA is not supported ?

Nope, it's supported.

>  Is it not possible to hardware auto activate it near sending SCK ?

It's possible by setting AUTOSS bit in SPIx_SSCTL register. But SPI will deassert CS pin while FIFO is empty. And sometimes this is an unwanted behavior.

yachen, 2021-03-11:
Hi,

Try deselect SPI0 enable pin for the second chip select.

Sincerely,

Yi-An Chen
________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

sushantdasputepatil, 2021-03-11:
Hi Yi-An Chen,

I have already tried that, no effect.

Sincerely,
Sushant Daspute

cwweng-nuvoton, 2021-03-11:
Hi,

If you want to use PC5(SPI0_SS0) as a chip select pin, please follow below 2 steps.
1. Update arch/arm/mach-nuc980/dev.c
Update .chip_select to 0 (use SPI0_SS0) 
[image]

2. Select SPI0 IO port to "Port C" in menuconfig
[image]

cwweng-nuvoton, 2021-03-11:
As for Non-OS BSP, I update the sample code to use SPI0. The SPI0_SS0(PC5) works fine.
[image]

Attached is the updated sample code that use SPI0_SS0(PC5).

[SPI_PDMA.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6122281/SPI_PDMA.zip)

sushantdasputepatil, 2021-03-11:
Is above information applicable to Nuvoton Chili Board? I have tried above thing but after that there is no spidev1.1 in /dev directory.

closed 2021-07-30 by yachen