OpenNuvoton/NUC980-linux-4.4.y #58: SPI Problem

tkrelcon, 2021-03-24:
Hi,

While trying to upload code and boot its showing error in NuWriter. I am using NUC980DK61YC evaluation board.  

NUC980 IBR 20180813
Boot from USB
DDR-OK
=======================================
Run firmware code
CPU: 300MHz, DDR: 300MHz, SYS: 300MHz, PCLK: 150MHz
Parse NuWriter command line
=======================================
SPI NOR ID=0x0000ffff  _spi_type =0
SPI NOR ID=0x0000ffff  _spi_type =0
**Error! Read SPI ID(0xffff)**

SPI NAND: Auto Detect
ID=[0xefaa21]
BlockPerFlash = 1024, PagePerBlock = 64
PageSize = 2048, SpareArea = 64
QuadReadCmd    = 0x6b
ReadStatusCmd  = 0xff
WriteStatusCmd = 0xff
StatusValue    = 0xff
Dummybyte      = 0x1
Multi-Die      = 0

NAND ID=[f7][f7][f7][f7]
PowerOn setting 0x7f6003d4
BlockPerFlash=1023, PagePerBlock=64, PageSize=2048

eMMC0/SD0 Init Fail

Finish get INFO!!

SPI NOR ID=0x0000ffff  _spi_type =0
SPI NOR ID=0x0000ffff  _spi_type =0
 96
NUC980 IBR 20180813
Boot from USB

yachen, 2021-03-24:
Nothing wrong with the log. NuWriter found the Winbond W25N01G SPI NAND flash with ID 0xefaa21.

tkrelcon, 2021-03-24:
But not able to upload image in SPI NOR and NAND. Upload operation is not getting completed. ITs showing error message. 
If i try to boot it showing like this.
NUC980 IBR 20180813
Boot from NAND
Err-R/B#
Err-ID
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
NUC980 IBR 20180813
Boot from NAND
Err-R/B#
Err-ID

NUC980 IBR 20180813
Boot from NAND
Err-R/B#
Err-ID

NUC980 IBR 20180813
Boot from SPI-NOR
DDR-OK
finish SPI dow▒
NUC980 IBR 20180813
Boot from USB

NUC980 IBR 20180813
Boot from NAND
Err-R/B#
Err-ID

NUC980 IBR 20180813
Boot from SPI-NOR
DDR-OK
finish SPI dow▒

yachen, 2021-03-25:
Your board has a SPI NAND connect with QSPI0 interface, so it it cannot boot from SPI NOR.
Both IBR and NuWriter failed to identify a NAND flash on your board, please check if SW3/4 is set to ON.

tkrelcon, 2021-03-25:
Ok. Thanks but it was not booting from NAND flash also. Switch PG 8& 9 Both are ON. Still let me try to again with suggested changes.

tkrelcon, 2021-03-25:
I tried to boot from SPI NAND Still system not able to boot up.. What to do now?

NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow▒c{
NAND boot!       #S

NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
4-bit
finish SPI dow▒c{
NAND boot!       #S

tkrelcon, 2021-03-25:
IF trying use DDR than also it was not able to boot..

finish SDRA

Only this message is coming..

yachen, 2021-03-25:
> I tried to boot from SPI NAND Still system not able to boot up.. What to do now?
> 
> NUC980 IBR 20180813
> Boot from SPI-NAND
> DDR-OK
> finish SPI dow▒c{
> NAND boot! #S
> 
> NUC980 IBR 20180813
> Boot from SPI-NAND
> DDR-OK
> 4-bit
> finish SPI dow▒c{
> NAND boot! #S

What images did you program into SPI NAND?

yachen, 2021-03-25:
> IF trying use DDR than also it was not able to boot..
> 
> finish SDRA
> 
> Only this message is coming..

What image did you load and execute in DDR, and what's the execution address?

tkrelcon, 2021-03-25:
> > I tried to boot from SPI NAND Still system not able to boot up.. What to do now?
> > NUC980 IBR 20180813
> > Boot from SPI-NAND
> > DDR-OK
> > finish SPI dow▒c{
> > NAND boot! #S
> > NUC980 IBR 20180813
> > Boot from SPI-NAND
> > DDR-OK
> > 4-bit
> > finish SPI dow▒c{
> > NAND boot! #S
> 
> What images did you program into SPI NAND?

uboot.spl.bin, uboot.bin,980uimage,env

tkrelcon, 2021-03-25:
> > IF trying use DDR than also it was not able to boot..
> > finish SDRA
> > Only this message is coming..
> 
> What image did you load and execute in DDR, and what's the execution address?

image file.

tkrelcon, 2021-03-25:
> > > IF trying use DDR than also it was not able to boot..
> > > finish SDRA
> > > Only this message is coming..
> > 
> > 
> > What image did you load and execute in DDR, and what's the execution address?
> 
> image file.

80000

tkrelcon, 2021-03-25:
> uboot.spl.bin, uboot.bin,980uimage,env

[Images.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6203609/Images.zip)

yachen, 2021-03-25:
> [Images.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6203609/Images.zip)

The uboot image you used doesn't support SPI NAND flash. Please rebuild uboot with a configuration that supports SPI NAND, for example nuc980_iot_defconfig.

yachen, 2021-03-25:
> > > > IF trying use DDR than also it was not able to boot..
> > > > finish SDRA
> > > > Only this message is coming..
> > > 
> > > 
> > > What image did you load and execute in DDR, and what's the execution address?
> > 
> > 
> > image file.
> 
> 80000

What image file were you testing? NUC980's Linux kernel should load to 0x8000 instead of 0x80000

tkrelcon, 2021-03-25:
Ok I Got the solution. Thank you sir.

closed 2021-03-25 by tkrelcon