OpenNuvoton/NUC980-linux-4.4.y #62: GPIO 边沿触发中断，edge 设置为 rising 发现上升沿和下降沿都会触发中断

loogg, 2021-05-21:
我使用的引脚为 PD.8，下面是我在内核中增加的打印代码。

1.  在 `static void nuc980_irq_demux_intgroupD(struct irq_desc *desc)` 增加打印信息
```C
static void nuc980_irq_demux_intgroupD(struct irq_desc *desc)
{
	unsigned int j, isr;
	unsigned int flag=0;
	ENTRY();
	isr = __raw_readl(REG_GPIOD_INTSRC);
	__raw_writel(isr,REG_GPIOD_INTSRC);
	for (j = 0; j < 16; j++) {
		if (isr  & 0x1) {
			flag=1;
			printk("occur interrupt\r\n"); //打印
			generic_handle_irq(IRQ_GPIO_START +
			                   3 * 0x20 + j);
		}
		isr = isr >> 1;
	}
	if(flag==0)
		__raw_writel(0x01, REG_AIC_EOIS);
	LEAVE();
}
```

发现下降沿和上升沿都会打印。

2. 在 `static int nuc980_irq_gpio_type(struct irq_data *d, unsigned int type)` 结束前增加了打印信息
```C
unsigned int int_type = __raw_readl((volatile unsigned int *)(Port[port] + 0x18));
unsigned int int_en = __raw_readl((volatile unsigned int *)(Port[port] + 0x1C));

printk("port:%d, num:%d, type:0x%08X, en:0x%08X\r\n", port, num, int_type, int_en);
```

打印结果如下，我查看了手测发现寄存器数据是正确的
```
port:3, num:8, now set edge rising
port:3, num:8, type:0x00000000, en:0x01000000
```

3. 应用测试程序如下, 上升沿和下降沿都会打印信息，并且我查看过 edge 文件的内容，确实是rising。 gpio的几个封装函数，我放在最下面。

```C
void key_init(void)
{
    gpio_unexport(104);
    gpio_export(104);
    gpio_direction(104, 0); // 输入
    gpio_edge(104, 1); //rising

    int fd = open("/sys/class/gpio/gpio104/value", O_RDWR | O_NONBLOCK);
    if(fd < 0)
        return;

    int flags = fcntl(fd, F_GETFD);
    flags |= FD_CLOEXEC;
    fcntl(fd, F_SETFD, flags);

    char buff[10];
    lseek(fd, 0, SEEK_SET);
    read(fd, buff, sizeof(buff));
    
    // select使用
    fd_set exceptset;

    while(1)
    {
        FD_ZERO(&exceptset);

        FD_SET(fd, &exceptset);

        int rc = select(fd + 1, NULL, NULL, &exceptset, NULL);
        if(rc == -1)
        {
            if(errno == EINTR)
            {
                continue;
            }

            break;
        }

        if(rc > 0)
        {
            if(FD_ISSET(fd, &exceptset))
            {
                lseek(fd, 0, SEEK_SET);
                read(fd, buff, sizeof(buff));
                
                printf("exceptset, value:%s\r\n", buff);
            }  
        }

        printf("next\r\n");
    }
    
    close(fd);
}
```

gpio封装函数
```C
#include <stdio.h>
#include <stdlib.h>

int gpio_export(int pin)
{
    FILE *fp = fopen("/sys/class/gpio/export", "w");
    if(fp == NULL)
        return -1;
    
    fprintf(fp, "%d", pin);
    fclose(fp);

    return 0;
}

int gpio_unexport(int pin)
{
    FILE *fp = fopen("/sys/class/gpio/unexport", "w");
    if(fp == NULL)
        return -1;
    
    fprintf(fp, "%d", pin);
    fclose(fp);

    return 0;
}

//dir: 0-->in, 1-->out
int gpio_direction(int pin, int dir)
{
    char path[64] = "";
    snprintf(path, sizeof(path), "/sys/class/gpio/gpio%d/direction", pin);
    FILE *fp = fopen(path, "w");
    if(fp == NULL)
        return -1;
    
    if(dir)
        fprintf(fp, "out");
    else
        fprintf(fp, "in");
    fclose(fp);

    return 0;
}

int gpio_write(int pin, int value)
{
    char path[64] = "";
    snprintf(path, sizeof(path), "/sys/class/gpio/gpio%d/value", pin);
    FILE *fp = fopen(path, "w");
    if(fp == NULL)
        return -1;

    if(value)
        fprintf(fp, "1");
    else
        fprintf(fp, "0");
    fclose(fp);

    return 0;
}

int gpio_read(int pin)
{
    char path[64] = "";
    snprintf(path, sizeof(path), "/sys/class/gpio/gpio%d/value", pin);
    FILE *fp = fopen(path, "rb");
    if(fp == NULL)
        return -1;
    
    char buffer[10] = "";
    fread(buffer, sizeof(char), sizeof(buffer) - 1, fp);
    fclose(fp);

    return atoi(buffer);
}

// 0-->none, 1-->rising, 2-->falling, 3-->both
int gpio_edge(int pin, int edge)
{
    char path[64] = "";
    snprintf(path, sizeof(path), "/sys/class/gpio/gpio%d/edge", pin);
    FILE *fp = fopen(path, "w");
    if(fp == NULL)
        return -1;
    
    switch(edge)
    {
        case 1:
            fprintf(fp, "rising");
        break;

        case 2:
            fprintf(fp, "falling");
        break;

        case 3:
            fprintf(fp, "both");
        break;

        default:
            fprintf(fp, "none");
        break;
    }
    fclose(fp);

    return 0;
}
```

schung1218, 2021-05-21:
@loogg 
可以幫忙檢查一下 /proc/interrupts  行為是對的嗎?

loogg, 2021-05-21:
```
/proc # cat interrupts
           CPU0
 15:       6281         - Edge      nuc980rtc
 19:       2900         - Edge      nuc980-emac0
 21:        165         - Edge      nuc980-emac0
 23:          0         - Edge      ehci_hcd:usb1
 24:          0         - Edge      ohci_hcd:usb2
 25:       8500         - Edge      PDMA0
 26:        776         - Edge      PDMA1
 27:          0         - Edge      mmc0
 32:    1210350         - Edge      nuc980-timer4
 36:        407         - Edge      ttyS0
 47:          0         - Edge      nuc980-i2c2
360:         28  GPIO-IRQ Edge      gpiolib

```

schung1218, 2021-05-21:
@loogg  sorry!沒有說清楚! 想要知道下降觸發的時候, /proc/interrupt 裡面對應的中斷數也會增加1嗎?

loogg, 2021-05-21:
ok，我现在在路上，等会到家试一下

loogg, 2021-05-21:
hello，我发现 按下和松开后，GPIO_IRQ 前面的数字增加了 2

loogg, 2021-05-21:
> hello，我发现 按下和松开后，GPIO_IRQ 前面的数字增加了 2

也就是按下和松开都增加 1

schung1218, 2021-05-24:
Hi @loogg 
我使用下面三個指令去測試沒有遇到你說的問題
echo "104" > /sys/class/gpio/export
echo "in" > /sys/class/gpio/gpio104/direction
echo "rising" > /sys/class/gpio/gpio104/edge
cat /proc/interrupts

loogg, 2021-05-27:
OK，应该是硬件电路的问题，按键这有电容电路。我测试单纯IO的时候就没这问题。谢谢

closed 2021-05-27 by loogg