OpenNuvoton/NUC980-linux-4.4.y #64: 串口设置 1200 波特率结果变成了 9600 波特率

loogg, 2021-05-27:
Hi，我现在遇到一个问题，我在将串口 4 和串口 9 设置为 1200 波特率后，发现 1200 无法通信成功，9600 才能通信上。
下面是我的初始化代码，我打印了speed，我确认是 1200。
```C
int serial_init(const char *device,
                int baud, char parity, int data_bit,
                int stop_bit, struct termios *old_tios)
{

    struct termios tios;
    speed_t speed;
    int flags;

    /* The O_NOCTTY flag tells UNIX that this program doesn't want
       to be the "controlling terminal" for that port. If you
       don't specify this then any input (such as keyboard abort
       signals and so forth) will affect your process

       Timeouts are ignored in canonical input mode or when the
       NDELAY option is set on the file via open or fcntl */
    flags = O_RDWR | O_NOCTTY | O_NDELAY | O_EXCL;
#ifdef O_CLOEXEC
    flags |= O_CLOEXEC;
#endif

    int s = open(device, flags);
    if (s == -1) {
        LOG_E("ERROR Can't open the device %s (%s)", device, strerror(errno));
        return -1;
    }

    flags = fcntl(s, F_GETFL, 0);
    flags |= O_NONBLOCK;
    fcntl(s, F_SETFL, flags);

    flags = fcntl(s, F_GETFD);
    flags |= FD_CLOEXEC;
    fcntl(s, F_SETFD, flags);

    /* Save */
    tcgetattr(s, old_tios);

    memset(&tios, 0, sizeof(struct termios));

    /* C_ISPEED     Input baud (new interface)
       C_OSPEED     Output baud (new interface)
    */
    switch (baud) {
    case 110:
        speed = B110;
        break;
    case 300:
        speed = B300;
        break;
    case 600:
        speed = B600;
        break;
    case 1200:
        speed = B1200;
        break;
    case 2400:
        speed = B2400;
        break;
    case 4800:
        speed = B4800;
        break;
    case 9600:
        speed = B9600;
        break;
    case 19200:
        speed = B19200;
[... 239 lines cut ...]
       driver you need exactly N bytes and any read call will
       return 0 or N bytes. However, the timeout only applies to
       the first character read, so if for some reason the driver
       misses one character inside the N byte packet then the read
       call could block forever waiting for additional input
       characters.

       VTIME specifies the amount of time to wait for incoming
       characters in tenths of seconds. If VTIME is set to 0 (the
       default), reads will block (wait) indefinitely unless the
       NDELAY option is set on the port with open or fcntl.
    */
    /* Unused because we use open with the NDELAY option */
    tios.c_cc[VMIN] = 0;
    tios.c_cc[VTIME] = 0;

    if (tcsetattr(s, TCSANOW, &tios) < 0) {
        close(s);
        s = -1;
        return -1;
    }

    return s;
}
```

yachen, 2021-05-27:
Hi, 現在 driver 裡的 UART clock 是吃 150MHz 的 PLL. 除不到那麼低速. 要跑低速的 port (以 PORT 4 為例) 驅動中改個 clock source 試試.

```C
	if(up->port.line == 4) {
		clk = clk_get(NULL, "uart4");
		clk_prepare(clk);
		clk_enable(clk);

		clk = clk_get(NULL, "uart4_eclk");
		clk_prepare(clk);
		clk_enable(clk);

		clkmux = clk_get(NULL, "uart4_eclk_mux");
		upll_clk = clk_get(NULL, "xin");
		clk_set_parent(clkmux, upll_clk);

		clk = clk_get(NULL, "uart4_eclk_div");

		clk_set_rate(clk, 12000000);
		up->port.uartclk = clk_get_rate(clk);
	}
```

loogg, 2021-05-27:
好的，谢谢

closed 2021-07-30 by yachen