OpenNuvoton/NUC980-linux-4.4.y #65: 请教：nuc980 系统休眠后，怎样唤醒？

redworlf007, 2021-06-17:
1、nuc980 用[*] Suspend to RAM and standby     $ echo mem > /sys/power/state  休眠后，怎么唤醒？
2、用alarm可以唤醒么？
3、内核配置中的，opportunistic sleep  有demo示例么？
4、内核配置中的，user space wakeup sources interface  有demo例子么？
5、我想实现的应用程序功能是：应用程序串口通讯采集数据后，cpu休眠十分钟，然后再采集数据，如此反复，应该用哪种唤醒方式好么？电池供电的。 谢谢。

yachen, 2021-06-17:
Hi, menuconfig->device drivers->real time clock->enable RTC wake-up function 要打開, 使用 RTC 的 alarm 功能就可以喚醒系統.

redworlf007, 2021-06-17:
好的，十分感谢，这个我搞好了，再请教下，如果我要做按键唤醒，除了要把按键设置为中断外，还要打开RTC的哪些配置么？

yachen, 2021-06-17:
按鍵不是用 GPIO? menuconfig->Device Drivers->GPIO support->Memory mapped GPIO drivers->NUC980 external I/O wake-up support 要開.

redworlf007, 2021-06-17:
按键是用gpio

redworlf007, 2021-06-17:
我用demo里面的rtc例子，出问题的时候，retval<0，然后就挂了。。。
retval = ioctl(fd, RTC_ALM_SET, &rtc_tm);
        if (retval <0) {
                printf("ioctl RTC_ALM_SET  faild!!!\n");
                close(fd);
				exit(0);
        }

yachen, 2021-06-17:
Hi, 請幫忙看一下 ioctl 返回什麼錯誤碼.

redworlf007, 2021-06-17:
好，我改下代码，跑一晚上看看。

redworlf007, 2021-06-17:
ioctl=-1
ioctl RTC_ALM_SET  faild!!!
你看看，这个返回值。

redworlf007, 2021-06-17:
******** RTC Test Demo ***********
Test second alarm(hour,min,sec)..
**************************************
Select :
read current time is: 2005-01-01 06:59:32

ioctl=0
read current alarm time is: 2005-01-01 06:59:42

Waiting 10 seconds for alarm...

PM: Syncing filesystems ... done.
Freezing user space processes ... (elapsed 0.001 seconds) done.
Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Suspending console(s) (use no_console_suspend to debug)
PM: suspend of devices complete after 33.783 msecs
PM: late suspend of devices complete after 3.491 msecs
PM: noirq suspend of devices complete after 3.466 msecs
Wake up source: 00000000  00000080
PM: noirq resume of devices complete after 2.716 msecs
PM: early resume of devices complete after 2.908 msecs
PM: resume of devices complete after 63.083 msecs
Restarting tasks ...

*Alarm interrupt come on,the func running!*
done.
Test second alarm(hour,min,sec) test OK!

******** RTC Test Demo ***********
Test second alarm(hour,min,sec)..
**************************************
Select :
read current time is: 2005-01-01 06:59:50

ioctl=-1
ioctl RTC_ALM_SET  faild!!!
PM: Syncing filesystems ... done.
Freezing user space processes ... (elapsed 0.001 seconds) done.
Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Suspending console(s) (use no_console_suspend to debug)

redworlf007, 2021-06-18:
hello，上班了么？

redworlf007, 2021-06-18:
大哥，帮忙给诊断下啥原因，我快要被老板杀了祭天了。。。救命呀

yachen, 2021-06-18:
Hi, 請確認一下出問題時 32k 晶振還有沒有在動. 像是等不到 access enable.
```c
        err = check_rtc_access_enable(rtc);
        if (IS_ERR(err))
                return PTR_ERR(err);
```

redworlf007, 2021-06-18:
==我看看access enable在哪个文件里面

yachen, 2021-06-18:
drivers/rtc/rtc-nuc980.c

redworlf007, 2021-06-18:
咋样能上传文件呢？我把测试代码发给你跑跑看。

redworlf007, 2021-06-18:
SuspendToRAM 程序如下：是在rtc demo基础上改的。
void alarm_func()
{
	//sleep(1);
	printf("\n\n*Alarm interrupt come on,the func running!*\n");
	system("./Bee &");	
    //system("ps au &"); 		
}

int main(void)
{

        int i, fd, retval, irqcount = 0;
        unsigned long tmp, data;
        struct rtc_time rtc_tm;
        int scale_type = HR24;
        struct rtc_wkalrm rtc_alarmtm;
        unsigned int select;
        fd = open ("/dev/rtc0", O_RDONLY);
        fd_all=fd;
        if (fd<0) {
                printf("open rtc faild!!!\n");
                return 0;
        }
        signal (SIGTERM, ctrl_c_signal); /* for the TERM signal.. */
        signal (SIGINT, ctrl_c_signal); /* for the CTRL+C signal.. */

        printf("\n******** RTC Test Demo ***********\n");
        printf("Test second alarm(hour,min,sec)..\n");
        printf("**************************************\n");
        printf("Select : \n");

       
        setup_alarm(fd,10,alarm_func);
              
        return 0;

} /* end main */

redworlf007, 2021-06-18:
Bee 程序代码如下：
int main(void)
{
	//system("ps au &");	
        initGpio(8);
	setGpioDirection(8, "out");
	for (uint16_t i = 0; i < 20; i++)
		{
		setGpioValueToHigh(8);
		usleep(50000);
		setGpioValueToLow(8);
		usleep(50000);
		}
	
	system("./SuspendToRAM &");	
	sleep(1);
	system("echo mem > /sys/power/state");	

	return 0;
}

redworlf007, 2021-06-18:
开机执行：./Bee

redworlf007, 2021-06-18:
setup_alarm函数里面：
/*
*setup_alarm():set up the sec,func
*/
void setup_alarm(int fd,int sec,void(*func)())
{

        int i,retval;
        unsigned long tmp, data;
        struct rtc_time rtc_tm;
        int scale_type = HR24;
        /*read current time*/

        retval=ioctl(fd,RTC_RD_TIME,&rtc_tm);
        if (retval <0) {
                printf("ioctl RTC_RD_TIME  faild!!!");
				close(fd);
				exit(0);
        }

        printf("read current time is: %04d-%02d-%02d %02d:%02d:%02d\n\n",
               rtc_tm.tm_year + 1900,
               rtc_tm.tm_mon + 1,
               rtc_tm.tm_mday,
               rtc_tm.tm_hour,
               rtc_tm.tm_min,
               rtc_tm.tm_sec);

        rtc_tm.tm_sec = rtc_tm.tm_sec+sec;//alarm to 5 sec in the future
        if (rtc_tm.tm_sec>60) {
                rtc_tm.tm_sec=rtc_tm.tm_sec-60;
                rtc_tm.tm_min=rtc_tm.tm_min+1;
        }
        //rtc_alarmtm.enabled=0;
        //rtc_alarmtm.time=rtc_tm;

        /* Set the alarm to 5 sec in the future */

        retval = ioctl(fd, RTC_ALM_SET, &rtc_tm);
	printf("ioctl=%d\n",retval);
		
        if (retval <0) {
                printf("1-ioctl RTC_ALM_SET  faild!!!\n");
               close(fd);
              exit(0);
				}
        }
        
        /* Read the current alarm settings */

        retval = ioctl(fd, RTC_ALM_READ, &rtc_tm);
        if (retval <0) {
                printf("ioctl  RTC_ALM_READ faild!!!\n");
                close(fd);
		exit(0);
        }

        printf("read current alarm time is: %04d-%02d-%02d %02d:%02d:%02d\n\n",
               rtc_tm.tm_year + 1900,
               rtc_tm.tm_mon + 1,
               rtc_tm.tm_mday,
               rtc_tm.tm_hour,
               rtc_tm.tm_min,
               rtc_tm.tm_sec);

        /* Enable alarm interrupts */

        retval = ioctl(fd, RTC_AIE_ON, 0);
        if (retval <0) {
                printf("ioctl  RTC_AIE_ON faild!!!\n");
                close(fd);
				exit(0);
        }

        fprintf(stderr, "Waiting %d seconds for alarm...\n\n",sec);
        fflush(stderr);
        /* This blocks until the alarm ring causes an interrupt */
        retval = read(fd, &data, sizeof(unsigned long));
        if (retval >0) {
			    close(fd);
                func();
        } else {
                printf("!!!alarm faild!!!\n");
                close(fd);
				exit(0);
        }
       
        /**********************************/
        fd = open ("/dev/rtc0", O_RDONLY);
        /**********************************/

         /* Disable alarm interrupts */
        retval = ioctl(fd, RTC_AIE_OFF, 0);
        if (retval == -1) {
                printf("ioctl RTC_AIE_OFF faild!!!\n");
                close(fd);
				exit(0);
        }
		close(fd);
		
        printf("Test second alarm(hour,min,sec) test OK!\n");
		exit(0);
}

redworlf007, 2021-06-18:
32.768的晶振，我示波器测了，有波形，硬件上应该是正常工作的，软件上我看看咋样确认那个函数返回值。

redworlf007, 2021-06-18:
RTCdemo里面 打开rtc0 是  O_RDONLY
fd = open ("/dev/rtc0", O_RDONLY);

etup_alarm 函数里面，要给rtc0里面写东西，是不是这里有问题？
retval = ioctl(fd, RTC_ALM_SET, &rtc_tm);

redworlf007, 2021-06-18:
fd = open ("/dev/rtc0", O_RDWR);  我改成这个样子，还是不行。。。

redworlf007, 2021-06-18:
[nuc980_Sleep.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6675614/nuc980_Sleep.zip)

redworlf007, 2021-06-18:
我把编译好的bin文件上传了。

redworlf007, 2021-06-18:
[SuspendToRAM.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6675624/SuspendToRAM.zip)
[Bee.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6675628/Bee.zip)

redworlf007, 2021-06-18:
源代码，你看看。

redworlf007, 2021-06-18:
/* Set the alarm to 5 sec in the future */

        /**********************************/
        fd = open ("/dev/rtc0", O_RDONLY);
        /**********************************/

		if (fd<0) {
            printf("open rtc faild!!!\n");
            close(fd);
			exit(0);
        }

        retval = ioctl(fd, RTC_ALM_SET, &rtc_tm);
		printf("ioctl=%d\n",retval);
		
        if (retval <0) {
                printf("1-ioctl RTC_ALM_SET  faild!!!\n");
                close(fd);
				exit(0);
				}	
open文件，都成功，往里面写配置，随机失败。。。。

redworlf007, 2021-06-18:
Select :
read current time is: 2005-01-02 03:18:55

open rtc success!!!
ioctl=-1
ioctl RTC_ALM_SET  faild!!!
PM: Syncing filesystems ... done.
Freezing user space processes ... (elapsed 0.001 seconds) done.
Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Suspending console(s) (use no_console_suspend to debug)

open文件，都成功，往里面写配置失败。。。

redworlf007, 2021-06-18:
https://blog.csdn.net/pcwung/article/details/78712231

你看看这个帖子，我看网上说要改内核驱动。。。。

redworlf007, 2021-06-18:
err = check_rtc_access_enable(rtc);
        if (IS_ERR(err))
                return PTR_ERR(err);

我在这里加啥打印函数，可以把内核这里的信息，打印到串口控制台上？

redworlf007, 2021-06-18:
244 static int nuc980_rtc_set_alarm(struct device *dev, struct rtc_wkalrm *alrm)
245 {
246         struct nuc980_rtc *rtc = dev_get_drvdata(dev);
247         struct nuc980_bcd_time tm;
248         unsigned long val;
249         int *err;
250 
251         nuc980_rtc_bin2bcd(dev, &alrm->time, &tm);
252 
253         err = check_rtc_access_enable(rtc);
254         if (IS_ERR(err))
255               {
256               printk("rtc_set_alarm IS_ERR(err)\n");
257               return PTR_ERR(err);
258                }
259          else
260             {
261              printk("rtc_set_alarm IS_OK\n");
262             }
263         val = tm.bcd_mday | tm.bcd_mon | tm.bcd_year;
264         val |= (1 << 31); // mask alarm week day
265         rtc_reg_write(rtc, REG_RTC_CAR, val);
266 
267         val = tm.bcd_sec | tm.bcd_min | tm.bcd_hour;
268         rtc_reg_write(rtc, REG_RTC_TAR, val);
269 
270         rtc_reg_write(rtc, REG_RTC_PWRCTL, (__raw_readl(rtc->rtc_reg + REG_RTC_PWRCTL) | (1 << 3)));
271 
272         return 0;
273 }
我在内核驱动这里加了打印，我先测测看。

redworlf007, 2021-06-18:
open rtc success!!!
ioctl=0
ioctl RTC_ALM_SET  success!!!
read current alarm time is: 2005-01-02 06:20:25

Waiting 5 seconds for alarm...

into  rtc_set_alarm
rtc_set_alarm IS_OK
正常情况打印如下

redworlf007, 2021-06-18:
open rtc success!!!
ioctl=0
ioctl RTC_ALM_SET  success!!!
read current alarm time is: 2005-01-02 06:33:50

Waiting 5 seconds for alarm...

into  rtc_set_alarm
rtc_set_alarm IS_OK
PM: Syncing filesystems ... done.
Freezing user space processes ... (elapsed 0.001 seconds) done.
Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Suspending console(s) (use no_console_suspend to debug)
PM: suspend of devices complete after 34.908 msecs
PM: late suspend of devices complete after 3.508 msecs
PM: noirq suspend of devices complete after 3.466 msecs
Wake up source: 00000000  00000080
PM: noirq resume of devices complete after 2.733 msecs
PM: early resume of devices complete after 2.908 msecs
PM: resume of devices complete after 63.066 msecs
Restarting tasks ...

*Alarm interrupt come on,the func running!*
done.
Test second alarm(hour,min,sec) test OK!

******** RTC Test Demo ***********
Test second alarm(hour,min,sec)..
**************************************
Select :
read current time is: 2005-01-02 06:33:55

open rtc success!!!
ioctl=-1
ioctl RTC_ALM_SET  faild!!!
PM: Syncing filesystems ... done.
Freezing user space processes ... (elapsed 0.001 seconds) done.
Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Suspending console(s) (use no_console_suspend to debug)

应该是err = check_rtc_access_enable(rtc); 这后面挂了，还没有打印就休眠了。

redworlf007, 2021-06-18:
static int *check_rtc_access_enable(struct nuc980_rtc *nuc980_rtc)
{
	unsigned int timeout = 0x800;

	rtc_reg_write(nuc980_rtc, REG_RTC_INIR, INIRRESET);

	mdelay(10);

	rtc_reg_write(nuc980_rtc, REG_RTC_AER, AERPOWERON);

	while (!(__raw_readl(nuc980_rtc->rtc_reg + REG_RTC_AER) & AERRWENB)
								&& timeout--)
		mdelay(1);

	if (!timeout)
		return ERR_PTR(-EPERM);

	return 0;
}
那就是这个函数里面报错了。。。咋搞？

redworlf007, 2021-06-18:
while (!(__raw_readl(nuc980_rtc->rtc_reg + REG_RTC_AER) & AERRWENB)&& timeout--)

这里超时了。。。

redworlf007, 2021-06-21:
早上好，我追到这里有问题，请教下，下面怎么处理呢？

redworlf007, 2021-06-21:
我又测试了一下，即便我不休眠，只是单纯的跑alarm程序，每5秒，闹钟起来一次，还是会挂在那里。

yachen, 2021-06-21:
有試過別張板子嗎? 由設計來看, 32k 在這個 flag 就會舉起來.

redworlf007, 2021-06-21:
2个项目上，3个板子，都试了，一样的情况，我改了下程序，遇到这个问题，直接reboot -f重启后，开机重新启动应用程序，能好。。。

redworlf007, 2021-06-21:
我上传的编译好的bin程序，你有开发板的话，跑跑看，能复现的。

redworlf007, 2021-06-21:
是程序运行一会，随机出这种错误。

redworlf007, 2021-06-21:
open rtc success!!!
ioctl=-1
ioctl RTC_ALM_SET  faild!!!
open rtc success!!!
ioctl=-1

出现这个问题的时候，我close rtc，再重新open后，再设置，还是不行。

redworlf007, 2021-06-21:
程序运行出这个错误，退出后，我手动./Bee，重新启动，又能正常运行，运行一段时间又会出现这个问题，应该和硬件没关系。

redworlf007, 2021-06-21:
你搞个简单的alarm程序试了没？我试了这个问题很容易复现。
我是用你们GitHub上的buildroot搞的内核，内核版本是4.4.x的。

redworlf007, 2021-06-22:
早上好，我再咋测呢？有没有啥办法处理这个问题呢？

redworlf007, 2021-06-22:
static int nuc980_rtc_set_alarm(struct device *dev, struct rtc_wkalrm *alrm)
{
	struct nuc980_rtc *rtc = dev_get_drvdata(dev);
	struct nuc980_bcd_time tm;
	unsigned long val;
	int *err;
        
        printk("into  rtc_set_alarm\n");

	nuc980_rtc_bin2bcd(dev, &alrm->time, &tm);

	err = check_rtc_access_enable(rtc);
	if (IS_ERR(err))
              {
              printk("rtc_set_alarm IS_ERR(err)\n");
	      return PTR_ERR(err);
               }
         else
            {
             printk("rtc_set_alarm IS_OK\n");
            }

我在内核这里加了打印，不知道为啥反复进入那个函数。

Select :
read current time is: 2005-01-01 15:15rtc_set_alarm IS_OK
:11

open rtc success!!!
ioctl=0
ioctl RTC_ALM_SET  success!!!
read current alarm time is: 2005-01-01 15:15:16

Waiting 5 seconds for alarm...

into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK

redworlf007, 2021-06-22:
*Alarm interrupt come on,the func running!*
Test second alarm(hour,min,sec) test OK!

******** RTC Tesinto  rtc_set_alarm
t Demo ***********
Test second alarm(hour,min,sec)..
**************************************
Select :
read current time is: 2005-01-01 15:32rtc_set_alarm IS_OK
:27

open rtc success!!!
ioctl=0
ioctl RTC_ALM_SET  success!!!
read current alarm time is: 2005-01-01 15:32:32

Waiting 5 seconds for alarm...

into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK
into  rtc_set_alarm
rtc_set_alarm IS_OK

*Alarm interrupt come on,the func running!*
Test second alarm(hour,min,sec) test OK!

yachen, 2021-06-22:
我試一下好像每秒會進去一次, 但都是設定一樣的值. RTC subsystem 沒什麼研究, 不確定為何有這樣的行為. 回到原本無法設定 alarm, 我改了 applications/demo/rtc 下的代碼, 在測試 alarm 的地方加 while(1), 每三秒發一次, 跑了一個多小時都還正常.  你有開發板可以做對照嗎?

redworlf007, 2021-06-22:
你把测试代码的bin发来，我直接在我的板子上跑。

yachen, 2021-06-22:
[rtc_demo.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6691275/rtc_demo.zip)
請參考附件.

redworlf007, 2021-06-22:
main
{
fd = open("/dev/rtc0", O_RDWR);
while(1)
{
setup_alarm(fd, 5, alarm_func);  //设置5秒闹钟
sleep(10);

}
}

void alarm_func()
{

	printf("\n\n*Alarm interrupt come on,the func running!*\n");
}

你这样试试看，应该可以触发我说的那个问题

yachen, 2021-06-22:
alarm 延長到 5sec. 加 10 sec sleep 跑 20 分鐘了, 都正常.

redworlf007, 2021-06-22:
int main(void)
{

        int i, fd, retval, irqcount = 0;
        unsigned long tmp, data;
        struct rtc_time rtc_tm;
        int scale_type = HR24;
        struct rtc_wkalrm rtc_alarmtm;
        unsigned int select;
        fd = open("/dev/rtc0", O_RDWR);
        fd_all = fd;
        if (fd < 0)
        {
                printf("open rtc faild!!!\n");
                return 0;
        }
        signal(SIGTERM, ctrl_c_signal); /* for the TERM signal.. */
        signal(SIGINT, ctrl_c_signal);  /* for the CTRL+C signal.. */

        printf("\n******** RTC Test Demo ***********\n");
        printf("Test second alarm(hour,min,sec)..\n");
        printf("**************************************\n");
        printf("Select : \n");

        while (1) {

        setup_alarm(fd, 5, alarm_func);
        printf("1234567890\n");

        }
        

        return 0;

} /* end main */

你看我这样测试的，修改的demo例子。

redworlf007, 2021-06-22:
ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 00:59:59

ioctl RTC_ALM_SET  faild!!!
1234567890
read current time is: 2005-01-01 01:00:00

read current alarm time is: 2005-01-01 01:00:05

Waiting 5 seconds for alarm...

*Alarm interrupt come on,the func running!*
Test second alarm(hour,min,sec) test OK!
1234567890
read current time is: 2005-01-01 01:00:05

read current alarm time is: 2005-01-01 01:00:10

Waiting 5 seconds for alarm...

*Alarm interrupt come on,the func running!*
Test second alarm(hour,min,sec) test OK!
1234567890
read current time is: 2005-01-01 01:00:10

read current alarm time is: 2005-01-01 01:00:15

Waiting 5 seconds for alarm...

还是会出现那种情况。

redworlf007, 2021-06-22:
alarm函数
void alarm_func()
{
        //sleep(1);
        printf("\n\n*Alarm interrupt come on,the func running!*\n");
        //system("./Bee &");
        //system("ps au &");
}

redworlf007, 2021-06-23:
while (1) {

        setup_alarm(fd, 5, alarm_func);
        printf("1234567890\n");
        sleep(5);

        }

我里面加了个sleep5 ，从昨晚12点多，测到现在正常。

redworlf007, 2021-06-23:
现在测了差不多24个小时了，目前正常，我再继续测下去看。

redworlf007, 2021-06-28:
我感觉我找到问题的所在了：
setup_alarm(fd, 5, alarm_func);
设置完这个以后，等待alarm闹钟到了，func响应后，要加sleep 延时后，再setup_alarm，就不会报错。

redworlf007, 2021-06-29:
你好，再请教个问题，这个按键唤醒，从休眠状态唤醒，我按bsp手册搞好了，为啥这个唤醒，没有个回调函数？

yachen, 2021-06-29:
Hi, 要什麼樣的回調函數? 可以直接加在eint 的中斷函數中. 或是以下的代碼之後處理. 這是在 nuc980_suspend_enter() 最後幾行. 

```c
	printk(KERN_INFO "Wake up source: %08x  %08x\n", __raw_readl(REG_WKUPSSR0), __raw_readl(REG_WKUPSSR1));
	// clear wake up source
	__raw_writel(__raw_readl(REG_WKUPSSR0), REG_WKUPSSR0);
	__raw_writel(__raw_readl(REG_WKUPSSR1), REG_WKUPSSR1);

```

[3 more replies]