OpenNuvoton/NUC980-linux-4.4.y #66: NUC980 boot from NAND problems

anton7191, 2021-07-30:
Kernel starts to boot, but can't mount rootfs from mtdblock2. 
Could you please help me to solve the problem.
Please see the logs.

I use cmdNuwriter.
#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 5
image0_path= ./nand/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x200
image0_start_offset = 0x000000

image1_type = data
image1_path= ./nand/u-boot.bin
image1_exe_addr = 0x00000
image1_start_offset = 0x100000

image2_type = data
image2_path= ../image/980uimage
image2_exe_addr = 0x000000
image2_start_offset = 0x200000

image3_type = env
image3_path= ./nand/env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= ./kernel/rootfs_yaffs.img
image4_exe_addr = 0x000000
image4_start_offset = 0x1400000

--------------------LOG from putty----------------------------------
Loading from nand0, offset 0x200000
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2131560 Bytes = 2 MiB
   Load Address: 00008000
   Entry Point:  00008000
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2131560 Bytes = 2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Linux version 4.4.207+ (anton@anton-Latitude-5580) (gcc version 4.9.3 (ctng-1.21.0-229g-FA) ) #5 PREEMPT Fri Jul 30 08:05:12 MSK 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 60368K/65536K available (2986K kernel code, 245K rwdata, 940K rodata, 136K init, 198K bss, 5168K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03ddda4   (3928 kB)
      .init : 0xc03de000 - 0xc0400000   ( 136 kB)
      .data : 0xc0400000 - 0xc043d700   ( 246 kB)
       .bss : 0xc043d700 - 0xc046f160   ( 199 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8400 - 0x843c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource nuc980-timer5
NET: Registered protocol family 2
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered (default)
nuc980_gpio_probe - pdev = nuc980-gpio
nuc980-uart.0: ttyS0 at I/O 0x0 (irq = 36, base_baud = 750000) is a NUC980
brd: module loaded
nuc980 mtd nand driver version: 20180301
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xf1
nand: Winbond NAND 128MiB 3,3V 8-bit
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: SMRA size 64, 60
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
fmi-sm: registered successfully! mtdid=nand0
libphy: nuc980_rmii0: probed
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
nuc980-ehci nuc980-ehci: Nuvoton NUC980 EHCI Host Controller
nuc980-ehci nuc980-ehci: new USB bus registered, assigned bus number 1
nuc980-ehci nuc980-ehci: irq 23, io mem 0xb0015000
nuc980-ehci nuc980-ehci: USB 2.0 started, EHCI 0.95
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 2 ports detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
usb_hcd_nuc980_probe, id = 0, name: nuc980-ohci.0, 0
nuc980-ohci nuc980-ohci.0: Nuvoton NUC980 OHCI Host Controller
nuc980-ohci nuc980-ohci.0: new USB bus registered, assigned bus number 2
nuc980-ohci nuc980-ohci.0: irq 24, io mem 0xb0017000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 8 ports detected
usb_hcd_nuc980_probe, id = 1, name: nuc980-ohci.1, 0
usb_hcd_nuc980_probe, id = 2, name: nuc980-ohci.2, 0
usb_hcd_nuc980_probe, id = 3, name: nuc980-ohci.3, 0
usb_hcd_nuc980_probe, id = 4, name: nuc980-ohci.4, 0
usb_hcd_nuc980_probe, id = 5, name: nuc980-ohci.5, 0
usb_hcd_nuc980_probe, id = 6, name: nuc980-ohci.6, 0
usbcore: registered new interface driver usb-storage
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
NET: Registered protocol family 17
VFS: Cannot open root device "mtdblock2" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:
0100           16384 ram0  (driver?)
0101           16384 ram1  (driver?)
0102           16384 ram2  (driver?)
0103           16384 ram3  (driver?)
0104           16384 ram4  (driver?)
0105           16384 ram5  (driver?)
0106           16384 ram6  (driver?)
0107           16384 ram7  (driver?)
0108           16384 ram8  (driver?)
0109           16384 ram9  (driver?)
010a           16384 ram10  (driver?)
010b           16384 ram11  (driver?)
010c           16384 ram12  (driver?)
010d           16384 ram13  (driver?)
010e           16384 ram14  (driver?)
010f           16384 ram15  (driver?)
1f00          131072 mtdblock0  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.207+ #5
Hardware name: NUC980
Backtrace:
[<c0012fe4>] (dump_backtrace) from [<c00131dc>] (show_stack+0x18/0x1c)
 r6:c3a9b000 r5:c3a9b000 r4:00008000 r3:00000000
[<c00131c4>] (show_stack) from [<c017d124>] (dump_stack+0x20/0x28)
[<c017d104>] (dump_stack) from [<c0076fb0>] (panic+0xac/0x238)
[<c0076f08>] (panic) from [<c03df1f4>] (mount_block_root+0x21c/0x2d0)
 r3:00000001 r2:00000000 r1:c383fec0 r0:c0384144
 r7:c03f8858
[<c03defd8>] (mount_block_root) from [<c03df40c>] (mount_root+0x68/0x84)
 r10:00000000 r9:c03de5d8 r8:c03ff2ac r7:c043d700 r6:c03f8858 r5:c03f8858
 r4:00000000
[<c03df3a4>] (mount_root) from [<c03df588>] (prepare_namespace+0x160/0x1c0)
 r4:c043d724 r3:00000000
[<c03df428>] (prepare_namespace) from [<c03dee1c>] (kernel_init_freeable+0x188/0x1d4)
 r6:c03f8834 r5:c043d700 r4:00000008
[<c03dec94>] (kernel_init_freeable) from [<c02ec348>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02ec338
 r4:00000000
[<c02ec338>] (kernel_init) from [<c000fed8>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

-------------ENV.TXT-------------
baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootcmd=nboot 0x7fc0 0 0x200000;bootm 0x7fc0
bootargs=noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init

yachen, 2021-07-30:
Cannot find MTD partition information from your console log. Please make sure you either specify the partition table in device tree or through bootarg. You should see similar message as below if MTD driver received the partition info.
```
Scanning device for bad blocks
Creating 4 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000003200000 : "Kernel_1"
0x000003200000-0x000006200000 : "Kernel_2"
0x000006200000-0x000008000000 : "user"
```

anton7191, 2021-07-30:
U-Boot 2016.11-gbf0ff9f (Sep 09 2020 - 09:30:58 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
=>
=>
=>
=> mtdparts

device nand0 <nand0>, # parts = 3
 #: name                size            offset          mask_flags
 0: u-boot              0x00200000      0x00000000      0
 1: kernel              0x01400000      0x00200000      0
 2: user                0x06a00000      0x01600000      0

active partition: nand0,0 - (u-boot) 0x00200000 @ 0x00000000

defaults:
mtdids  : nand0=nand0
mtdparts: mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user)
=>

yachen, 2021-07-30:
That's the configuration in U-Boot. You need to pass these info to kernel using bootarg (or set them in kernel's device tree)

anton7191, 2021-08-01:
I added to my env.txt file:
baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 console=ttyS0,115200n8 rdinit=/sbin/init mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
bootcmd=nboot 0x7fc0 0 0x200000;bootm 0x7fc0

and I have next error.
Could you please help me to solve the problem.

## LOG from putty

U-Boot 2016.11-gbf0ff9f (Sep 09 2020 - 09:30:58 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

Loading from nand0, offset 0x200000
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2131528 Bytes = 2 MiB
   Load Address: 00008000
   Entry Point:  00008000
Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2131528 Bytes = 2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Linux version 4.4.207+ (anton@anton-Latitude-5580) (gcc version 4.9.3 (ctng-1.21                                                  .0-229g-FA) ) #1 PREEMPT Sun Aug 1 14:16:55 MSK 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c043cf08, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 console=t                                                  tyS0,115200n8 rdinit=/sbin/init mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x                                                  200000(kernel),-(user) ignore_loglevel
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 60368K/65536K available (2986K kernel code, 245K rwdata, 940K rodata, 13                                                  6K init, 198K bss, 5168K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03ddda4   (3928 kB)
      .init : 0xc03de000 - 0xc0400000   ( 136 kB)
[... 81 lines cut ...]
yaffs: yaffs: Attempting MTD mount of 31.2,"mtdblock2"
yaffs: block 845 is bad
yaffs: block 846 is bad
yaffs: block 847 is bad
yaffs: block 848 is bad
yaffs: 8 blocks to be sorted...
yaffs: yaffs_read_super: is_checkpointed 0
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: error mounting -2
Freeing unused kernel memory: 136K
Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidace.
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.207+ #1
Hardware name: NUC980
Backtrace:
[<c0012fe4>] (dump_backtrace) from [<c00131dc>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c02ec338 r4:c043d700 r3:00000000
[<c00131c4>] (show_stack) from [<c017d124>] (dump_stack+0x20/0x28)
[<c017d104>] (dump_stack) from [<c0076fb0>] (panic+0xac/0x238)
[<c0076f08>] (panic) from [<c02ec408>] (kernel_init+0xd0/0xf4)
 r3:00000000 r2:60000013 r1:00001291 r0:c0383f14
 r7:00000000
[<c02ec338>] (kernel_init) from [<c000fed8>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt or guidance.
random: nonblocking pool is

anton7191, 2021-08-01:
I have starting linux, but when I mount mtdblock2 I see next warning:
Could you please help me to solve the problem.

nuc980 mtd nand driver version: 20180301
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xf1
nand: Winbond NAND 128MiB 3,3V 8-bit
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: SMRA size 64, 60
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
3 cmdlinepart partitions found on MTD device nand0
Creating 3 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000001600000 : "kernel"
0x000001600000-0x000008000000 : "user"
fmi-sm: registered successfully! mtdid=nand0
libphy: nuc980_rmii0: probed
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
nuc980-ehci nuc980-ehci: Nuvoton NUC980 EHCI Host Controller
nuc980-ehci nuc980-ehci: new USB bus registered, assigned bus number 1
nuc980-ehci nuc980-ehci: irq 23, io mem 0xb0015000
nuc980-ehci nuc980-ehci: USB 2.0 started, EHCI 0.95
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 2 ports detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
usb_hcd_nuc980_probe, id = 0, name: nuc980-ohci.0, 0
nuc980-ohci nuc980-ohci.0: Nuvoton NUC980 OHCI Host Controller
nuc980-ohci nuc980-ohci.0: new USB bus registered, assigned bus number 2
nuc980-ohci nuc980-ohci.0: irq 24, io mem 0xb0017000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 8 ports detected
usb_hcd_nuc980_probe, id = 1, name: nuc980-ohci.1, 0
usb_hcd_nuc980_probe, id = 2, name: nuc980-ohci.2, 0
usb_hcd_nuc980_probe, id = 3, name: nuc980-ohci.3, 0
usb_hcd_nuc980_probe, id = 4, name: nuc980-ohci.4, 0
usb_hcd_nuc980_probe, id = 5, name: nuc980-ohci.5, 0
usb_hcd_nuc980_probe, id = 6, name: nuc980-ohci.6, 0
usbcore: registered new interface driver usb-storage
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
NET: Registered protocol family 17
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags "inband-tags"
yaffs: yaffs: Attempting MTD mount of 31.2,"mtdblock2"
yaffs: block 845 is bad
yaffs: block 846 is bad
yaffs: block 847 is bad
yaffs: block 848 is bad
yaffs: 151 blocks to be sorted...
yaffs: yaffs_read_super: is_checkpointed 0
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: mounted
Freeing unused kernel memory: 136K
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
[Mount YAFFS2]: /dev/mtdblock2 --> /mnt/mtdblock2

BusyBox v1.22.1 (2016-02-03 14:11:04 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

~ #
~ #
~ # ls
bin         etc         lib         lost+found  proc        sys         usr
dev         flash       linuxrc     mnt         sbin        tmp         var
~ # mount -t yaffs -o "inband-tags" /dev/mtdblock2 /flash
mount: mounting /dev/mtdblock2 on /flash failed: Device or resource busy

yachen, 2021-08-02:
mtdblock2 is already mounted as rootfs. You don't need to mount it again.
```
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags "inband-tags"
yaffs: yaffs: Attempting MTD mount of 31.2,"mtdblock2"
yaffs: block 845 is bad
yaffs: block 846 is bad
yaffs: block 847 is bad
yaffs: block 848 is bad
yaffs: 151 blocks to be sorted...
yaffs: yaffs_read_super: is_checkpointed 0
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
```

anton7191, 2021-08-04:
Thank you very much. I solved my problem very quickly. I changed my env.txt file on this:

baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootcmd=nboot 0x7fc0 0 0x200000;bootm 0x7fc0
bootargs=noinitrd ubi.mtd=2 root=ubi0:system rw rootfstype=ubifs console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel

closed 2021-08-04 by anton7191

zwjin1210, 2023-08-22:
CPU: NUC980
Board: NUC980
DRAM:  128 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x200000, size 0x800000
 8388608 bytes read: OK
Unknown command 'noinitrd' - try 'help'
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.289+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1970864 Bytes = 1.9 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Linux version 4.4.289+ (nuc980@zwjin-vm) (gcc version 4.8.4 (GCC) ) #8 PREEMPT Tue Aug 22 15:00:57 CST 2023
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c03f2768, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 60664K/65536K available (2886K kernel code, 179K rwdata, 804K rodata, 140K init, 200K bss, 4872K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03a2d94   (3692 kB)
      .init : 0xc03a3000 - 0xc03c6000   ( 140 kB)
      .data : 0xc03c6000 - 0xc03f2f60   ( 180 kB)
       .bss : 0xc03f2f60 - 0xc0425040   ( 201 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8400 - 0x843c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
SCSI subsystem initialized
pps_core: LinuxPPS API ver. 1 registered
pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
PTP clock support registered
clocksource: Switched to clocksource nuc980-timer5
NET: Registered protocol family 2
IP idents hash table entries: 2048 (order: 2, 16384 bytes)
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
yaffs: yaffs Installing.
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
io scheduler noop registered (default)
nuc980_gpio_probe - pdev = nuc980-gpio
nuc980_dma_probe - pdev = nuc980-dma
nuc980-dma nuc980-dma: NUC980 DMA ready
nuc980-uart.0: ttyS0 at I/O 0x0 (irq = 36, base_baud = 750000) is a NUC980
nuc980-uart.1: ttyS1 at I/O 0x0 (irq = 37, base_baud = 9375000) is a NUC980
nuc980-uart.2: ttyS2 at I/O 0x0 (irq = 38, base_baud = 9375000) is a NUC980
nuc980-uart.3: ttyS3 at I/O 0x0 (irq = 43, base_baud = 9375000) is a NUC980
brd: module loaded
nuc980 mtd nand driver version: 20180301
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xf1
nand: Winbond NAND 128MiB 3,3V 8-bit
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: SMRA size 64, 60
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
3 cmdlinepart partitions found on MTD device nand0
Creating 3 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000001600000 : "kernel"
0x000001600000-0x000008000000 : "user"
fmi-sm: registered successfully! mtdid=nand0
libphy: nuc980_rmii0: probed
nuc980-rtc nuc980-rtc: rtc core: registered nuc980-rtc as rtc0
NET: Registered protocol family 17
nuc980-rtc nuc980-rtc: setting system clock to 2005-01-01 00:00:00 UTC (1104537600)
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags "inband-tags"
yaffs: yaffs: Attempting MTD mount of 31.2,"mtdblock2"
yaffs: block 845 is bad
yaffs: block 846 is bad
yaffs: block 847 is bad
yaffs: block 848 is bad
yaffs: 103 blocks to be sorted...
yaffs: yaffs_read_super: is_checkpointed 0
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: mounted
Freeing unused kernel memory: 140K

上面是我的启动日志 为啥每块板子启动都存在yaffs: block 845 is bad
yaffs: block 846 is bad
yaffs: block 847 is bad
yaffs: block 848 is bad