OpenNuvoton/NUC980-linux-4.4.y #68: How change UART2 baudrate

anton7191, 2021-08-04:
Hi, I turn on UART2 without DMA, but it work on baudrate=9600. And I can't find where changing this parameter. 
Would you help me please?

yachen, 2021-08-04:
Hi, did you use cfsetospeed() and cfsetispeed() to set buadrate? Please refer to sample code in https://github.com/OpenNuvoton/NUC980_Linux_Applications/blob/master/demos/uart/uart.c

anton7191, 2021-08-09:
I made this C code and added bin-file to my rootfs system (rootfs/sbin). But this program is not working. 
~:Illegal instruction

yachen, 2021-08-10:
Please use file instruction on PC to check the file type.

anton7191, 2021-08-12:
Hi, thanks for your help. I resolved this problem when I built the project with the necessary compiler.

closed 2021-08-12 by anton7191