OpenNuvoton/NUC980-linux-4.4.y #69: Unable to boot from DTB NUC980 IOT Board

swapnilpanwal, 2021-08-05:
Development board: NuMaker-IIoT-NUC980, pcb silk screen model: NK-980IOT V1.0
Configuration: make nuc980_iot_defconfig
Enabled : Boot options->[*] Flattened Device Tree support
dts file compiled with : dtc nuc980-iot-v1.0.dts -o nuc980-iot-v1.0.dtb
we have used bellow run.ini and env file but unable to boot successfully
**run.ini **
[SPINAND]
# This mode set "run" to control SPINAND flash and specify the type of the image. 
# These types can be recognized by bootloader or Linux. 
# The image type is set loader, data, environment or pack.

# program : write a new image to SPINAND
# program_verify : write a new image to SPINAND and verify image
# read : read data from SPINAND to your's computer.
# erase : erase SPINAND.
# run: program, program_verify, read, erase, #
run = program

#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 6
image0_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/build/uboot-master/spl/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x000200
image0_start_offset = 0x00000

image1_type = data
image1_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/build/uboot-master/u-boot.bin
image1_exe_addr = 0x0000
image1_start_offset = 0x100000

image2_type = data
image2_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/images/uImage
image2_exe_addr = 0x0000
image2_start_offset = 0x200000

image3_type = env
image3_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/images/rootfs.yaffs2
image4_exe_addr = 0x00000
image4_start_offset = 0x1600000

image5_type = data
image5_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/images/nuc980-iot-v1.0.dtb
image5_exe_addr = 0x00000
image5_start_offset = 0x1400000

# "yes" if you are using user defined to config SPINAND parameters
# using_user_defined: yes, no #
using_user_defined = no
page_size=2048
spare_area=64
quad_read_command = 0x6b
read_status_command = 0xff
write_status_command = 0xff
status_value = 0xff
dummy_bytes = 0x1
blockperflash=1024
pageperflash=64

#------------------- read -------------------#
read_path =  ./pattern/record_spinand.bin
read_start_blocks = 0
read_offset_blocks = 10

#------------------- erase ------------------#
# erase_all: yes, no(Will use start blocks and offset blocks) #
erase_all = yes
erase_start_blocks = 0
erase_offset_blocks = 10

**uboot env**
baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
setspi=sf probe 0 30000000
loadkernel=sf read 0x7fc0 0x200000 0x800000
loaddtb=sf read 1400000 0x40000 0x10000
bootcmd=run setspi;run loadkernel;run loaddtb;bootm 0x7fc0 - 0x1400000

**boot logs**
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow�load

 SPL load main U-Boot from SPI NAND Flash! (Aug 05 2021 14:38:02)

U-Boot 2016.11 (Aug 05 2021 - 14:38:02 +0530)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x800000
SF: 8388608 bytes @ 0x200000 Read: OK
device 0 offset 0x40000, size 0x10000
SF: 65536 bytes @ 0x40000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2675304 Bytes = 2.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
=>

yachen, 2021-08-06:
Can you use following commands to check if the dtb is read from SPI flash correctly.
```
sf read 1400000 40000 10000
md 1400000 
```

flownder, 2021-08-10:
HI swapnilpanwal, problem solved?
I have a similar problem:
Booting from NAND
my board: NUVOTON NK-NUC98R with NUC98RDF61YC
Kernel configuration: nuc980_defconfig
Enabled : Boot options->[*] Flattened Device Tree support
build DTB: make -j8 ARCH=arm CROSS_COMPILE=arm-nuvoton-linux-uclibcgnueabi- nuc980-dev-v1.0.dtb
My ini file:
`[NAND]
# This mode set "run" to control NAND flash and specify the type of the image. 
# These types can be recognized by bootloader or Linux. 
# The image type is set loader, data, environment or pack.

# program : write a new image to NAND flash
# program_verify : write a new image to NAND flash and verify image
# read : read data from  NAND flash to your's computer.
# erase : erase NAND flash.
# run: program, program_verify, read, erase, #
run = program

#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 5
image0_path= ../new_image/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x200
image0_start_offset = 0x000000

image1_type = data
image1_path= ../new_image/u-boot.bin
image1_exe_addr = 0x00000
image1_start_offset = 0x100000

image2_type = data
image2_path= ../new_image/uImage
image2_exe_addr = 0x000000
image2_start_offset = 0x200000

image3_type = env
image3_path= ../new_image/env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= ../new_image/nuc980-dev-v1.0.dtb
image4_exe_addr = 0x00000
image4_start_offset = 0x1400000

# "yes" if you are using user defined to config NAND flash parameters
# using_user_defined: yes, no #
using_user_defined = no
blockperflash=1024
pageperblock=64

#------------------- read -------------------#
read_path =  ./pattern/record_nand.bin
read_start_blocks = 0
read_offset_blocks = 10

#------------------- erase ------------------#
# erase_all: yes, no #
erase_all = yes
erase_start_blocks = 0
erase_offset_blocks = 10`

My question: where set dtb address?
`baudrate=115200
bootcmd=nboot 0x7fc0 0 0x200000 0x1400000;bootm 0x7fc0; <- ????
bootdelay=3
stderr=serial
stdin=serial
stdout=serial`

flownder, 2021-08-10:
HI swapnilpanwal, problem solved?
I have a similar problem:
Booting from NAND
my board: NUVOTON NK-NUC98R with NUC98RDF61YC
Kernel configuration: nuc980_defconfig
Enabled : Boot options->[*] Flattened Device Tree support
build DTB: make -j8 ARCH=arm CROSS_COMPILE=arm-nuvoton-linux-uclibcgnueabi- nuc980-dev-v1.0.dtb
My ini file:
[NAND]
# This mode set "run" to control NAND flash and specify the type of the image. 
# These types can be recognized by bootloader or Linux. 
# The image type is set loader, data, environment or pack.

# program : write a new image to NAND flash
# program_verify : write a new image to NAND flash and verify image
# read : read data from  NAND flash to your's computer.
# erase : erase NAND flash.
# run: program, program_verify, read, erase, #
run = program

#------------------- program -------------------#
# type: data, env, loader, pack #
image_num = 5
image0_path= ../new_image/u-boot-spl.bin
image0_type = loader
image0_exe_addr = 0x200
image0_start_offset = 0x000000

image1_type = data
image1_path= ../new_image/u-boot.bin
image1_exe_addr = 0x00000
image1_start_offset = 0x100000

image2_type = data
image2_path= ../new_image/uImage
image2_exe_addr = 0x000000
image2_start_offset = 0x200000

image3_type = env
image3_path= ../new_image/env.txt
image3_exe_addr = 0x00000
image3_start_offset = 0x80000

image4_type = data
image4_path= ../new_image/nuc980-dev-v1.0.dtb
image4_exe_addr = 0x00000
image4_start_offset = 0x1400000

# "yes" if you are using user defined to config NAND flash parameters
# using_user_defined: yes, no #
using_user_defined = no
blockperflash=1024
pageperblock=64

#------------------- read -------------------#
read_path =  ./pattern/record_nand.bin
read_start_blocks = 0
read_offset_blocks = 10

#------------------- erase ------------------#
# erase_all: yes, no #
erase_all = yes
erase_start_blocks = 0
erase_offset_blocks = 10

My question: where set dtb address in env.txt?

baudrate=115200
bootcmd=nboot 0x7fc0 0 0x200000 0x1400000;bootm 0x7fc0; <- ???
bootdelay=3
stderr=serial
stdin=serial
stdout=serial

swapnilpanwal, 2021-08-10:
you can use this
nand read 0x7fc0 0x200000 0x800000
nand read 0x1400000 0x40000 0x10000
bootm 0x7fc0 - 0x1400000
and in run.ini use **0x40000**
image4_type = data
image4_path= ../new_image/nuc980-dev-v1.0.dtb
image4_exe_addr = 0x00000
image4_start_offset = **0x40000**

swapnilpanwal, 2021-08-11:
> Can you use following commands to check if the dtb is read from SPI flash correctly.
> 
> ```
> sf read 1400000 40000 10000
> md 1400000 
> ```

swapnilpanwal, 2021-08-11:
> > Can you use following commands to check if the dtb is read from SPI flash correctly.
> > ```
> > sf read 1400000 40000 10000
> > md 1400000 
> > ```

thanks for your support

problem in run.ini  we are passing wrong offset address for dtb
by changing address **0x40000** it works fine with dtb
image5_type = data
image5_path= /home/swapnil/Nuvoton/NUC980_Buildroot_dts/NUC970_Buildroot/output/images/nuc980-iot-v1.0.dtb
image5_exe_addr = 0x00000
image5_start_offset = **0x40000**

closed 2021-08-11 by swapnilpanwal

flownder, 2021-08-12:
Unfortunately it does not work
nuc980-iot-v1.0.dtb invalid or unable to find
Maybe am forgot select anything else from kernel config?

`NUC980 IBR 20180813
Boot from NAND
DDR-OK
finish NAND dow▒c{
NAND boot!        #S

U-Boot 2016.11-gd2ff7a9 (Jul 19 2021 - 11:36:00 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

Loading from nand0, offset 0x200000
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4856760 Bytes = 4.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
 #Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4856760 Bytes = 4.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

StartiBooting Linux on physical CPU 0x0
Linux version 4.4.207 (yuriy@yuriy-desktop) (gcc version 7.5.0 (Buildroot 2020.02) ) #14 PREEMPT Thu Aug 12 12:22:51 MSK 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 57760K/65536K available (2906K kernel code, 228K rwdata, 900K rodata, 2888K init, 198K bss, 7776K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03bfdcc   (3808 kB)
      .init : 0xc03c0000 - 0xc0692000   (2888 kB)
      .data : 0xc0692000 - 0xc06cb220   ( 229 kB)
       .bss : 0xc06cb220 - 0xc06fcbdc   ( 199 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
Console: colour dummy device 80x30
console [ttyS0] enabled
**Calibrating delay loop...**`

swapnilpanwal, 2021-08-12:
As per your previous comment i think you are using nuc980-dev-v1.0.dtb so please replace that in your run.ini

> Unfortunately it does not work
> nuc980-iot-v1.0.dtb invalid or unable to find
> Maybe am forgot select anything else from kernel config?
>
> `NUC980 IBR 20180813
> Boot from NAND
> DDR-OK
> finish NAND dow▒c{ [...]

 i think you are using nuc980-dev-v1.0.dtb so you can use **nuc980-dev-v1.0.dtb** instead of nuc980-iot-v1.0.dtb

flownder, 2021-08-12:
nuc980-iot-v1.0.dtb My mistake in this post (copy-paste)
In run.ini **image4_path= ../new_image/nuc980-dev-v1.0.dtb**

swapnilpanwal, 2021-08-12:
please share your env.txt,run.ini and .config file(buildroot,linux,uboot)

flownder, 2021-08-12:
**U-Boot:**
CONFIG_ARM=y
CONFIG_TARGET_NUC980=y
CONFIG_MMC=y
CONFIG_SD_BOOT=y
CONFIG_BOOTDELAY=3
CONFIG_SYS_CONSOLE_OVERWRITE_ROUTINE=y
CONFIG_SYS_NO_FLASH=y
CONFIG_SPL=y
CONFIG_HUSH_PARSER=y
#CONFIG_CMD_IMLS is not set
#CONFIG_CMD_FLASH is not set
CONFIG_CMD_MMC=y
#CONFIG_CMD_SETEXPR is not set
CONFIG_CMD_TIMER=y
CONFIG_NUC980_GPIO=y
CONFIG_GENERIC_MMC=y
CONFIG_NUC980_MMC=y
CONFIG_NUC980_SD_PORT0=y
CONFIG_NAND_NUC980=y
CONFIG_NUC980_CONSOLE=y

**Buildroot:**
BR2_arm=y
BR2_TOOLCHAIN_BUILDROOT_VENDOR="nuvoton"
BR2_KERNEL_HEADERS_4_4=y
BR2_TOOLCHAIN_BUILDROOT_LOCALE=y
BR2_GCC_VERSION_7_X=y
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="file:///home/yuriy/Work/NUC98R/nuc980-buildroot/NUC980-linux-4.4.y-master.tar.gz"
BR2_LINUX_KERNEL_DEFCONFIG="nuc980"
BR2_LINUX_KERNEL_UIMAGE=y
BR2_PACKAGE_BUSYBOX_CONFIG="board/nuvoton/busybox-nuc980.config"
BR2_PACKAGE_OPENSSL=y
BR2_PACKAGE_LIBCURL=y
BR2_PACKAGE_MOSQUITTO=y
BR2_TARGET_ROOTFS_INITRAMFS=y

**Kernel:**
#CONFIG_LOCALVERSION_AUTO is not set
#CONFIG_SWAP is not set
CONFIG_SYSVIPC=y
#CONFIG_CROSS_MEMORY_ATTACH is not set
CONFIG_FHANDLE=y
CONFIG_NO_HZ_IDLE=y
CONFIG_HIGH_RES_TIMERS=y
CONFIG_BSD_PROCESS_ACCT=y
CONFIG_BSD_PROCESS_ACCT_V3=y
#CONFIG_UTS_NS is not set
#CONFIG_IPC_NS is not set
#CONFIG_PID_NS is not set
#CONFIG_NET_NS is not set
CONFIG_BLK_DEV_INITRD=y
CONFIG_INITRAMFS_SOURCE="${BR_BINARIES_DIR}/rootfs.cpio"
#CONFIG_RD_BZIP2 is not set
#CONFIG_RD_LZMA is not set
#CONFIG_RD_XZ is not set
#CONFIG_RD_LZO is not set
#CONFIG_RD_LZ4 is not set
#CONFIG_COMPAT_BRK is not set
CONFIG_MODULES=y
CONFIG_MODULE_UNLOAD=y
#CONFIG_IOSCHED_DEADLINE is not set
#CONFIG_IOSCHED_CFQ is not set
CONFIG_ARCH_NUC980=y
CONFIG_PREEMPT=y
CONFIG_AEABI=y
CONFIG_OABI_COMPAT=y
#CONFIG_CPU_SW_DOMAIN_PAN is not set
CONFIG_USE_OF=y
CONFIG_CMDLINE="root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M"
CONFIG_KEXEC=y
#CONFIG_ATAGS_PROC is not set
CONFIG_AUTO_ZRELADDR=y
CONFIG_FPE_NWFPE=y
#CONFIG_CORE_DUMP_DEFAULT_ELF_HEADERS is not set
#CONFIG_SUSPEND is not set
CONFIG_NET=y
CONFIG_PACKET=y
CONFIG_PACKET_DIAG=y
CONFIG_UNIX=y
CONFIG_UNIX_DIAG=y
CONFIG_INET=y
CONFIG_IP_MULTICAST=y
#CONFIG_INET_XFRM_MODE_TRANSPORT is not set
#CONFIG_INET_XFRM_MODE_TUNNEL is not set
#CONFIG_INET_XFRM_MODE_BEET is not set
#CONFIG_INET_LRO is not set
#CONFIG_INET_DIAG is not set
#CONFIG_IPV6 is not set
#CONFIG_WIRELESS is not set
CONFIG_UEVENT_HELPER_PATH="/sbin/hotplug"
CONFIG_DEVTMPFS=y
CONFIG_DEVTMPFS_MOUNT=y
#CONFIG_STANDALONE is not set
#CONFIG_PREVENT_FIRMWARE_BUILD is not set
#CONFIG_FIRMWARE_IN_KERNEL is not set
CONFIG_BLK_DEV_RAM=y
CONFIG_BLK_DEV_RAM_SIZE=16384
CONFIG_SCSI=y
#CONFIG_SCSI_PROC_FS is not set
CONFIG_BLK_DEV_SD=y
#CONFIG_SCSI_LOWLEVEL is not set
CONFIG_NETDEVICES=y
CONFIG_DUMMY=y
#CONFIG_NET_VENDOR_ARC is not set
#CONFIG_NET_CADENCE is not set
#CONFIG_NET_VENDOR_BROADCOM is not set
#CONFIG_NET_VENDOR_CIRRUS is not set
#CONFIG_NET_VENDOR_EZCHIP is not set
#CONFIG_NET_VENDOR_FARADAY is not set
#CONFIG_NET_VENDOR_INTEL is not set
#CONFIG_NET_VENDOR_MARVELL is not set
#CONFIG_NET_VENDOR_MICREL is not set
#CONFIG_NET_VENDOR_NATSEMI is not set
CONFIG_NUC980_ETH0=y
#CONFIG_NET_VENDOR_QUALCOMM is not set
#CONFIG_NET_VENDOR_RENESAS is not set
#CONFIG_NET_VENDOR_ROCKER is not set
#CONFIG_NET_VENDOR_SAMSUNG is not set
#CONFIG_NET_VENDOR_SEEQ is not set
#CONFIG_NET_VENDOR_SMSC is not set
#CONFIG_NET_VENDOR_STMICRO is not set
#CONFIG_NET_VENDOR_SYNOPSYS is not set
#CONFIG_NET_VENDOR_VIA is not set
#CONFIG_NET_VENDOR_WIZNET is not set
CONFIG_ICPLUS_PHY=y
#CONFIG_USB_NET_DRIVERS is not set
#CONFIG_WLAN is not set
#CONFIG_INPUT_MOUSEDEV is not set
#CONFIG_INPUT_KEYBOARD is not set
#CONFIG_INPUT_MOUSE is not set
#CONFIG_SERIO is not set
CONFIG_VT_HW_CONSOLE_BINDING=y
CONFIG_DEVPTS_MULTIPLE_INSTANCES=y
CONFIG_SERIAL_NUC980=y
CONFIG_SERIAL_NUC980_CONSOLE=y
#CONFIG_HW_RANDOM is not set
CONFIG_PINCTRL_NUC980=y
CONFIG_GPIO_SYSFS=y
CONFIG_GPIO_NUC980=y
#CONFIG_HWMON is not set
#CONFIG_HID_GENERIC is not set
CONFIG_USB=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_NUC980_EHCI=y
CONFIG_USB_OHCI_HCD=y
CONFIG_USB_NUC980_OHCI=y
CONFIG_USB_STORAGE=y
#CONFIG_DNOTIFY is not set
#CONFIG_INOTIFY_USER is not set
CONFIG_MSDOS_FS=y
CONFIG_VFAT_FS=y
CONFIG_TMPFS=y
CONFIG_TMPFS_POSIX_ACL=y
#CONFIG_NETWORK_FILESYSTEMS is not set
CONFIG_NLS_CODEPAGE_437=y
CONFIG_NLS_ISO8859_1=y
#CONFIG_ENABLE_WARN_DEPRECATED is not set
#CONFIG_ENABLE_MUST_CHECK is not set
CONFIG_DEBUG_FS=y
CONFIG_STRICT_DEVMEM=y
#CONFIG_ARM_UNWIND is not set
#CONFIG_CRYPTO_MANAGER_DISABLE_TESTS is not set
CONFIG_CRYPTO_GF128MUL=y
CONFIG_CRYPTO_SEQIV=y
#CONFIG_CRYPTO_ECHAINIV is not set
#CONFIG_CRYPTO_HW is not set
CONFIG_CRC_CCITT=y
CONFIG_XZ_DEC=y

**My last worked env.txt (without kernel dtb option):**
baudrate=115200
bootcmd=nboot 0x7fc0 0 0x200000;bootm 0x7fc0;
bootdelay=5
stderr=serial
stdin=serial
stdout=serial

The commands input manually in uboot prompt:
nand read 0x7fc0 0x200000 0x800000
nand read 0x1400000 0x40000 0x10000
bootm 0x7fc0 - 0x1400000

My configs files:
[cfg.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/6974876/cfg.zip)

[image]

gloabt, 2021-08-12:
Have you enabled CONFIG_OF_FLATTREE=y in the kernel? 
i.e Enabled: Boot options->[*] Flattened Device Tree support.
which board & flash you are using.  try SPI NAND in run.ini

flownder, 2021-08-12:
> Have you enabled CONFIG_OF_FLATTREE=y in the kernel?
> i.e Enabled: Boot options->[*] Flattened Device Tree support.
> which board & flash you are using. try SPI NAND in run.ini

Yes, option  **Boot options->[*] Flattened Device Tree support** enabled
[image]

swapnilpanwal, 2021-08-13:
is **nuc980-iot-v1.0.dtb** present in this location **NUC970_Buildroot/output/build/linux-master/arch/arm/boot/dts**

flownder, 2021-08-13:
i see this file after compilation **make -j8 ARCH=arm CROSS_COMPILE=arm-nuvoton-linux-uclibcgnueabi- nuc980-dev-v1.0.dtb** and copy to directory new_image (set path run.ini)

flownder, 2021-08-13:
i will try to rebuild all

swapnilpanwal, 2021-08-13:
> i see this file after compilation **make -j8 ARCH=arm CROSS_COMPILE=arm-nuvoton-linux-uclibcgnueabi- nuc980-dev-v1.0.dtb** and copy to directory new_image (set path run.ini)

Again you mention here **nuc980-dev-v1.0.dtb**

swapnilpanwal, 2021-08-13:
please check this commands
sf probe
sf read 1400000 40000 10000
md 1400000 
its looks like bellow
=> sf probe
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
=>sf read 1400000 40000 10000 
device 0 offset 0x40000, size 0x10000
SF: 65536 bytes @ 0x40000 Read: OK
=> md 1400000
01400000: edfe0dd0 86760000 38000000 5c740000    ......v....8..t\
01400010: 28000000 11000000 10000000 00000000    ...(............
01400020: 2a020000 24740000 00000000 00000000    ...*..t$........
01400030: 00000000 00000000 01000000 00000000    ................
01400040: 03000000 04000000 00000000 01000000    ................
01400050: 03000000 04000000 0f000000 01000000    ................
01400060: 03000000 18000000 1b000000 6f76754e    ............Nuvo
01400070: 206e6f74 3943554e 49203038 5620544f    ton NUC980 IOT V
01400080: 00302e31 03000000 27000000 21000000    1.0........'...!
01400090: 6f76756e 2c6e6f74 3963756e 642d3038    nuvoton,nuc980-d
014000a0: 762d7665 00302e31 6f76756e 2c6e6f74    ev-v1.0.nuvoton,
014000b0: 3963756e 00003038 03000000 04000000    nuc980..........
014000c0: 2c000000 01000000 01000000 736f6863    ...,........chos
014000d0: 00006e65 03000000 4b000000 3d000000    en.........K...=
014000e0: 746f6f72 65642f3d 61722f76 6320306d    root=/dev/ram0 c
014000f0: 6f736e6f 743d656c 30537974 3531312c    onsole=ttyS0,115

swapnilpanwal, 2021-08-13:
which board & flash you are using

flownder, 2021-08-13:
> which board & flash you are using

> please check this commands
> sf probe
> sf read 1400000 40000 10000
> md 1400000
> its looks like bellow
> => sf probe
> SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
> =>sf read [...]

[image]

SF command not found. Because uboot build without optoin SPI flash suppurt

flownder, 2021-08-13:
> which board & flash you are using

NK-NUC980
use NAND Flash
[image]

yachen, 2021-08-13:
Your system is booting from NAND flash, please use "nand read" instead of "sf read"

flownder, 2021-08-13:
> Your system is booting from NAND flash, please use "nand read" instead of "sf read"

I already did this
[image]

yachen, 2021-08-13:
Before bootm command, can you use md to check the content in 14000000? It looks like the kernel didn't find the dtb.

flownder, 2021-08-13:
> Before bootm command, can you use md to check the content in 14000000? It looks like the kernel didn't find the dtb.

md output on screen above.It seems to me that it is not correct.

flownder, 2021-08-13:
> Before bootm command, can you use md to check the content in 14000000? It looks like the kernel didn't find the dtb.

md output on screen above (revious screen). It seems to me that it is not correct

flownder, 2021-08-13:
I rebuild all images step by step from User manual, but no sucsess. Image_type is right?
**image4_type = data**
image4_path= ../new_image/nuc980-dev-v1.0.dtb
image4_exe_addr = 0x00000
image4_start_offset = 0x40000

gloabt, 2021-08-13:
enable following in uboot to require FDT command and do make  ..it will work.
CONFIG_FIT=y
CONFIG_OF_LIBFDT=y

flownder, 2021-08-13:
> enable following in uboot to require FDT command and do make ..it will work.
> 
> CONFIG_OF_LIBFDT=y

Enable this function, rebuild, nothing has changed (

flownder, 2021-08-13:
I try run dtb and uImage from SDRAM, but unsuccessfully
NUC980 IBR 20180813
Boot from USB
DDR-OK
=======================================
Run firmware code
CPU: 300MHz, DDR: 300MHz, SYS: 300MHz, PCLK: 150MHz
Parse NuWriter command line
=======================================
SPI NOR ID=0x0000ef17  _spi_type =0
SPI NOR ID=0x0000ef17  _spi_type =0, Auto Detect

SPI NAND: Error MXIC spiNAND_ReadyBusy_Check timeout
SPI NAND ID 0x401800 not support!!

NAND ID=[ef][f1][0][95]
PowerOn setting 0x7f6003d4
Auto Detect:
BlockPerFlash= 1023, PagePerBlock= 64
PageSize= 2048, ECC= T8, ExtraDataSize= 64, SectorPerBlock= 256

BlockPerFlash=1023, PagePerBlock=64, PageSize=2048

eMMC0/SD0 Init Fail

Finish get INFO!!

finish SDRAM download ...

finish SDRA

...No booting process

My run.ini
[SDRAM]
option = run
sdram_path = nuc980bsp/buildroot/output/images/uImage
exe_addr = 0x8000

using_dtb = yes
dtb_path = nuc980bsp/buildroot/output/images/nuc980-dev-v1.0.dtb
dtb_addr = 0x1400000

yachen, 2021-08-17:
Please use Image instead of uImage while execute from RAM.

> [SDRAM]
option = run
sdram_path = nuc980bsp/buildroot/output/images/uImage
exe_addr = 0x8000

flownder, 2021-08-17:
dtb will be append to Image inside?

yachen, 2021-08-17:
No.

flownder, 2021-08-18:
Thanks, it work

swapnilpanwal, 2021-12-22:
@flownder need help from you 
how can you use **Buildroot 2020.02** for **NUC980**

flownder, 2021-12-22:
> @flownder need help from you how can you use **Buildroot 2020.02** for **NUC980**

Hi, im download tarrbal from buildroot.org and copy nuvoton_nuc980_defconfig to /buildroot/configs

swapnilpanwal, 2021-12-22:
thanks @flownder  
we are facing bellow error while buildroot make 
**Makefile.legacy:9: *** "You have legacy configuration in your .config! Please check your configuration.".  Stop.**

flownder, 2021-12-22:
@swapnilpanwal
my mistake...don't copy nuvoton_nuc980_defconfig to /buildroot/configs
1.just clone buildroot repo or download tarball 2020.02
2. make menuconfig
3. and set options below

`BR2_arm=y
BR2_TOOLCHAIN_BUILDROOT_VENDOR="nuvoton"
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_4_4=y
BR2_UCLIBC_CONFIG="board/nuvoton/uClibc-nuc980.config"
BR2_TOOLCHAIN_BUILDROOT_INET_RPC=y
BR2_TOOLCHAIN_BUILDROOT_LOCALE=y
BR2_GCC_VERSION_4_8_X=y
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_PACKAGE_HOST_GDB=y
BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_MDEV=y
# BR2_TARGET_ENABLE_ROOT_LOGIN is not set
# BR2_TARGET_GENERIC_GETTY is not set
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_GIT=y
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/OpenNuvoton/NUC980-linux-4.4.y.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="master"
BR2_LINUX_KERNEL_DEFCONFIG="nuc980"
BR2_LINUX_KERNEL_UIMAGE=y
BR2_PACKAGE_APPLICATIONS=y
NUC980_APPLICATIONS=y
NUC980_APPLICATIONS_BUILD_DEMOS=y
NUC980_APPLICATIONS_BUILD_YAFFS2UTILS=y
BR2_PACKAGE_BUSYBOX_CONFIG="board/nuvoton/busybox-nuc980.config"
BR2_PACKAGE_ALSA_UTILS=y
BR2_PACKAGE_BZIP2=y
BR2_PACKAGE_ZIP=y
BR2_PACKAGE_ZLIB=y
BR2_PACKAGE_MOSQUITTO=y
BR2_TARGET_ROOTFS_INITRAMFS=y
BR2_TARGET_UBOOT=y
BR2_TARGET_UBOOT_BUILD_SYSTEM_KCONFIG=y
BR2_TARGET_UBOOT_CUSTOM_GIT=y
BR2_TARGET_UBOOT_CUSTOM_REPO_URL="https://github.com/OpenNuvoton/NUC970_U-Boot_v2016.11.git"
BR2_TARGET_UBOOT_CUSTOM_REPO_VERSION="master"
BR2_TARGET_UBOOT_BOARD_DEFCONFIG="nuc980"`

Note:
BR2_UCLIBC_CONFIG="board/nuvoton/uClibc-nuc980.config"
BR2_PACKAGE_BUSYBOX_CONFIG="board/nuvoton/busybox-nuc980.config"

Files uClibc-nuc980.config and busybox-nuc980.config copied from https://github.com/OpenNuvoton/NUC970_Buildroot/tree/master/board/nuvoton

munababy, 2023-01-25:
[NUC972 Device Tree.pdf](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/10495769/NUC972.Device.Tree.pdf)
For nandflash。