OpenNuvoton/NUC980-linux-4.4.y #71: 使用设备树时，内核中SPI开启DMA时 会导致SPI错误，SPI 设备无法工作

QQ466005858, 2021-08-07:
qspi0: qspi0@b0060000 {
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			pinctrl-names = "default";		
			pinctrl-0 = <&pinctrl_qspi0>;
			num_cs  = <1>;
			lsb	 = <0>;
			txneg   = <1>;
			rxneg   = <0>;
			clkpol  = <0>;
			divider = <4>;
			sleep   = <0>;
			txnum   = <0>;
			txbitlen = <8>;
			bus_num = <0>;

			flash: mt29f@0 {
				compatible = "mt29f";
				#address-cells = <1>;
				#size-cells = <1>;
				reg = <0>;
				spi-max-frequency = <100000000>;
				spi-rx-bus-width = <4>;
				spi-tx-bus-width = <4>;
				partition@0 {
					label = "u-boot";
					reg = <0x00000000 0x0200000>;
					read-only;
				};
				partition@200000 {
					label = "kernel";
					reg = <0x0200000 0x1400000>;
				};
				partition@1600000 {
					label = "ubi";
					reg = <0x1600000 0x6A00000>;
				};
			};
				
		};

QQ466005858, 2021-08-07:
在内核中关闭DMA选项，则启动正常，开启QSPI的DMA或者SPI的DMA，问题复现，是不是设备树配置错了。

ychuang3, 2022-01-24:
Hi,

請 update 2022/1/22 commit，已修改 SPI Driver 及 device tree  fix 此問題。

YC

QQ466005858, 2022-05-05:
謝謝

closed 2022-05-05 by QQ466005858