OpenNuvoton/NUC980-linux-4.4.y #72: How I can load out-of-tree kernel module at boot time in NUC980 with 4.4.y linux kernel?

miteshdube, 2021-08-20:
I am trying to load the kernel module at boot time. I have copied the kernel module in the location as follows:
`cp mymodule.ko /lib/modules/4.4.207/kernel/`

And created the file `modules` under /etc/ directory :
```
~ # cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

mymodule
```
But it is not loading the mymodule at boot time.

yachen, 2021-08-23:
I think currently there's no process/script parsing that file during boot time.
You can manually out the "insmod mymodule" command in init.d/rcS file.

miteshdube, 2021-08-23:
Issue solved. Thank you.

closed 2021-08-23 by miteshdube