OpenNuvoton/NUC980-linux-4.4.y #73: Problems booting rootfs from SD

anton7191, 2021-08-24:
Hi, I'm having problems booting rootfs from SD card. My Linux image started but rootfs couldn't be loaded  from SD card. 
U-boot and Kernel are stored SPI NOR flash, and rootfs - on SD card.
U-boot detects SD card and Kernel does not.
How can I solve this problem?

U-Boot 2016.11-g4680eed-dirty (Aug 23 2021 - 21:36:07 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
MMC:   NUC980 SD: 0, NUC980 eMMC: 1
SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
=>
=>
=>
=> mmc info
Device: NUC980 SD
Manufacturer ID: 3
OEM: 5344
Name: SC16G
Tran Speed: 25000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.8 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
=>

# Putty #

NUC980 IBR 20180813
Boot from SPI-NOR
DDR-OK
finish SPI dow▒oad

U-Boot 2016.11-g4680eed-dirty (Aug 23 2021 - 21:36:07 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
MMC:   NUC980 SD: 0, NUC980 eMMC: 1
SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
device 0 offset 0x200000, size 0x400000
SF: 4194304 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2391176 Bytes = 2.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
[... 126 lines cut ...]
1f02             128 mtdblock2  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.207+ #8
Hardware name: NUC980
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:c3ad8000 r5:c3ad8000 r4:c03f613c r3:00000000
[<c0013014>] (show_stack) from [<c01bc45c>] (dump_stack+0x20/0x28)
[<c01bc43c>] (dump_stack) from [<c0076290>] (panic+0xb0/0x240)
[<c00761e4>] (panic) from [<c045c20c>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fec0 r0:c03f613c
 r7:c3fecb00
[<c045bfe8>] (mount_block_root) from [<c045c424>] (mount_root+0x68/0x84)
 r10:0000005c r9:c0478838 r8:c047f8a8 r7:c0478830 r6:c0478858 r5:c0478858
 r4:00000000
[<c045c3bc>] (mount_root) from [<c045c5a0>] (prepare_namespace+0x160/0x1c0)
 r4:c04bad44 r3:00000000
[<c045c440>] (prepare_namespace) from [<c045be34>] (kernel_init_freeable+0x184/0x1cc)
 r6:c04bad20 r5:c04bad20 r4:00000007
[<c045bcb0>] (kernel_init_freeable) from [<c03507d0>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03507c0
 r4:00000000
[<c03507c0>] (kernel_init) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

yachen, 2021-08-25:
Which SD interface are you using, GPC or GPF?

anton7191, 2021-08-25:
I use dev. board NK-NUC980_V1.1. And I use SD1 (GPF).

yachen, 2021-08-25:
Can you attach your kernel config here? Thanks.

anton7191, 2021-08-25:
[kernel.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/7044132/kernel.zip)

yachen, 2021-08-25:
Please make sure SW6 is ON and SW7 is OFF. And I just noticed you're using fat32 as file system, I don't think this is gonna work.   Maybe try ext2?
```
rootfstype=fat32 
```

anton7191, 2021-08-25:
Are these switches installed correctly?

[image]

anton7191, 2021-08-25:
I try to change Default kernel command sttring on this, but nothing changed:

Kernel command line: bootargs=noinitrd root=mmcblk0p1 rootfstype=ext2 rw console=ttyS0,115200n8 rdinit=/sbin/init mem=64M rootflags=inband-tags ignore_loglevel

And my SD card is detected like eMMC, is it normally:

U-Boot 2016.11-g4680eed-dirty (Aug 23 2021 - 21:36:07 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
MMC:   NUC980 SD: 0, NUC980 eMMC: 1
SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
=>
=> mmc list
NUC980 SD: 0
NUC980 eMMC: 1
=> mmc info
Device: NUC980 SD
Manufacturer ID: 3
OEM: 5344
Name: SC16G
Tran Speed: 25000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.8 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
=> mmc part

Partition Map for MMC device 0  --   Partition Type: DOS

Part    Start Sector    Num Sectors     UUID            Type
  1     2048            4194304         1aa3554e-01     83 Boot
  2     4196352         26919936        1aa3554e-02     83
=> mmc dev
switch to partitions #0, OK
mmc0 is current device
=>