OpenNuvoton/NUC980-linux-4.4.y #74: Problem to enable rs485 at boot time for UART4 NUC980

swapnilpanwal, 2021-08-28:
Development board: nuMaker-nuc980-serial-server
with demo example its woking fine 
but we want enable rs485 at boot time for this we have make changes in dts file as bellows
**.dts file**
uart4: serial@b0074000 {
			pinctrl-names = "default";
			pinctrl-0 = <&pinctrl_uart4_PD>;
			pdma-enable = <1>;		
			linux,rs485-enabled-at-boot-time;
			rs485-rts-delay = <0 200>;
			status = "okay";
		};
but its not working.
please suggest if we miss something else

yachen, 2021-08-30:
In Linux-4.4, serial_core.c doesn't check this option, and since NUC980's UART driver doesn't check this flag either, adding this in dts won't work.

swapnilpanwal, 2021-08-30:
Hello yachen ,
thanks for reply
we are implementing modbus rtu using libmodbus, we want to toggle rts line automatically when it send data through UART4 + RS485 physical channel
please suggest another way to implement this

yachen, 2021-08-30:
Hi, The NUC980 UART driver toggles RTS automatically in RS485 mode by default.
We have a reference sample utilize libmodbus execute on N9H30 platform in following repository.  You might want to take a look.
https://github.com/OpenNuvoton/N9H_emWin_Template/tree/master/Thermostat_N9H30_Linux/libmodbus

swapnilpanwal, 2021-08-31:
Hello Yachen,
thanks for suggestion finally we are sucessful to toggle RTS using libmodbus
bellow are the sample code to read temperature and humidity (rs485 based TH sensor)

/*********************************************/
#include <stdio.h>
#include <errno.h>
#include <modbus.h>
#include <termios.h>
#include <linux/serial.h>

#define TIOCGRS485      0x542E
#define TIOCSRS485      0x542F

int main(void) {
  modbus_t *ctx;
  uint16_t tab_reg[32];
  int ret;
  int fd;
  int status;

  struct serial_rs485 rs485conf;
  struct serial_rs485 rs485conf_bak;

  ctx = modbus_new_rtu("/dev/ttyS4", 9600, 'N', 8, 1);
  if (ctx == NULL)
  {
      fprintf(stderr, "Unable to create the libmodbus context\n");
      return -1;
  }
  modbus_set_debug (ctx, TRUE);

  modbus_set_slave(ctx, 3);

  if (modbus_connect(ctx) == -1)
  {
	  printf("Connection failed: %s\n", modbus_strerror(errno));
      modbus_free(ctx);
      return -1;
  }

  ret = modbus_rtu_set_serial_mode(ctx,MODBUS_RTU_RS485);
  ret = modbus_rtu_get_serial_mode(ctx);

  fd = modbus_get_socket(ctx);

  if (ioctl (fd, TIOCGRS485, &rs485conf) < 0)
  {
	/* Error handling.*/
	printf("ioctl TIOCGRS485 error.\n");
  }
  /* Enable RS485 mode: */
  rs485conf.flags |= SER_RS485_ENABLED;

  /* Set logical level for RTS pin equal to 1 when sending: */
  rs485conf.flags |= SER_RS485_RTS_ON_SEND;

  /* set logical level for RTS pin equal to 0 after sending: */
  rs485conf.flags &= ~(SER_RS485_RTS_AFTER_SEND);

   /* Set rts delay after send, if needed: */
   rs485conf.delay_rts_after_send = 0x80;

   if (ioctl (fd, TIOCSRS485, &rs485conf) < 0)
   {
	/* Error handling.*/
	printf("ioctl TIOCSRS485 error.\n");
   }

  if (ioctl (fd, TIOCGRS485, &rs485conf_bak) < 0)
   {
	/* Error handling.*/
	printf("ioctl TIOCGRS485 error.\n");
    }
    while(1)
    {
	 modbus_read_input_registers(ctx, 1, 2, tab_reg);
	 sleep(1);
    }
    modbus_close(ctx);
    modbus_free(ctx);
}

closed 2021-08-31 by swapnilpanwal