OpenNuvoton/NUC980-linux-4.4.y #75: System stalled after 120 hours run .  系统运行120小时后挂起

handsomehui, 2021-09-06:
Hi，
Our system is compiled base on newest NUC980 code. After 120 hours run, the system stalled. In stalled state, the system is still pingable and can be loged in with telnet, but commands like 'ifconfig/free -m/dmesg' won't get any response. My application also can't be scheduled in this state, no any logs can be printed.  Meanwhile some common command 'pwd / ls' still works fine. Our board has a externel equipment which is connected by serial, baud rate is set to 57600. In whole 120 hours, the equipment is sending data with 8000bps rate. Can you help to analysis this issue? The dumptrace is attached at the end

Thanks

请教各位：
我们使用的是基于NUC980最新代码编译的内核，运行时发现系统运行120小时之后挂起。挂起之后，系统依然可以ping通以及telnet登录，但是ifconfig、free -m 、 dmesg等常见命令都会卡死，应用程序也得不到调度不再打印log。但是pwd、ls等命令工作正常。我们的板子外接一个串口通信的设备，波特率57600，这120小时里面一直持续以8000bps的速率发送数据。求提供问题解决思路，dumptrace如下：

非常感谢

BUG: scheduling while atomic: kworker/u2:1/3536/0x00000101
Modules linked in:
CPU: 0 PID: 3536 Comm: kworker/u2:1 Not tainted 4.4.207+ #49
Hardware name: NUC980
Workqueue: writeback wb_workfn (flush-ubifs_0_0)
Backtrace:
[<c0012e60>] (dump_backtrace) from [<c001304c>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c05b7108 r4:c332e440 r3:c247c000
[<c0013034>] (show_stack) from [<c01e981c>] (dump_stack+0x20/0x28)
[<c01e97fc>] (dump_stack) from [<c0077a48>] (__schedule_bug+0x4c/0x64)
[<c00779fc>] (__schedule_bug) from [<c0438970>] (__schedule+0x3bc/0x46c)
[<c04385b4>] (__schedule) from [<c0438a74>] (schedule+0x54/0xb8)
 r8:ffffffff r7:00000002 r6:c332e440 r5:c0622538 r4:c247c000 r3:00000101
[<c0438a20>] (schedule) from [<c0438db0>] (schedule_preempt_disabled+0x28/0x44)
 r4:c247c000 r3:c247c000
[<c0438d88>] (schedule_preempt_disabled) from [<c043a60c>] (__mutex_lock_slowpath+0xb4/0x184)
[<c043a558>] (__mutex_lock_slowpath) from [<c043a714>] (mutex_lock+0x38/0x3c)
 r9:c247d240 r8:c38978b8 r7:0330c000 r6:c247d228 r5:c3308180 r4:c38ab1e4
[<c043a6dc>] (mutex_lock) from [<c021fc38>] (hw_submit+0x234/0x3c0)
[<c021fa04>] (hw_submit) from [<c021f2d0>] (nuc980_dma_tx_submit+0x78/0x7c)
 r6:c247d228 r5:004e88dd r4:c38ab1e4 r3:c021fa04
[<c021f258>] (nuc980_dma_tx_submit) from [<c0240fc8>] (nuc980_prepare_RX_dma+0x14c/0x1b8)
 r5:c06235d4 r4:c06234b0
[<c0240e7c>] (nuc980_prepare_RX_dma) from [<c024111c>] (nuc980_Rx_dma_callback+0xe8/0x110)
 r7:13613a20 r6:00000071 r5:c062369c r4:c06234b0
[<c0241034>] (nuc980_Rx_dma_callback) from [<c021f230>] (nuc980_dma_tasklet+0x128/0x150)
 r8:c062369c r7:c0241034 r6:00000002 r5:c247d280 r4:c38ab1e4 r3:c247d280
[<c021f108>] (nuc980_dma_tasklet) from [<c00217c0>] (tasklet_action+0x80/0xe8)
 r9:c05f5aa0 r8:00000100 r7:00000000 r6:c05f5a80 r5:c05b5660 r4:00000000
[<c0021740>] (tasklet_action) from [<c0021a54>] (__do_softirq+0xa4/0x220)
 r7:c247c000 r6:40000006 r5:c05f5ab8 r4:00000000
[<c00219b0>] (__do_softirq) from [<c0021e94>] (irq_exit+0xa8/0x100)
 r10:c05dd108 r9:c3923d4c r8:00000000 r7:c247d394 r6:00000000 r5:c05dd128
 r4:00000000
[<c0021dec>] (irq_exit) from [<c004c310>] (__handle_domain_irq+0x64/0xa8)
 r4:00000000 r3:c247c000
[<c004c2ac>] (__handle_domain_irq) from [<c0009024>] (asm_do_IRQ+0x24/0x28)
 r6:f0000000 r5:60000013 r4:c021f2d8 r3:c247d360
[<c0009000>] (asm_do_IRQ) from [<c0013a70>] (__irq_svc+0x30/0x70)
Exception stack(0xc247d360 to 0xc247d3a8)
d360: f0008000 00000000 00000015 00000003 c38ab0b4 c3883dc0 f0008000 c247d644
d380: 00000000 c3923d4c c05dd108 c247d3cc c247d3b0 c247d3b0 c021f5f0 c021f2d8
d3a0: 60000013 ffffffff
[<c021f46c>] (fill_desc) from [<c021fa3c>] (hw_submit+0x38/0x3c0)
 r6:c0625288 r5:c3883d80 r4:c38ab0b4 r3:00000000
[<c021fa04>] (hw_submit) from [<c021f2d0>] (nuc980_dma_tx_submit+0x78/0x7c)
 r6:c0625288 r5:000075ab r4:c38ab0b4 r3:c021fa04
[<c021f258>] (nuc980_dma_tx_submit) from [<c02930b8>] (nuc980_qspi0_txrx+0x2c8/0x78c)
 r5:c3923e10 r4:c0625288
[<c0292df0>] (nuc980_qspi0_txrx) from [<c02925d8>] (spi_bitbang_transfer_one+0x80/0xa4)
 r9:c3923d4c r8:c39e3400 r7:c3923e10 r6:c3923c00 r5:c247d644 r4:00000000
[<c0292558>] (spi_bitbang_transfer_one) from [<c028ff78>] (spi_transfer_one_message+0x164/0x4e0)
 r8:c39e3510 r7:c3923d80 r6:c247d644 r5:c3923c00 r4:c247d560 r3:00000000
[<c028fe14>] (spi_transfer_one_message) from [<c0290610>] (__spi_pump_messages+0x31c/0x514)
 r10:c028f16c r9:00000000 r8:00000000 r7:c3923c00 r6:c247d560 r5:60000013
 r4:c3923c00
[<c02902f4>] (__spi_pump_messages) from [<c0290b10>] (__spi_sync+0x2ec/0x2f8)
 r10:c028f16c r9:00000000 r8:00000000 r7:c3923c00 r6:c247d50c r5:c247d560
 r4:c39e3400
[<c0290824>] (__spi_sync) from [<c0290b30>] (spi_sync+0x14/0x18)
[... 59 lines cut ...]
[<c0082e50>] (generic_writepages) from [<c0084730>] (do_writepages+0x3c/0x40)
 r5:c3812038 r4:c2c6c178
[<c00846f4>] (do_writepages) from [<c00df660>] (__writeback_single_inode+0x30/0x1cc)
[<c00df630>] (__writeback_single_inode) from [<c00dfa54>] (writeback_sb_inodes+0x258/0x4e0)
 r8:c247c000 r7:c3ac5c00 r6:c381204c r5:c3812038 r4:c2c6c1f8 r3:c2c6c178
[<c00df7fc>] (writeback_sb_inodes) from [<c00dfd60>] (__writeback_inodes_wb+0x84/0xc8)
 r10:c2c6c1f8 r9:c05c9900 r8:c381204c r7:c247de9c r6:00000000 r5:c3ac5c00
 r4:c3812038
[<c00dfcdc>] (__writeback_inodes_wb) from [<c00dff98>] (wb_writeback+0x1f4/0x21c)
 r10:c05ca3a4 r9:c3812038 r8:c05c9900 r7:02669671 r6:02669671 r5:c247c000
 r4:c247de9c
[<c00dfda4>] (wb_writeback) from [<c00e0a98>] (wb_workfn+0x19c/0x35c)
 r10:c05c9900 r9:c00e09a8 r8:c38120bc r7:c38120c4 r6:c3812038 r5:c247c000
 r4:00000000
[<c00e08fc>] (wb_workfn) from [<c00337d8>] (process_one_work+0x140/0x388)
 r10:c24b86c0 r9:00000000 r8:00000000 r7:c3831900 r6:c3802400 r5:c38120c4
 r4:c24b86c0
[<c0033698>] (process_one_work) from [<c0033a8c>] (worker_thread+0x6c/0x5d4)
 r10:c24b86c0 r9:c3802400 r8:00000088 r7:c24b86d8 r6:c05f51d2 r5:c3802410
 r4:c3802400
[<c0033a20>] (worker_thread) from [<c0038f8c>] (kthread+0xc8/0xe4)
 r10:00000000 r9:00000000 r8:00000000 r7:c0033a20 r6:c24b86c0 r5:00000000
 r4:c2450000
[<c0038ec4>] (kthread) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
 r7:00000000 r6:00000000 r5:c0038ec4 r4:c2450000

yachen, 2021-09-06:
Please pull 81769b8 and test again. This patch replace mutex with spinlock to fix the _scheduling while atomic_ issue.

handsomehui, 2021-09-06:
Thank you yachen，
I have pulled the patch and test again.
This issue may take several days to reproduce. So I also need at least a week to verify the fix.
Will let you know when I have a solid test result.

Thanks