OpenNuvoton/NUC980-linux-4.4.y #81: NUC980 USBH Lite0无法识别

xiaolu88856, 2021-12-08:
目前在调试USB1.1读取U盘的功能，

**menuconfig配置如下：**

>  <\*>    OHCI HCD (USB 1.1) support                                                      
>                        <\*>       Support for NUC980 OHCI (USB 1.1)       
>                        [\*]         NUC980 USBH Lite0 (OHCI port3) support        
>                                      Select USBH Lite0 D+/D- pin (D+/D- is PD.15/PD.14)  --->   
>                        [ ]         NUC980 USBH Lite1 (OHCI port4) support  
>                        [ ]         NUC980 USBH Lite2 (OHCI port5) support
>                       [ ]         NUC980 USBH Lite3 (OHCI port6) support
>                       [ ]         NUC980 USBH Lite4 (OHCI port7) support
>                       [ ]         NUC980 USBH Lite5 (OHCI port8) support
>                       <\*>       Generic OHCI driver for a platform device

**开启后，日志显示：**
>usb_hcd_nuc980_probe, id = 1, name: nuc980-ohci.1, 0
>nuc980-ohci nuc980-ohci.1: unable to reserve usbh_lite0 pin
>usb_hcd_nuc980_probe, id = 2, name: nuc980-ohci.2, 0
>usb_hcd_nuc980_probe, id = 3, name: nuc980-ohci.3, 0
>usb_hcd_nuc980_probe, id = 4, name: nuc980-ohci.4, 0
>usb_hcd_nuc980_probe, id = 5, name: nuc980-ohci.5, 0
>usb_hcd_nuc980_probe, id = 6, name: nuc980-ohci.6, 0

请问是否哪里配错了？还是硬件问题？

ychuang3, 2021-12-08:
我發現 OHCI driver 在這裡剛好有個 typo error 
  drivers/usb/host/ohci-nuc980.c  line 63
p = devm_pinctrl_get_select(&pdev->dev, "usbh_lite0_pd15_pd15");
應改為 
p = devm_pinctrl_get_select(&pdev->dev, "usbh_lite0_pd15_pd14");

這應該是導致 mutil-function pin 設定失敗的原因

xiaolu88856, 2021-12-09:
> 我發現 OHCI driver 在這裡剛好有個 typo error drivers/usb/host/ohci-nuc980.c line 63 p = devm_pinctrl_get_select(&pdev->dev, "usbh_lite0_pd15_pd15"); 應改為 p = devm_pinctrl_get_select(&pdev->dev, "usbh_lite0_pd15_pd14");
> 
> 這應該是導致 mutil-function pin 設定失敗的原因

非常感谢，目前能成功识别，但是不支持热插拔，请问还有啥要配置的么？

closed 2021-12-09 by xiaolu88856